mcguirepr89/BirdNET-Pi #438: Improper species names in polish translation

macmat76, 2022-07-12:
Hi,
In the polish translation of the species names there's a redundant letter in the every name, please see below the examples:
Current name                          Proper name 
Turdus merula_Keos                Turdus merula_Kos   
Pica pica_Seroka (zwyczajna)   Pica pica_Sroka (zwyczajna)

I can modify the model/labels_l18n:labels_pl file and provide you the fix, however I'm not sure whether this file is generated from the other system, so it doesn't make sense to change it manually.

nilspupils, 2022-07-13:
Hi @macmat76: Thanks for drawing attention to these mistakes. Can i ask you to do the correction for the whole dataset and upload the file here? If that's not possible can you specify the exact error (an extra "e" on the second place at the translated Name?).
Thank you!
Best, Nils

macmat76, 2022-07-14:
Hi,
Please find new labels_pl.txt version in the attachment.
- by a regular expression in every line I've removed second character after '_' sign if it's 'e' letter
- I've prepared a simple bash script to verify whether new labels are valid (for every label it sends a request to polish wikipedia)
- there were about 200 wrong labels (see output in out.log)
- these suspicious labels were verified manually

I hope the final file is ok, please verify.

[labels_pl.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/9114380/labels_pl.txt)
[verify_labels.zip](https://github.com/mcguirepr89/BirdNET-Pi/files/9114384/verify_labels.zip)

nilspupils, 2022-07-15:
@macmat76: Awesome work and thank you so much for taking the time to do it. @mcguirepr89: Can you include the file in the updates?
Best, Nils

nilspupils, 2022-07-19:
@ehpersonal38: Ethan can you include it in the updates and close the issue? Thank you!

ehpersonal38, 2022-07-19:
Okay, I think I did that correctly. Reopen the issue if you have any problems.