mcguirepr89/BirdNET-Pi #439: Stuck after reboot without seeing BirdNET Pi screen

trophyguy, 2022-07-13:
Hello. I'm a noob with Raspberry PI so bear with me. I did the SD card install per the instructions.
Put the card in my Raspberry PI 3B+ and let it do its thing. It does some updating and looks to be installing BirdNET PI.

I've gotten stuck a couple times after trying fresh installs on this screen below.
[image]
.

Am I missing something along the way? Any help would be appreciated. 
THANK YOU!!

ehpersonal38, 2022-07-13:
Step 21 of the installation guide - https://github.com/mcguirepr89/BirdNET-Pi/wiki/Installation-Guide#:~:text=When%20the%20installation,Pi%27s%20IP%20address.

ehpersonal38, 2022-07-13:
Hmm, could you attach the installation log located at `$HOME/installation$(date "+%F").log` ?

ehpersonal38, 2022-07-13:
I will ask you to check to see if you can reach your Pi at its IP address or http://birdnetpi.local in a web browser. I think there might be a misunderstanding of how the web interface can be accessed. If that fails and there was an issue with the install then we'll proceed from there.

mcguirepr89, 2022-07-13:
Hi, @trophyguy Thanks for your interest in BirdNET-Pi and for wading into unfamiliar territory to try this new thing out for yourself. I think the others share my suspicion that there's been a bit of a misunderstanding of how you'll access and interact with your BirdNET-Pi installation.

The BirdNET-Pi will actually act as a "headless" server, meaning there is no value in having it attached to a monitor and keyboard. Instead, the BirdNET-Pi is a server on your (home?) network. It is presenting all of its data via the web server so that you can access it and interact with the installation from _other_ device on the same (home?) network. So, now that your installation is serving data on your network, you should be able to access its contents and User Interface by opening a web browser (Chrome, Firefox, or Safari) on your laptop, PC, or smartphone, and try to reach the installation's _web server_ via http://birdnetpi.local

Let us know if positing that slight misunderstanding makes things more clear

Sincerely,
Patrick

trophyguy, 2022-07-13:
I've done networking within my home Windows based before, so please bear with me. Sorry

In the screen shot above, the Raspberry PI is all set then and waiting for me to log in
over my home network from my regular desktop correct? I have the PI hooked up to an ethernet cable
and was planning on doing that when it's outdoors, so as long as it's on the same network connection
I can log in with the birdnetpi.local credentials I created when I wrote to the SD card?

trophyguy, 2022-07-13:
I have tried logging in at http://birdnetpi.local (what I set the hostname to) on my other desktop
and get Site Can Not Be Reached. I tried the IP address also and with both I get refused to connect.
I did enable SSL during setup which means it needs the username and password I created?

The desktop and PI are both wired to the same router and they both have IP's of 192.168.1.xx

Sorry again.  I'm very good with tech but Linux is throwing me for a bit of a loop.
THANKS!
Tony

trophyguy, 2022-07-14:
I started from scratch. Used Windows Power Shell to successfully log into the PI.
I was also able to add birdnetpi1.local to the list of known hosts. It looks like an IP V6 address also.
The connection then closes without asking me to use the password I created.

[image]

mcguirepr89, 2022-07-14:
Hi, @trophyguy -- Based on the screen shot you provided above, I would assume that you have `birdnetpi1.local` pointed to the wrong device -- if you didn't get an SSH password prompt from port 22 from that host, it wasn't the BirdNET-Pi. That is the only explanation I can come up with based on what I see. It is possible to add just about any device to a list of known_hosts so long as the system is *nix -- unfortunately, that doesn't tell us anything about the server you're communicating with other than it is not accepting SSH connections from you.

But, to try to clarify, after installing BirdNET-Pi, your access to it thereafter will be from _web browsers_ on other devices, even for administrative tasks at the system level. If you can't access the web user interface, then there is either a problem with the web server itself or issues with how your network is configured for this type of traffic.

>I did enable SSL during setup which means it needs the username and password I created?

How exactly did you enable SSL during setup? This is a step with which I am unfamiliar and knowing what steps you took to try to enable SSL during setup may be helpful during this troubleshooting.

mcguirepr89, 2022-07-14:
I clicked the wrong button and didn't not intend to close the issue!! Sorry about that!

trophyguy, 2022-07-14:
> How exactly did you enable SSL during setup? This is a step with which I am unfamiliar and knowing what steps you took to try to enable SSL during setup may be helpful during this troubleshooting.

I mis typed and said SSL when I meant SSH in step number 7 of your installation instructions.

I can see the BirdNET-PI on my local network when logging into my router. 
Seems I have some kind of network issue that is preventing me being able to access it.
I'll keep plugging along. Thank you for your help Patrick.

mcguirepr89, 2022-07-14:
You're quite welcome, and that typo makes perfect sense, now.

I'm happy to take part in a remote desktop session if you would like me to take a look at what might be going on. If interested, please [email me](mailto:mcguirepr89@gmail.com) and we will arrange a remote work session to see what we can find. Otherwise, please just keep us updated on what you observe and we'll do our best to get your system working.

Sincerely, 
Patrick

mcguirepr89, 2022-07-17:
@trophyguy Tony is all set now with a working installation -- closing this issue~~

closed 2022-07-17 by mcguirepr89

trophyguy, 2022-10-11:
> I think it's installed successfully. You need to access it from a browser at: http://birdnetpi.local/

Would I do that here on the Raspberry Pi terminal or remote?

Should I install the OS with a visible desktop since I did it without? After using Windows for 30+ years this is quite different for me.

Thank you
Tony

mcguirepr89, 2022-10-11:
Hi, @trophyguy Tony, 

I'm not sure exactly which parts of the installation are proving difficult, but I would be very happy to help give the installation a go in a remote desktop session. If interested, please [email me](mailto:mcguirepr89@gmail.com) and I will provide you with some instructions to share your (Windows) desktop. If that is not something you want to do, I'm afraid I only know to suggest running through the installation guide again and then reaching out during the specific steps where you are unsure.

My best regards,
Patrick

trophyguy, 2022-10-11:
I have no clue where that comment came from. I may have posted that when I initially started this months ago. I didn't submit that comment again so no idea why thats there. I was sleeping when that posted.

ehpersonal38, 2022-10-11:
Yeah there's some weird things going on with GitHub today. It's almost like email replies to posts got stalled for months and are only now being posted.