mcguirepr89/BirdNET-Pi #442: Unable to start birdnet_server.service

jds3d, 2022-07-13:
### Discussed in https://github.com/mcguirepr89/BirdNET-Pi/discussions/441

<div type='discussions-op-text'>

<sup>Originally posted by **jds3d** July 13, 2022</sup>
I had some issues with my install and decided to uninstall and reinstall.  Now I cannot get the web server to load, so birdnetpi1.local shows nothing.  I tried to restart_services.sh and see:

+ logger '[/usr/local/bin/restart_services.sh] Unable to start birdnet_server.service... Looping until it start properly'

I do not know what to do to get that service to start or whether I'm even barking up the right tree.</div>

ehpersonal38, 2022-07-13:
Hey there, sorry to hear you're having trouble! If you could supply the output of `sudo systemctl status birdnet_server.service`, I think that will help make progress towards solving this issue.

jds3d, 2022-07-13:
● birdnet_server.service - BirdNET Analysis Server
     Loaded: loaded (/home/birder/BirdNET-Pi/templates/birdnet_server.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-07-13 16:02:51 EDT; 3h 37min ago
   Main PID: 142129 (python3)
      Tasks: 8 (limit: 4164)
        CPU: 29.988s
     CGroup: /system.slice/birdnet_server.service
             └─142129 /home/birder/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/server.py

Jul 13 16:02:51 birdnetpi1 systemd[1]: Started BirdNET Analysis Server.
Jul 13 16:03:21 birdnetpi1 python3[142129]: INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
Jul 13 16:03:21 birdnetpi1 python3[142129]: INFO: Created TensorFlow Lite delegate for select TF ops.
Jul 13 16:03:21 birdnetpi1 python3[142129]: INFO: TfLiteFlexDelegate delegate: 1 nodes delegated out of 182 node>

Hmm.  That looks pretty good, but I still get:
This site can’t be reached
birdnetpi1.local refused to connect.
Try:

Checking the connection
[Checking the proxy and the firewall](chrome-error://chromewebdata/#buttons)
ERR_CONNECTION_REFUSED

mcguirepr89, 2022-07-14:
If I may step in here, I don't think the issue with the web server loading is related to the `birdnet_server.service`

Would you mind providing some log output from the webserver and from PHP?

```
journalctl --no-hostname --no-pager -n200 -ucaddy -uphp7.4-fpm
```

jds3d, 2022-07-14:
Thanks, Patrick!  I appreciate both of your help with this.  

```
birder@birdnetpi1:~/BirdNET-Pi/scripts$ journalctl --no-hostname --no-pager -n200 -ucaddy -uphp7.4-fpm
-- Journal begins at Sat 2022-07-02 01:31:01 EDT, ends at Thu 2022-07-14 08:31:00 EDT. --
Jul 11 10:35:01 caddy[941]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
Jul 11 10:35:01 caddy[941]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
Jul 11 10:35:01 caddy[941]: runtime.GOOS=linux
Jul 11 10:35:01 caddy[941]: runtime.GOARCH=arm64
Jul 11 10:35:01 caddy[941]: runtime.Compiler=gc
Jul 11 10:35:01 caddy[941]: runtime.NumCPU=4
Jul 11 10:35:01 caddy[941]: runtime.GOMAXPROCS=4
Jul 11 10:35:01 caddy[941]: runtime.Version=go1.18.1
Jul 11 10:35:01 caddy[941]: os.Getwd=/
Jul 11 10:35:01 caddy[941]: LANG=en_GB.UTF-8
Jul 11 10:35:01 caddy[941]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
Jul 11 10:35:01 caddy[941]: NOTIFY_SOCKET=/run/systemd/notify
Jul 11 10:35:01 caddy[941]: HOME=/var/lib/caddy
Jul 11 10:35:01 caddy[941]: LOGNAME=caddy
Jul 11 10:35:01 caddy[941]: USER=caddy
Jul 11 10:35:01 caddy[941]: INVOCATION_ID=1c72e6ed350d41ae8cd8aa49425c55c0
Jul 11 10:35:01 caddy[941]: JOURNAL_STREAM=8:14303
Jul 11 10:35:01 caddy[941]: run: reading config file: open /etc/caddy/Caddyfile: no such file or directory
Jul 11 10:35:01 systemd[1]: caddy.service: Main process exited, code=exited, status=1/FAILURE
Jul 11 10:35:01 systemd[1]: caddy.service: Failed with result 'exit-code'.
Jul 11 10:35:01 systemd[1]: Failed to start Caddy.
Jul 11 10:56:12 systemd[1]: Stopping The PHP 7.4 FastCGI Process Manager...
Jul 11 10:56:12 systemd[1]: php7.4-fpm.service: Succeeded.
Jul 11 10:56:12 systemd[1]: Stopped The PHP 7.4 FastCGI Process Manager.
-- Boot b2afb416d9c645d7bb5fdc1c79a8b754 --
Jul 11 10:56:18 systemd[1]: Starting The PHP 7.4 FastCGI Process Manager...
Jul 11 10:56:20 systemd[1]: Started The PHP 7.4 FastCGI Process Manager.
Jul 11 10:56:34 systemd[1]: Starting Caddy...
Jul 11 10:57:00 caddy[959]: caddy.HomeDir=/var/lib/caddy
Jul 11 10:57:00 caddy[959]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
Jul 11 10:57:00 caddy[959]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
Jul 11 10:57:00 caddy[959]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
Jul 11 10:57:00 caddy[959]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
Jul 11 10:57:00 caddy[959]: runtime.GOOS=linux
Jul 11 10:57:00 caddy[959]: runtime.GOARCH=arm64
Jul 11 10:57:00 caddy[959]: runtime.Compiler=gc
Jul 11 10:57:00 caddy[959]: runtime.NumCPU=4
Jul 11 10:57:00 caddy[959]: runtime.GOMAXPROCS=4
Jul 11 10:57:00 caddy[959]: runtime.Version=go1.18.1
Jul 11 10:57:00 caddy[959]: os.Getwd=/
Jul 11 10:57:00 caddy[959]: LANG=en_GB.UTF-8
Jul 11 10:57:00 caddy[959]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
Jul 11 10:57:00 caddy[959]: NOTIFY_SOCKET=/run/systemd/notify
Jul 11 10:57:00 caddy[959]: HOME=/var/lib/caddy
Jul 11 10:57:00 caddy[959]: LOGNAME=caddy
Jul 11 10:57:00 caddy[959]: USER=caddy
Jul 11 10:57:00 caddy[959]: INVOCATION_ID=ccc0dbd1f1ce43019077f4dd55e6307e
Jul 11 10:57:00 caddy[959]: JOURNAL_STREAM=8:9189
Jul 11 10:57:00 caddy[959]: run: reading config file: open /etc/caddy/Caddyfile: no such file or directory
Jul 11 10:57:00 systemd[1]: caddy.service: Main process exited, code=exited, status=1/FAILURE
Jul 11 10:57:00 systemd[1]: caddy.service: Failed with result 'exit-code'.
Jul 11 10:57:00 systemd[1]: Failed to start Caddy.
-- Boot 663d753418c44027ae1f881eeadc574c --
Jul 12 17:36:29 systemd[1]: Starting The PHP 7.4 FastCGI Process Manager...
Jul 12 17:36:31 systemd[1]: Started The PHP 7.4 FastCGI Process Manager.
Jul 12 17:37:00 systemd[1]: Starting Caddy...
Jul 12 17:37:01 caddy[1166]: caddy.HomeDir=/var/lib/caddy
Jul 12 17:37:01 caddy[1166]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
Jul 12 17:37:01 caddy[1166]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
Jul 12 17:37:01 caddy[1166]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
Jul 12 17:37:01 caddy[1166]: caddy.Version=v2.5.1 h1:bAWwslD1jNeCzDa+jDCNwb8M3UJ2tPa8UZFFzPVmGKs=
Jul 12 17:37:01 caddy[1166]: runtime.GOOS=linux
Jul 12 17:37:01 caddy[1166]: runtime.GOARCH=arm64
Jul 12 17:37:01 caddy[1166]: runtime.Compiler=gc
Jul 12 17:37:01 caddy[1166]: runtime.NumCPU=4
[... 168 lines cut ...]
I was able to revert to an old version, and it appears to be working!

```
birder@birdnetpi1:/etc/caddy$ sudo rm Caddyfile
birder@birdnetpi1:/etc/caddy$ sudo cp Caddyfile.original Caddyfile
birder@birdnetpi1:/etc/caddy$ cat Caddyfile
http://  {
  root * /home/birder/BirdSongs/Extracted
  file_server browse
  handle /By_Date/* {
    file_server browse
  }
  handle /Charts/* {
    file_server browse
  }
  reverse_proxy /stream localhost:8000
  php_fastcgi unix//run/php/php7.4-fpm.sock
  reverse_proxy /log* localhost:8080
  reverse_proxy /stats* localhost:8501
  reverse_proxy /terminal* localhost:8888
}

```

It's a mystery for me what I did to break it, but your log knowledge helped me find and fix it, so thanks!

mcguirepr89, 2022-07-14:
That's great!!! If you need anything else, please let us know as we're always eager to help. I will go ahead and close this issue, but please do reopen if anything else seems to be related to this (of course feel invited to open a new issue for any unrelated problems).

My very best regards,
Patrick

closed 2022-07-14 by mcguirepr89