mcguirepr89/BirdNET-Pi #446: Use SSD drive?

jungle-boogie, 2022-07-14:
hello,

I discovered this cool project only today, so I don't know a whole lot about it yet.

I'm a little curious about this statement:
> 24/7 recording and BirdNET-Lite analysis

I didn't think it would be safe to write to an SD card all the time, especially something streaming to the SD card full time.

Has anyone noticed SD cards dying at a faster rate when having it under a heavy write load?

I now think the raspberry pi installer will let you write the image directly to an SSD drive, and not require the micro SD card.

On an unrelated note, it looks like `gotty` hasn't been updated since December 2017, with the last release in August 2017.
Do you think this is still a good project to use in `birdnet`?

Thanks!

nilspupils, 2022-07-14:
Hi @jungle-boogie: So far i think we had reports about one broken sd-card after months of use. You still have a valid point though and one of the next versions of the system will do most of the work in RAM and only write to the card when it actually has an extraction.
Best, Nils

jds3d, 2022-07-14:
@nilspupils was it proven that the card failed due to overuse?  These cards are subjected to the elements, rain, lightning, excessive heat and cold.  It very easily could have been fried by something other than reading and writing.

jungle-boogie, 2022-07-14:
Thank you both for your input. Good news about doing most of the work in RAM.

Second question, if you're recording 24/7, how are the recordings rolled over?

jungle-boogie, 2022-07-14:
@jds3d, I did a search and the answers range from 5 - 10 years, or ~10,000 write/read cycles. All the sites I read indicate flash memory isn't viable for long term storage.

CaiusX, 2022-07-14:
It was my card that died, and highly likely due to being on an unhoused pi close to an open window (rain essentially). I have been running several other installations - at least a total of a year and a half of continuous operation  for one machine with no other problems.

Best
AP

ehpersonal38, 2022-07-14:
@jungle-boogie There's an option in Tools -> Settings -> Advanced settings to change this behavior:
[image]

With the purge option, your "Best Recordings" will be excluded automatically, and you can manually exclude specific recordings as well. But all other audio files will get deleted permanently if your disk becomes full (>95% used).

jungle-boogie, 2022-07-14:
I see this was asked about in a discussion as well:
https://github.com/mcguirepr89/BirdNET-Pi/discussions/247

closed 2022-07-14 by jungle-boogie