mcguirepr89/BirdNET-Pi #456: Enhancement - throttle alerts on infrequent visit identification

GeekNJ, 2022-07-18:
The notification setting for "Notify each new infrequent species detection (<5 visits per week)" triggers multiple emails on a single visit. I have the option enabled and often when triggered I get 3+ emails in a row. 

An adjustment might be that an alert/notification is sent only after a specified minimum time from the prior identification. As an example, using data already available in the db, trigger the alert based on the current criteria and also "not within the last 2 minutes". Not sure if 2 mins or 5 mins would be optimal, but it would help reduce the flurry of emails for a single new visit.

ehpersonal38, 2022-07-18:
I see the use of this, but I also want to consider the fact that by setting a "maximum notifications sent per minute," we are losing data. So if a species has say 4 detections all within the span of a few minutes, you'll get 1 notification but 4 DB entries. That may be a deal breaker for people who want the most accurate data.

I'd love to hear more thoughts from people that actively use the "Notify each new infrequent species" option though.

ehpersonal38, 2022-08-17:
Okay I've done work on this, you can test it out on the development branch:
`bash ~/BirdNET-Pi/scripts/update_birdnet.sh -b development`

Then you'll find a new setting:
[image]

Thinking I'll merge it into main in about a week seeing as there are no issues reported by anyone testing it. I'll run it for a few days too and see. 🐤

closed 2022-09-30 by ehpersonal38