mcguirepr89/BirdNET-Pi #462: Enhancement - standardization of of info/links displayed above recordings

GeekNJ, 2022-07-19:
The info shared above a recording window is different across the pages. I'm not sure if that was intentional or just how things played out. Here's what I notice across the various pages/areas and it would be nice if it was standardized where the data fields makes sense. I really miss the links in some spots and and why this is being requested. How do others feel?

**Suggestion** :
Common Name (Link to AllAboutBirds)
Scientific Name (Link to Wikipedia)
Confidence % or Max Confidence %
Date Time
Number of Occurrences for the species 

The main **overview** shows above each recording: 
Common Name (Link to AllAboutBirds)
Scientific Name (Link to Wikipedia)
Confidence %

The **Today's Detections** page shows over each recording:
Time
Common Name (Link to AllAboutBirds)
Scientific Name (Link to Wikipedia)
Confidence %

**Best Recording**s page shows over each recording:
Common Name (not linked)
Scientific Name (Link to Wikipedia)
Number of Occurrences
Max confidence %
Date Time
All About Birds (link)

**Recordings** page shows over each recording:
Common Name (nor linked)
Date Time
Confidence %

ehpersonal38, 2022-07-19:
I think this is a good idea! The "Common Name" on Overview actually links to the Best Recording page for that species. Is that something anyone is interested in preserving or is the AllAboutBirds link preferred?

Sp4rkR4t, 2022-07-20:
I'm not a fan of the allaboutbirds links, for info I always clicked through the latin for the wiki. Standardising the info for each section would be a great help in navigating the UI.

closed 2022-12-13 by mcguirepr89