mcguirepr89/BirdNET-Pi #474: Arecord hangs

matthew73210, 2022-07-30:
Here's a quick issue for futur reference, I'm on an orange pi 4 lts so probs not a supported device yet.

Issue : 
Every so ofter I notice that birdnet has stopped recording, the spectorgrah will have a date/time combo that is way off. If I go into system info arecord is stopped. Here's a dump from journalctl --no-pager

`Jul 30 09:19:55 birdnetpi systemd[1]: birdnet_recording.service: Main process exited, code=exited, status=1/FAILURE
Jul 30 09:19:55 birdnetpi systemd[1]: birdnet_recording.service: Failed with result 'exit-code'.
Jul 30 09:19:55 birdnetpi systemd[1]: birdnet_analysis.service: Succeeded.
Jul 30 09:19:56 birdnetpi kernel: sprdwl:sprdwl_tx_ba_mgmt, 1636, tx_addba failed, reason=1, lut_index=6, tid=0, map=0
Jul 30 09:19:57 birdnetpi systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 7196.
Jul 30 09:19:57 birdnetpi systemd[1]: Stopped BirdNET Analysis.
Jul 30 09:19:57 birdnetpi systemd[1]: Started BirdNET Analysis.
Jul 30 09:19:58 birdnetpi systemd[1]: birdnet_recording.service: Scheduled restart job, restart counter is at 6091.
Jul 30 09:19:58 birdnetpi systemd[1]: Stopped BirdNET Recording.
Jul 30 09:19:58 birdnetpi systemd[1]: Started BirdNET Recording.
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: + source /etc/birdnet/birdnet.conf
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ LATITUDE=45.5558
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ LONGITUDE=6.7395
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ BIRDWEATHER_ID=6Mnq3vi4YFSyoPnEPxW4xpXT
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ CADDY_PWD=
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ ICE_PWD=birdnetpi
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ BIRDNETPI_URL=
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ RTSP_STREAM=
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ APPRISE_NOTIFICATION_TITLE='New BirdNET-Pi Detection'
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ APPRISE_NOTIFICATION_BODY='A $sciname $comname was just detected with a confidence of $confidence at $time. $listenurl'
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ APPRISE_NOTIFY_EACH_DETECTION=0
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ APPRISE_NOTIFY_NEW_SPECIES=0
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ APPRISE_WEEKLY_REPORT=1
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ FLICKR_API_KEY=6c8751e7a408b5c9801794a80484daef
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ FLICKR_FILTER_EMAIL=
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ RECS_DIR=/home/matthew/BirdSongs
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ RECS_DIR=/home/matthew/BirdSongs
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ REC_CARD=default
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ PROCESSED=/home/matthew/BirdSongs/Processed
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ EXTRACTED=/home/matthew/BirdSongs/Extracted
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ OVERLAP=0.0
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ CONFIDENCE=0.75
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ SENSITIVITY=1.25
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ CHANNELS=2
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ FULL_DISK=purge
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ PRIVACY_THRESHOLD=0
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ RECORDING_LENGTH=15
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ EXTRACTION_LENGTH=
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ AUDIOFMT=mp3
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ DATABASE_LANG=en
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ SILENCE_UPDATE_INDICATOR=0
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ LAST_RUN=
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ THIS_RUN=
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ IDFILE=/home/matthew/BirdNET-Pi/IdentifiedSoFar.txt
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: ++ APPRISE_NOTIFY_NEW_SPECIES_EACH_DAY=1
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: + '[' -z 15 ']'
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: + '[' '!' -z ']'
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: + pulseaudio --check
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575044]: E: [pulseaudio] core-util.c: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: + pulseaudio --start
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575047]: E: [autospawn] core-util.c: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575047]: W: [autospawn] lock-autospawn.c: Cannot access autospawn lock.
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575047]: E: [pulseaudio] main.c: Failed to acquire autospawn lock
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: + pgrep arecord
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: + grep 5050 /dev/fd/63
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575050]: ++ netstat -tulpn
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575051]: tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      1647/python3
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: + '[' -z default ']'
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575043]: + arecord -f S16_LE -c2 -r48000 -t wav --max-file-time 15 -D default --use-strftime /home/matthew/BirdSongs/%B-%Y/%d-%A/%F-birdnet-%H:%M:%S.wav
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575053]: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Jul 30 09:19:58 birdnetpi birdnet_recording.sh[575053]: arecord: main:830: audio open error: Device or resource busy
Jul 30 09:19:58 birdnetpi systemd[1]: birdnet_recording.service: Main process exited, code=exited, status=1/FAILURE
Jul 30 09:19:58 birdnetpi systemd[1]: birdnet_recording.service: Failed with result 'exit-code'.
Jul 30 09:19:58 birdnetpi systemd[1]: birdnet_analysis.service: Succeeded.`

This whole bit keeps on repeating itself, and if I go into view log nothing is happening. When I log in via the web terminal or ssh the whole thing restarts again and starts working.

I'm getting a lot of this sort of error too 
`birdnetpi kernel: sprdwl:sprdwl_tx_ba_mgmt, 1636, tx_addba failed, reason=1, lut_index=6, tid=0, map=0`

The installation went through without a hitch, what I'm going to try and do is redo an install. I could parse the install log if you want prior to re-installing?

Cheers

matthew73210, 2022-07-30:
Funny enough if I leave myself connected via ssh the service never hangs, it's only when I dissconect.

matthew73210, 2022-07-30:
Re-install did nothing, so jumped back on my old pi 3 to use as a rtsp server, works just fine. For some reason when bird pi has finished with the audio it causes ffmpeg to crash but under a service it just restarts and it doesn't really bother me. So it seams that it was the microphone not being available causing all the issues.
Feel free to close if you want :)

matthew73210, 2022-07-31:
Okay so after reading through discussions it seams to be linked to this [375](https://github.com/mcguirepr89/BirdNET-Pi/discussions/375). Next weekend I'll reconnect the mic back to the orange pi and try using an autologin script pointing to tty1.

mcguirepr89, 2022-08-04:
@matthew73210 -- I'm sorry for the delay in responding, but you've figured out the issue.

PulseAudio needs a runtime directory in user space that can be satisfied with an autologin on tty1 -- alternatively, you can bypass PulseAudio altogether and go straight to ALSA and specify the device by its name from the command output for `arecord -L`:

On my machine, this produces:
```
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Playback/recording through the PulseAudio sound server
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front output / input
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
usbstream:CARD=Device
    USB Audio Device
    USB Stream Output
usbstream:CARD=vc4hdmi0
    vc4-hdmi-0
    USB Stream Output
usbstream:CARD=vc4hdmi1
    vc4-hdmi-1
    USB Stream Output
```

You will choose the _NON_-indented device name from your list that corresponds to the sound card you wish to use. I would use `dsnoop:CARD=Device,DEV=0` since I have a `dsnoop` option, otherwise one can always use the `hw:` option (for me that would be `hw:CARD=Device,DEV=0`).

The advantage of the `dsnoop` option is that it will serve the live audio stream playback and the analysis -- if your device doesn't support that, you won't be able to use the sound card for live audio streaming _and_ analysis, and would have to forego using the Live Audio stream. (This can be worked around with other ALSA plugins, but it's a little beyond the scope of the support I tend towards without deciding it is absolutely needed.)

Anywho, copy and paste the entry that you need into your "Tools" > "Settings" > "Advanced Settings" > "Audio Card" entry and "Update Settings" (full disclosure: you may need to restart services for those settings to take: "Tools" > "Services" > "Restart Core Services")

Fingers crossed for you on this hardware :crossed_fingers:

matthew73210, 2022-08-06:
I tried that a while ago but to no success, I think I grabbed the wrong card as I had over 20 options for the same card. I'll give it a try tomorrow.
I'll try getty for the autologin

matthew73210, 2022-08-07:
I'll close this issue as I don't use that setup right now. If in the future it pops up again when I change my setup, I'll refer to this issue.

closed 2022-08-07 by matthew73210