mcguirepr89/BirdNET-Pi #478: Filter by location and date

krummrey, 2022-08-03:
I think the original version of BirdNET filters possible results by location and date to lessen the chance of false positives.
I live in northern Germany and have a pink flamingo and a puffin in my detected list. Both could reliably be filtered out based on my location. 
Is that data available? Could that feature be implemented?
BTW How would I delete those false positives on my local installation?

jds3d, 2022-08-03:
You should enter your lat/lon in the tools/settings section.  I believe
that's all you need.

ehpersonal38, 2022-08-03:
I think the model we currently use does try to intelligently detect species based on the lat/lon/day of week you provide, but I've noticed that it's not very accurate. Hopefully the new model will be released soonish and it will be implemented, but for now you can use my hacky method here if you want better results: https://github.com/mcguirepr89/BirdNET-Pi/discussions/374#discussioncomment-2940777

You can delete detections by clicking the icon in the top right of one, then on the page that gets opened, click the trash can.

jds3d, 2022-08-03:
> you want better results: #374 (comment)

I think you mean "week of year", not "day of week"?

ehpersonal38, 2022-08-03:
Yep, week of year.

closed 2022-08-10 by mcguirepr89