mcguirepr89/BirdNET-Pi #479: BirdNET conked out on Update

r32moose, 2022-08-04:
BirdNet stopped analyzing new recordings.  Checked the System Tools on the webpage, showed a red badge on Updates (113).  Pressed Update, page became unresponsive.  Rebooted.  Update badge had disappeared but still not analyzing.  Checked the Log and this is the info that repeats every couple of minutes.  Pi is booted and running, but now the webpage is really sluggish and unresponsive, hangs.  Have to reboot Pi to get it to do anything.

[image]

ehpersonal38, 2022-08-04:
Sorry to see your installation is having trouble. Could you try another update from the command line and provide the output? 

`bash ~/BirdNET-Pi/scripts/update_birdnet.sh`

r32moose, 2022-08-04:
Tried it twice.  It runs the update and then ends trying to restart the birdnet server, but it just loops without restarting.  I went back to the Log and it shows the same file information missing as I previously posted.

mcguirepr89, 2022-08-04:
Hi, @r32moose -- after attempting the update, could you please provide the output (if you have to stop the update with Ctrl + C, then everything up to that point). I cannot say what broke during your update, but how it fails will probably be helpful information.

From there, I may ask that you force things along with a few commands, but we'll see what those may be with some more information about your update issues.

Frono, 2022-08-09:
I had the same issue. I ran the update and nothing worked. Mine was down for a few days, but when I pulled it down to look at it the disk was full. So I deleted all the birdsongs to get enough space to work with.

- Tried a lot of things, (rebooting, running the git update script, restarting services, etc) but never could get it to start all the BirdNetPi services (web etc)...
- $ cd ~
- $ cp ./BirdNET-Pi/birdnet.conf .
- $ cp ./BirdNET-Pi/scripts/birds.db .
-  $ sudo rm -R BirdNET-Pi
- $ rm -R Birdsongs
- Copied my birdnet.conf and birds.db to the new install in an attempt to at least keep the history of birds identified. Services still would not work.
- gave up and ran ~/BirdNET-Pi/scripts/clear_all_data.sh

Everything runs fine now, with my old conf file but no history exists.

mcguirepr89, 2022-08-09:
Hi, @Frono -- like with @r32moose, I can only say that I would need some of the information it seems is now totally lost in order to be able to help. Any bit of logging information is helpful when you notice a problem. If you encounter any issues in the future, please preserve some of the information about the problem (assuming you'd like us to help get to the bottom of it) as it will be needed to get beyond the "same here" phase of troubleshooting.

Frono, 2022-08-09:
@mcguirepr89 
No worries at all, I found this after I'd fixed it. I was more trying to socialize my method to get BirdNet running again after a failure. Was hoping for feedback... I have log files and the old DB if either of those help. I figured everything would be screwed up on mine since disk space was zero...

mcguirepr89, 2022-08-09:
@Frono thanks for providing this info for others ~~ May I ask approximately how long your system was running before it had trouble?

Frono, 2022-08-09:
It failed immediately after I upgraded it from the web interface.

Do you have some text that would have been logged for the upgrade in /var/log/ ?
I can try to find the upgrade event and get you some info...

mcguirepr89, 2022-08-09:
I'm sorry, I meant how long had you been using BirdNET-Pi before attempting to update

Depending on your version, the update would have logged in the web browser with an option to copy all of its output upon success or failure. If you didn't see anything like that, that tells me that your system was an older version (kind of really old, relatively speaking).

mcguirepr89, 2022-08-09:
>Do you have some text that would have been logged for the upgrade in /var/log/ ?
I can try to find the upgrade event and get you some info...

BTW -- I know a great suggestion/improvement when I see one -- thanks!! I will add a behind-the-scenes logger for the update to help with this type of thing. It has been great having the feedback go right into the web browser, but it seems that may be too soon to determine whether the update created any problems. :thinking:

Sorry, again, for the system implosion!

Frono, 2022-08-09:
I do not believe my version was very old as I have updated from the panel as recently July 7th.
Forgive me as I'm not super technical, but I'm not really sure how to see what the version of the software is from the Web frontend. As a result I'm not very plugged into version info beyond I ran the update when I saw that I was XX commits behind...

mcguirepr89, 2022-08-09:
@Frono -- the mist and uncertainty around the version is an unfortunate feature/characteristic of BirdNET-Pi. I've never known (or have been too recalcitrant towards better praxis) how to handle the many rapid changes the system undergoes during development and collaboration. Your scenario is a great impetus for revisiting this issue in search of a plan to which we can all adhere. @ehpersonal38 , this is so tough, but maybe we can figure out a release "schedule" to push changes to a development branch that gets pulled into `main` weekly -- then we can have version numbers that reflect the week and can then go into the code for that week to see what BirdNET-Pi looked like at that time. Thoughts?

Frono, 2022-08-09:
If you made it monthly, you could go with the standard 2022.01 versioning convention. Also maintaining the old versions as branches would allow people to load prior versions if there is a problem with current main.

Frono, 2022-08-09:
Oh, on the actual problem. I just remembered that a number of the files in the files in the scripts directory were 0 bytes when I looked this am, but disk integrity tested ok... Is there a chance that the hard disk is so full with recordings that there was not enough room to do the full update process?

mcguirepr89, 2022-08-10:
@Frono My supposition is that you had a bug in your older version that let the disk fill (thaat was me that birthed that bug -- but I also squashed it eventually), also preventing the update from succeeding, were I to make a guess at this stage.

krummrey, 2022-08-17:
Here ist my output of the update. It's been like that for every update. It hangs and I have to reboot. Works after the reboot though.
`+ logger '[/usr/local/bin/restart_services.sh] Unable to start birdnet_server.service... Looping until it start properly'
`
https://pastebin.com/5b398g9q

mcguirepr89, 2022-08-17:
Hi, @krummrey -- can you confirm which model is running your system? I may need to adjust some documentation based on your output from:

```
grep Model /proc/cpuinfo
```

krummrey, 2022-08-17:
`Model		: Raspberry Pi 3 Model B Rev 1.2`

mcguirepr89, 2022-08-22:
Closing this issue since it is not specific enough to help anyone with their issues.

closed 2022-08-22 by mcguirepr89