mcguirepr89/BirdNET-Pi #485: Add option to specify quantity of free disk space to trigger "purge or keep"

Frono, 2022-08-10:
I have had a problem where my disk filled up and it may or may not have been due to birdnetpi. I have a fairly large SSD and really do not need to keep a whole disk worth of recordings, as the detection statistics (Species, Times, Dates, etc...) are what I'm really interested in preserving year over year.

If everybody agrees, I'd like to see an option in the advanced menu that either specifies a level at which purge/keep operates based on how full the birdsong folder is, or an option to just say "Full disk is considered to be less than XXXgb free space" or 
"Full disk is considered to be less XX% free space" 
The measurement could be in percentage to make it easier for a default, or an absolute. It would be very handy for me either way.

DD4WH, 2022-08-10:
Have a look at Tools --> System Info 
In the upper part it says "Usage" of the SD card in %

Would that be sufficient ?

mcguirepr89, 2022-08-22:
This feature is available in the new repo -- I may backport the code to this repo, but wanted to let you all know that the new repo will allow this to be user-defined.

DD4WH, 2022-08-22:
@mcguirepr89 
Patrick, I feel a little lost.
What is the "new repo"? Did you start a different project?

mcguirepr89, 2022-08-22:
The new repo is my fork of the new model's repo, https://github.com/mcguirepr89/BirdNET-Analyzer-Pi.git

It is still very much in development, but Stefan released the 3K model earlier last week and I've been putting together a new BirdNET-Pi around that ever since.

BirdNET-Pi's success has surprised me, to say the absolute least. As (one of) its steward(s), I do want to improve how it is put together and how others develop around it.

I've been doing a shit load of homework over the last several weeks to get a feel for what type of development would suit the needs for this unexpected success while providing fertile scaffolding for continued growth.

With the homework complete, Laravel 9 is the framework I've chosen to use to rebuild BirdNET-Pi around the new model.

It will be superior in all regards, but will be in development for a long while (I suspect).

Basically, once this current repo became popular, more feature requests and bugs cropped up than I can sustainably address. That was, of course, a result of putting all this together while learning as I go.

The new Laravel platform will let me address bugs and feature requests with much less "hacking" and much better development practices. 

As it is now, were I to disappear, this project would need someone who understands how _I_ put the system together in order to make any meaningful adjustments. My goal with this new build is that _I_ will follow best practices as much as I possibly can so that my savoir faire is _not_ required for folks who wish to develop around (or customize) the software.

I will be sure to let you know when the new repo can be tested -- as it is now, the core services are intact and I am building out the UI--

There are still SOOOO many BirdNET-Pi features that I need to port over, but many of them will disappear because either a) the new model handles the scenarios for which we've built tool so it is no longer needed, or b), Laravel is better isolated from the underlying system, so less configuration is needed overall.

With all of that said, I've valued your testing since first making this project public, so I am of course 100% eager to get you testing sooner than later, though I don't think the model's species list is quite where it would need to be for your purposes.

Also, instead of an installer, I will (I'm about 90% sure on this) only offer pre-installed images so that I can better support the software. The mysteries that surround what folks _actually_ do with their installations is getting old :) -- I'd rather know that if the system booted, everything is installed correctly because I'm the one who installed it.

Anywho, there is SOOOOO much about the new iteration to discuss, but I'm not quite at a place where I'm open for discussion just yet -- I don't want to open the feature-request flood-gates (from the hoi polloi, not you), until there is a stable version and want to remove bells and whistles as I go and ask that users make solid cases for requests for new bells and whistles. I'd like your help in assessing those requests, btw -- I don't want BirdNET-Pi to turn into a toy and I want to prioritize research over hobby interests.

Lastly, I'm getting really excited about bats and finally have access to a mic that can listen in those frequencies -- depending on how my tinkering with that goes, I may also want to bring some of those things into the new version. (For now, it's basically a customized version of the custom-recording stuff I put together for you, where it checks for a second sound card, if it is there and is listed with the frequency range > 48000, it should record in the evenings and extract all events as spectrograms -- no species classification, just capturing spectrograms)

ehpersonal38, 2022-08-22:
> The new Laravel platform will let me address bugs and feature requests with much less "hacking" and much better development practices.

Ahoy Patrick! May I offer you the chance to implement my feature into the new repo? It's a cron job that calls `systemctl analyze_recordings start` every 3 minutes, and at the end of that service the service calls `stop` on itself, which is managed by another service that watchdogs when that stop is issued - then a .txt is updated with a '1' meaning the service is ready to be started again, which another cron job reads from (every 2 mins) and then queues the analysis for a `start` again. After this sequence the /etc/crontab is completely wiped and a new job is inserted which uses a new .txt file. Hoping it will make the new repo more efficient while more importantly protecting my job security. Let me know if you're interested. 🙂

mcguirepr89, 2022-08-22:
Sounds like a keeper!

mcguirepr89, 2022-08-22:
## Login:
[image]

## Detections:
[image]

## User profile management (username, password, Google Auth 2FA):
[image]

## Manage Browser/Session Cookies:
[image]

## Set user avatar image:
[image]

## Edit the BirdNET-Analyzer Config File (config.py):
[image]
[image]
[image]
[image]

## API TOKENS!!!!!! Super easy to issue and set permissions:
[image]
[image]
[image]

DD4WH, 2022-08-23:
@mcguirepr89 
Thanks Patrick, looks very good!
only one comment from my side on Bats/ultrasound and BirdNET-Pi:
* that would be a very welcome extension of BirdNET-Pi !
* one of my Bachelor students made a RPi device to automatically record and classify bat ultrasound calls
* the RPi uses a Dodotronic Ultramic microphone (with built-in ADC) and listens for ultrasound: whenever the sound level exceeds the threshold, it records the sound for some seconds, after that the recording is transformed into spectrograms and the spectrograms are classified using a pre-built random forest classifier
* the system relies on two developments: 1.) Clouded Bats WURB Recorder by Arnold Andreasson, and 2.) BatClassify: a random forest pretrained classifier for the bats of Great Britain written in C by Chris Scott
* you can find the source code here, maybe of interest to you and your bat detection developments, which I find very interesting and worthwhile!

https://github.com/FloMee/cloudedbats_wurb_2020
Here is a screenshot of the system from today, it produces a simple realtime graph:
[image]

1.) Clouded bats - the original
https://github.com/cloudedbats/cloudedbats_wurb_2020
2.) Bat Classify - the original
https://bitbucket.org/chrisscott/batclassify/src/master/

closed 2022-09-30 by ehpersonal38