mcguirepr89/BirdNET-Pi #486: Multiple BirdNet-Pi's on the same network

paulhurdubelea, 2022-08-10:
Hello! I tried to connect 3 Raspberry's on the same network. The first one is working properly, but the 2nd and 3rd one don't process the audio recieved. I know that the audio is recieved because the spectogram is displaying the sounds properly.

For the first RPI I kept the same hostname (the default one), but for the other 2 I changed the hostname in order to acces them from the same network, so I think that might be the problem, but I don't know how to solve it. Also in the Tools>System Info tab, 2 processes are off.
[image]

mcguirepr89, 2022-08-10:
Hi, @paulhurdubelea -- thanks for installing BirdNET-Pi.

>but for the other 2 I changed the hostname in order to access them from the same network . . .

Did you do this in [step 8 of the installation instructions](https://github.com/mcguirepr89/BirdNET-Pi/wiki/Installation-Guide#9), or post-install? (Hint, only the former is supported.)

If you did change the hostname during step 8 of the installation guide, could you please provide some log output from the two installations that are not working? To start, just let me know what you see in the "View Logs" page for each installation and we'll take things from there.

My very best regards,
Patrick

paulhurdubelea, 2022-08-10:
Hi @mcguirepr89 and thank you for answering! I changed the hostname during the installation, not post-install. I will attach photos with the log output.

[image]
For raspberry nr. 2

[image]
For raspberry nr. 3

mcguirepr89, 2022-08-10:
:dang:

Those both look like there were issues during the installation for the python virtual environment -- one cannot find the `librosa` module and the other cannot locate `tzlocal`.

On both of those devices, try issuing the following:
- ```
  uninstall.sh && cd ~/BirdNET-Pi && rm -drf ./birdnet && ./scripts/install_birdnet.sh
  ```

That command should uninstall what was successfully installed, remove the python virtual environment completely, and re-initiate the installation.

I'm hopeful that will clear up the issues with the python virtual environment that I suspect are the root of the trouble here.

Please let me know how the installations respond to that command :crossed_fingers:

paulhurdubelea, 2022-08-10:
Thank you for the suggestions and I will let you know if it works.

paulhurdubelea, 2022-08-11:
Ok so I tried using the commands and I got into some pretty nasty studd, other processes stopped working. 

I reinstalled the OS and used the default hostname, proceeded with the installation guide and after all of this I changed the hostname in the terminal. All the issues were solved and the BirdNet-Pi is working properly! 

Thank you for your help!

mcguirepr89, 2022-08-11:
Very glad you're up and running. I will close this issue now.

My very best regards as always,
Patrick

closed 2022-08-11 by mcguirepr89