mcguirepr89/BirdNET-Pi #491: Move web server login/authentication to a page with text fields

Bobblybook, 2022-08-12:
Just wondering if this is possible. I have been moving my passwords to a password manager, and as I plan on exposing this birdnet installation, I want to have the admin pw locked down. Unfortunately due to the way the page prompts for credentials (via the popup browser prompt), it will not interact with password managers, which makes logins tedious.

Moving the username/password fields to a page in the html instead would fix this completely.

ehpersonal38, 2022-08-12:
Okay I've started on this, will work on it sporadically when I get spare time.

Any security concerns or things that need to be considered? @mcguirepr89

ehpersonal38, 2022-08-12:
EDIT: I've done some work on this, try it out by switching to the test branch on your Pi:
`bash ~/BirdNET-Pi/scripts/update_birdnet.sh -b loginform`

Then try to access "Tools." Make sure you're logged out (cookies cleared) or are in incognito.

Let me know how that works for you.

Bobblybook, 2022-08-12:
> EDIT: I've done some work on this, try it out by switching to the test branch on your Pi: `bash ~/BirdNET-Pi/scripts/update_birdnet.sh -b loginform`
> 
> Then try to access "Tools." Make sure you're logged out (cookies cleared) or are in incognito.
> 
> Let me know how that works for you.

Thank you very much. I've just tested this now and it works perfectly.
Also by logging in via the web form, it has fixed the issue I was having with symbols not being accepted in passwords (issue #276). I can now use these passwords with no problem at all.

Currently hosted at birdnet.bobblybook.com. I haven't tested the login form very extensively but there probably isn't much to test - it works and does what it's supposed to!

mcguirepr89, 2022-08-22:
FYI I'm rebuilding BirdNET-Pi on Laravel -- it will have do everything with best practices in mind. The new login page will also allow you to set a "Remember Me" token if desired.