mcguirepr89/BirdNET-Pi #501: Flickr API Key shown in live log

krummrey, 2022-08-17:
Hi, I like to watch the live log to see what birds get cut off below the threshold line. Ever since I've set up my Flickr API key it shows up there as a missing command:

`10:01:27---/etc/birdnet/birdnet.conf: line 74: *********my_API_key*********: command not found`

Doesn't seem to do any damage, but also a little strange to see it there all the time. What is causing it? Can it be fixed?

mcguirepr89, 2022-08-17:
Hi, @krummrey -- I'm sorry you're seeing this issue. I must admit I'm actually not up to date, so I'll need to ask a bit about your `birdnet.conf` to see what is wrong with line 74.

Could you please provide the output from:

```
sed -n '74'p ~/BirdNET-Pi/birdnet.conf
```

Based on the error you've provided, I suspect that line 74 is just your Fickr API key -- if the output from the command above only prints out your Flickr API key, then we've found a bug!! There is not a reason to have a "naked" API key in your `birdnet.conf` -- it needs to be set as a variable (which I suppose would make it "not-naked" if we follow my nonsense verbage).

krummrey, 2022-08-17:
The line is:
`FLICKR_API_KEY= ****************`

You are right, it is not my actual key but something that looks like a hash or an encrypted key
Still strange that it logs it as a "command not found"

UncleSamulus, 2022-08-18:
It may be because of the space between FLICKER_API_KEY and its value.
bash variable need to be declared this way VARIABLE=VALUE with no space around `=` as far as I know.
I hope this could help...

krummrey, 2022-08-18:
YES! That fixed it.

closed 2022-08-18 by krummrey