mcguirepr89/BirdNET-Pi #502: When changing Database Language, database names are not updated

mgoldhacker, 2022-08-18:
Hi and first of all - thanks for the great work!

I recognized that after changing the Database Language BirdnetPi starts to record the detections in the new language (nice!) - but the already existing recordings are not transfered to the new language. This is, now my BirdNET-Pi has kind of duplicates in its database: recordings from old and new language. Is there any possibility to fix this?

Thanks and see you,

Markus

mcguirepr89, 2022-08-18:
Hi, @mgoldhacker -- thanks for installing BirdNET-Pi and for your kind words.

Would you be able to show a screen shot example of a duplicate database entry? I may be able to put together a script that can adjust the old entries for you, but I will want to see an example duplicate for reference beforehand.

If you can, please use the "Tools" > "Database Maintenance" tool to locate duplicate entries and take a screenshot. I'll see what I can come up with from there.

My very best regards,
Patrick

mgoldhacker, 2022-08-18:
Hi Patrick,

thanks for your fast reply. I attach two screenshots from the database maintenance tool.
I think this would be a good fix for everyone, who wants to switch language.

Thanks again for this great BirdNET-Pi implementation!

See you

Marku
[image]
[image]
s

mcguirepr89, 2022-08-18:
@mgoldhacker I'm afraid I don't see any duplicate database entries from either screenshot? Notice each entry has a distinct confidence score? Or perhaps I'm overlooking the duplicates (it is 7:16am here and I am just now getting my first cup of coffee, so totally possible)

mgoldhacker, 2022-08-18:
@mcguirepr89 Sorry, I was not clear enough in my explanation. I see the problem that BirdNET-Pi recognizes the German and English detection as two separate species. This screws up the statistics a bit, since the two actually equivalent species are treated as two separate. If I go into the streamlit Dashboard e.g. there are now English and German versions of the detections mixed
[image]
.

mcguirepr89, 2022-08-18:
@mgoldhacker maybe I'm misunderstanding the ask: if you just want to change all of the "Com_Name" entries you see from "Common Chiffchaff" to "Zilpzalp," try this:

Open your database maintenance tool -- on the left panel, click the "select" button where it says "select detections"

Use the top row of links to select "Search" and choose the "LIKE %%" option. Enter "Chiffchaff," and then click the "Select" action at the far right of that line. 
This will bring up all entries for that species by its common name.
My example below uses "Northern Cardinal"
[image]
at the bottom of the page you can click the check box for "Whole result."

It will bring up a page where you can edit all entries in bulk:
[image]

Save your changes and all of the old names will have the new name

mgoldhacker, 2022-08-18:
@mcguirepr89 Thanks for your reply. Very good that there is a manual option. I think this would be a nice feature to automate this manual renaming task, if the user switches languages. Maybe this would be a nice feature - so I think this is more like a feature request than a bug report :)

See you

Markus

mcguirepr89, 2022-08-18:
I think it would be SUPER tricky to automate this kind of task -- indeed I'm not even sure how I would go about doing that -- it would require looking through the database for any matches against the English labels.txt common name, find the scientific name it corresponds with, locate that species entry in the target language labels.txt file for translation, get the translated common name, then search and replace in the database for that word.

Phew!!

closed 2022-08-18 by mcguirepr89