mcguirepr89/BirdNET-Pi #503: BirdNET-stream: BirdNET-Analyzer based, BirdNET-Pi inspired application for any Linux computer

UncleSamulus, 2022-08-18:
Hello,
As I had no mic for my Raspberry Pi, but a homemade jack microphone and parabola working with an old laptop, I tried to use BirdNET-Pi on a Debian Bullseye x86_64 machine, with some BirdNET-Pi modification. 
Unfortunately, the webpage worked but no sound were recorded.

In order to make this system work, I rewrite the application, with great inspiration from BirdNET-Pi. An now this is working on my system.

Maybe I could help for BirdNET-Pi to be working on any x86_64 computer. I think there is not that much to do to adapt the code.

Bitwise, here is the repository of [BirdNET-stream](https://github.com/UncleSamulus/BirdNET-stream).

BirdNET-stream has a lot less features than BirdNET-Pi yet, but I hope it could help those who don't own a raspberry with microphone, to run BirdNET continuously as well.

It is not mature enough right now, but I am working on It.

mcguirepr89, 2022-08-18:
## Super cool!!

matthew73210, 2022-08-22:
#211 
I've just referenced this other issue as the author of this issue is currently doing work on VM / docker support