mcguirepr89/BirdNET-Pi #516: Feature request - per species thresholds

raybellis, 2022-08-23:
**Is your feature request related to a problem? Please describe.**
I get frequent mis-identifications, specifically:
- Muntjac deer barking reported as Raven
- Hawfinch reports that are likely European Robin, based on correlation of per-hour reports

**Describe the solution you'd like**
An option to specify a per-species recognition threshold would eliminate many of these incorrect reports.

**Describe alternatives you've considered**
An option to submit apparent errors back to BirdNET to improve the training model would help, too.

mcguirepr89, 2022-08-23:
Hello, @raybellis 

Let's get this out of the way: [This is so crazy to me!!!](https://en.wikipedia.org/wiki/Muntjac)

Okay . . . where was I . . . ah, yes . . .

Indeed, false positives are not uncommon. I like your request, though I think the newer model is actually going to be the most elegant solution to this issue. 

It might be possible to put this together without too much trouble, though -- I'll check with @CaiusX tomorrow when he and I meet and we'll see what makes most sense.

For what it is worth right now while you wait :) . . .

I'm putting together a new version of BirdNET-Pi that is built on the new model and its toolset. It has a more granular control over these types of things and would likely be a better place for us to put in the time refactoring :thinking: hard to say!!

I will follow up with you here tomorrow when I have a little more time to think on this and collaborate :)

My very best regards,
Patrick

mcguirepr89, 2022-09-01:
Sorry, @raybellis, for the long delay since my last comment. I discussed this with @CaiusX and I get the feeling we simply don't have enough bandwidth (so to speak) to address this while we're developing around the new model. I'm really hopeful that the new model will address this by virtue of its function and not any additional settings/features.

If we find that the new model does not provide better results across species, we can certainly revisit this solution as a model-agnostic user-definable tweak.

closed 2022-09-30 by ehpersonal38