mcguirepr89/BirdNET-Pi #521: Database indexes

raybellis, 2022-08-26:
Performance of the web interface _ought_ to be improved with the addition of the following two indexes:

```
CREATE INDEX "detections_Com_Name" ON "detections" ("Com_Name")
CREATE INDEX "detections_Date_Time" ON "detections" ("Date" DESC, "Time" DESC)
```
Such that queries by date or species do not have to perform a full table scan.

mcguirepr89, 2022-08-26:
Hi, @raybellis -- thanks for the suggestion!

Would the file, [`createdb.sh`](https://github.com/mcguirepr89/BirdNET-Pi/blob/main/scripts/createdb.sh) be the place to put that?

```
#!/usr/bin/env bash
source /etc/birdnet/birdnet.conf
sqlite3 $HOME/BirdNET-Pi/scripts/birds.db << EOF
DROP TABLE IF EXISTS detections;
CREATE TABLE IF NOT EXISTS detections (
  Date DATE,
  Time TIME,
  Sci_Name VARCHAR(100) NOT NULL,
  Com_Name VARCHAR(100) NOT NULL,
  Confidence FLOAT,
  Lat FLOAT,
  Lon FLOAT,
  Cutoff FLOAT,
  Week INT,
  Sens FLOAT,
  Overlap FLOAT,
  File_Name VARCHAR(100) NOT NULL);
CREATE INDEX "detections_Com_Name" ON "detections" ("Com_Name")
CREATE INDEX "detections_Date_Time" ON "detections" ("Date" DESC, "Time" DESC)
EOF
chown $USER:$USER $HOME/BirdNET-Pi/scripts/birds.db
chmod g+w $HOME/BirdNET-Pi/scripts/birds.db
```

raybellis, 2022-08-31:
Yes, that looks appropriate.

raybellis, 2022-09-06:
albeit maybe missing some trailing semicolons - I'm not _that_ familiar with SQLite.

lloydbayley, 2022-11-11:
Has anyone with else tried this and have any feedback as a result? I added the indexes as @raybellis suggested but can't definitively prove or notice any speed improvement. It stands to reason that it would, however, I can't really replicate a scenario where I go 'WOW, that was fast!'

lloydbayley, 2022-12-30:
Well, that was a long walk down a windy beach to a café that was closed. No takers at all! 
Haven't heard fro the OP either. Ahh well.
If this can be converted into a discussion perhaps or just close the bugger. :)

ehpersonal38, 2023-02-10:
Oh wow, I just tested this and am seeing amazing speed improvements, contrary to what @lloydbayley experienced on his system. It's probably due to how large my birds.db is (90MB, 450,000 detections). But the improvement is something like 500% faster. Thank you so much!!

closed 2023-02-10 by ehpersonal38

lloydbayley, 2023-02-11:
Pffffft! Are you knocking my 26MB, 115,620 database?
To be fair, I did actually implement the changes some time ago, so I suppose it's just not getting as bad as it could have been, so therefore I wouldn't notice!

raybellis, 2023-03-19:
@lloydbayley GitHub seems to have stopped sending me notifications, but I'm glad this does actually seem to have helped those with large databases!

lloydbayley, 2023-03-19:
Well that's just rude!
Glad you found the issue again and have seen that your suggestion has been implemented officially into the project.

Github, sit in the corner and think about what you have done! (Or didn't do)