mcguirepr89/BirdNET-Pi #522: birdnetpi broke today after update

pentode1, 2022-08-30:
After 4+ months and  154,000 detections, my birdnetpi installation broke today after running an update.  I'm thinking maybe something happened and the update broke because there were "missing file" errors in the log and the system was apparently hung up trying to analyze a file that didn't exist.

I touched a couple missing files to see if things would start running again and it seems to be mostly running but I'm getting errors from a python script: 
`3:14:26---ANALYZING AUDIO... Exception in thread Thread-1:
13:14:26---Traceback (most recent call last):
13:14:26---  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
13:14:26---    self.run()
13:14:26---  File "/usr/lib/python3.9/threading.py", line 892, in run
13:14:26---    self._target(*self._args, **self._kwargs)
13:14:26---  File "/usr/local/bin/server.py", line 360, in handle_client
13:14:26---    myReturn += str(i) + '-' + str(detections[i][0]) + '\n'
13:14:26---IndexError: list index out of range`

2 questions:  any ideas what might be going on and assuming things are badly screwed up due to a failed update, is there a simple way to force a reinstall while keeping the birdsong data?

mcguirepr89, 2022-08-31:
Hi, @pentode1 I'm very sorry to hear that attempting an update has caused so much trouble.

The log excerpt you've provided (for which I thank you, by the way) isn't quite enough for me to know what went wrong (disappointing, I know -- believe me, I wish I could always tell the root cause of an issue from a little log except, but unfortunately I'm not that good and I didn't not put BirdNET-Pi together that with that much cleverness!)

Perhaps we can try a manual update to see if we can get a little more info out of that installation:
- Could you please try running an update manually from the command line? You'll SSH into the installation and issue "`update_birdnet.sh`" and provide its output here. (Please redact any information you don't want to share, e.g., your $USER and $HOME path perhaps.)

pentode1, 2022-09-01:
Here's what I get: 

`+ sudo -u xxxx git -C /home/xxxx/BirdNET-Pi reset --hard
HEAD is now at 8956b31 Update README.md
+ set +x
+ sudo -u xxxx git -C /home/xxxx/BirdNET-Pi fetch origin main
From https://github.com/mcguirepr89/BirdNET-Pi
 * branch            main       -> FETCH_HEAD
+ set +x
+ sudo -u xxxx git -C /home/xxxx/BirdNET-Pi switch -C main --track origin/main
Branch 'main' set up to track remote branch 'main' from 'origin'.
Reset branch 'main'
Your branch is up to date with 'origin/main'.
+ set +x

+ my_dir=/home/xxxx/BirdNET-Pi/scripts
+ sudo systemctl stop birdnet_server.service
+ sudo systemctl stop birdnet_recording.service
+ services=(chart_viewer.service spectrogram_viewer.service icecast2.service extraction.service birdnet_recording.service birdnet_log.service)
+ for i in "${services[@]}"
+ sudo systemctl restart chart_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart spectrogram_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart icecast2.service
+ for i in "${services[@]}"
+ sudo systemctl restart extraction.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_recording.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_log.service
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      1148952/python3     
+ logger '[/usr/local/bin/restart_services.sh] birdnet_server.service is running'
+ break
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      1148952/python3     
+ status=0
+ ((  status != 0  ))
+ sudo systemctl restart birdnet_analysis.service`

mcguirepr89, 2022-09-01:
Thanks, @pentode1 -- that update looks successful, so let's keep digging.

Could you please do the following for me, it will help me get a better picture of your system:
(Note: the commands should "scrub" your personal info, but please review their output in case anything remains that you would like to manually obscure before sharing.)

1. Go ahead and manually reboot the system. The goal here is to get a fresh set of journal logs that show a clean booting process and then hopefully will reveal the issues you're encountering during the next step.
2. When the system boots, issue this command and please relay its output here:
   ```
   journalctl --no-hostname --no-pager --boot | sed "s|$HOME||g;s/$USER//g;/LATITUDE/d;/LONGITUDE/d;/KEY/d;/BIRDWEATHER/d;/PWD/d;/STREAM/d;/IDFILE/d"
   ```
3. Finally, it is helpful to check some vitals/resources while we're at it:
   ```
   grep Model /proc/cpuinfo && free -h && df -h
   ```

If we don't find the issue there, the last bit of recourse I can offer is a quick remote session wherein I will take a look and see if I can find the issue. :crossed_fingers: I'll look forward to the output from the steps above and will keep my fingers crossed that we find the issues somewhere therein.

My very best regards as always,
Patrick

pentode1, 2022-09-01:
Thanks very much, I appreciate the help.  Here's what I get:
`Sep 01 10:54:57 caddy[509]: {"level":"info","ts":1662040497.958446,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x40002c6af0"}
Sep 01 10:54:57 caddy[509]: {"level":"info","ts":1662040497.9667287,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}
Sep 01 10:54:57 caddy[509]: {"level":"info","ts":1662040497.9669483,"logger":"tls","msg":"finished cleaning storage units"}
Sep 01 10:54:57 caddy[509]: {"level":"info","ts":1662040497.9747376,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
Sep 01 10:54:57 caddy[509]: {"level":"info","ts":1662040497.9757485,"msg":"serving initial configuration"}
Sep 01 10:54:57 systemd[1]: Started Caddy.
Sep 01 10:54:57 login[541]: pam_unix(login:session): session opened for user (uid=1000) by LOGIN(uid=0)
Sep 01 10:54:58 systemd-logind[397]: New session 2 of user .
Sep 01 10:54:58 systemd[1]: Created slice User Slice of UID 1000.
Sep 01 10:54:58 dhcpcd[472]: wlan0: carrier acquired
Sep 01 10:54:58 dhcpcd[472]: wlan0: connected to Access Point `NorPacEx5'
Sep 01 10:54:58 kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Sep 01 10:54:58 systemd[1]: Starting User Runtime Directory /run/user/1000...
Sep 01 10:54:58 dhcpcd[472]: DUID 00:01:00:01:29:dd:c1:30:b8:27:eb:fe:5f:43
Sep 01 10:54:58 dhcpcd[472]: wlan0: IAID eb:fe:5f:43
Sep 01 10:54:58 dhcpcd[472]: wlan0: adding address fe80::a54b:7242:7d47:e2ce
Sep 01 10:54:58 avahi-daemon[366]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::a54b:7242:7d47:e2ce.
Sep 01 10:54:58 avahi-daemon[366]: New relevant interface wlan0.IPv6 for mDNS.
Sep 01 10:54:58 avahi-daemon[366]: Registering new address record for fe80::a54b:7242:7d47:e2ce on wlan0.*.
Sep 01 10:54:58 systemd[1]: Finished User Runtime Directory /run/user/1000.
Sep 01 10:54:58 systemd[1]: Starting User Manager for UID 1000...
Sep 01 10:54:58 systemd[637]: pam_unix(systemd-user:session): session opened for user (uid=1000) by (uid=0)
Sep 01 10:54:58 dhcpcd[472]: wlan0: soliciting an IPv6 router
Sep 01 10:54:58 dhcpcd[472]: wlan0: rebinding lease of 192.168.3.158
Sep 01 10:54:58 dhcpcd[472]: wlan0: probing address 192.168.3.158/24
Sep 01 10:54:58 systemd[1]: avahi-alias@birdnetpi.local.service: Scheduled restart job, restart counter is at 2.
Sep 01 10:54:58 systemd[1]: Stopped Publish birdnetpi.local as alias for birdnetpi.local via mdns.
Sep 01 10:54:58 systemd[1]: Started Publish birdnetpi.local as alias for birdnetpi.local via mdns.
Sep 01 10:54:58 bash[651]: Bad number of arguments
Sep 01 10:54:58 systemd[1]: avahi-alias@birdnetpi.local.service: Main process exited, code=exited, status=1/FAILURE
Sep 01 10:54:58 systemd[1]: avahi-alias@birdnetpi.local.service: Failed with result 'exit-code'.
Sep 01 10:54:58 birdnet_recording.sh[368]: + grep 5050 /dev/fd/63
Sep 01 10:54:58 birdnet_recording.sh[658]: ++ netstat -tulpn
Sep 01 10:54:58 birdnet_recording.sh[368]: + sleep 1
Sep 01 10:54:59 kernel: Bluetooth: Core ver 2.22
Sep 01 10:54:59 kernel: NET: Registered PF_BLUETOOTH protocol family
Sep 01 10:54:59 kernel: Bluetooth: HCI device and connection manager initialized
Sep 01 10:54:59 kernel: Bluetooth: HCI socket layer initialized
Sep 01 10:54:59 kernel: Bluetooth: L2CAP socket layer initialized
Sep 01 10:54:59 kernel: Bluetooth: SCO socket layer initialized
Sep 01 10:54:59 kernel: Bluetooth: HCI UART driver ver 2.3
Sep 01 10:54:59 kernel: Bluetooth: HCI UART protocol H4 registered
Sep 01 10:54:59 kernel: Bluetooth: HCI UART protocol Three-wire (H5) registered
Sep 01 10:54:59 kernel: Bluetooth: HCI UART protocol Broadcom registered
Sep 01 10:54:59 btuart[451]: bcm43xx_init
Sep 01 10:54:59 btuart[451]: Flash firmware /lib/firmware/brcm/BCM4345C0.hcd
Sep 01 10:54:59 btuart[451]: Set BDADDR UART: b8:27:eb:01:a0:bc
Sep 01 10:54:59 btuart[451]: Set Controller UART speed to 3000000 bit/s
Sep 01 10:54:59 btuart[451]: Device setup complete
Sep 01 10:54:59 systemd[1]: Starting Load/Save RF Kill Switch Status...
Sep 01 10:54:59 systemd[1]: Started Configure Bluetooth Modems connected by UART.
Sep 01 10:54:59 systemd[1]: Started Load/Save RF Kill Switch Status.
Sep 01 10:54:59 systemd[1]: Created slice system-bthelper.slice.
Sep 01 10:54:59 systemd[1]: Reached target Multi-User System.
Sep 01 10:54:59 systemd[1]: Starting Raspberry Pi bluetooth helper...
Sep 01 10:54:59 systemd[1]: Starting Update UTMP about System Runlevel Changes...
Sep 01 10:54:59 systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Sep 01 10:54:59 systemd[1]: Finished Update UTMP about System Runlevel Changes.
Sep 01 10:54:59 systemd[637]: Queued start job for default target Main User Target.
Sep 01 10:54:59 dbus-daemon[374]: [system] Activating via systemd: service name='org.freedesktop.RealtimeKit1' unit='rtkit-daemon.service' requested by ':1.8' (uid=1000 pid=683 comm="/usr/bin/pulseaudio --daemonize=no --log-target=jo")
Sep 01 10:54:59 systemd[637]: Created slice User Application Slice.
Sep 01 10:54:59 bthelper[674]: Raspberry Pi BDADDR already set
Sep 01 10:54:59 systemd[637]: Reached target Bluetooth.
Sep 01 10:54:59 systemd[637]: Reached target Paths.
Sep 01 10:54:59 dbus-daemon[374]: [system] Successfully activated service 'org.freedesktop.RealtimeKit1'
Sep 01 10:54:59 systemd[637]: Reached target Timers.
[... cut ...]
               total        used        free      shared  buff/cache   available
Mem:           909Mi       214Mi       325Mi       3.0Mi       370Mi       634Mi
Swap:           99Mi          0B        99Mi
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G   27G  1.7G  95% /
devtmpfs        326M     0  326M   0% /dev
tmpfs           455M     0  455M   0% /dev/shm
tmpfs           182M  792K  182M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  253M   31M  222M  12% /boot
tmpfs            91M  4.0K   91M   1% /run/user/1000`

pentode1, 2022-09-01:
Quick follow-up...  In looking through that log I noticed a bunch of sound card related errors which weren't there before-  the device I was using somehow disappeared despite no config changes from me.  However, I went back and changed the sound card back to default and it still doesn't work, however possibly slightly different info here?

`valid, non monotonically increasing dts to muxer in stream 0: 870396 >= 836351
Sep 01 11:06:55 livestream.sh[526]: [193B blob data]
Sep 01 11:06:55 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 838658
Sep 01 11:06:55 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 839801
Sep 01 11:06:55 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 840939
Sep 01 11:06:55 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 842102
Sep 01 11:06:55 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 843262
Sep 01 11:06:55 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 844409
Sep 01 11:06:55 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 845555
Sep 01 11:06:55 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 846704
Sep 01 11:06:55 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 847869
Sep 01 11:06:55 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 849033
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 850196
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 851358
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 852517
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 853676
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 854834
Sep 01 11:06:56 birdnet_recording.sh[562]: + grep 5050 /dev/fd/63
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 855985
Sep 01 11:06:56 birdnet_recording.sh[900]: ++ netstat -tulpn
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 857141
Sep 01 11:06:56 birdnet_recording.sh[562]: + sleep 1
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 858298
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 859453
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 860609
Sep 01 11:06:56 livestream.sh[526]: [193B blob data]
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 862915
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 864065
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 865220
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 866375
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 867528
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 868681
Sep 01 11:06:56 livestream.sh[526]: [mp3 @ 0x559b8109a0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 870396 >= 869833
Sep 01 11:06:57 birdnet_recording.sh[562]: + grep 5050 /dev/fd/63
Sep 01 11:06:57 birdnet_recording.sh[912]: ++ netstat -tulpn
Sep 01 11:06:57 birdnet_recording.sh[562]: + sleep 1
Sep 01 11:06:58 birdnet_recording.sh[562]: + grep 5050 /dev/fd/63
Sep 01 11:06:58 birdnet_recording.sh[920]: ++ netstat -tulpn
Sep 01 11:06:58 birdnet_recording.sh[562]: + sleep 1
Sep 01 11:06:59 birdnet_recording.sh[562]: + grep 5050 /dev/fd/63
Sep 01 11:06:59 birdnet_recording.sh[933]: ++ netstat -tulpn
[... cut ...]
Sep 01 11:07:35 pulseaudio[699]: ALSA woke us up to read new data from the device, but there was actually nothing to read.
Sep 01 11:07:35 pulseaudio[699]: Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers.
Sep 01 11:07:35 pulseaudio[699]: We were woken up with POLLIN set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
Sep 01 11:07:36 systemd[1]: spectrogram_viewer.service: Succeeded.
Sep 01 11:07:36 kernel: Transfer to device 4 endpoint 0x6 frame 1402 failed - FIQ reported NYET. Data may have been lost.
`

mcguirepr89, 2022-09-01:
Thanks for those logs. So far, it does seem to be related to your recording hardware, but I can't say that for certain just yet.

Can you please provide the output for:
```
arecord -L
```

pentode1, 2022-09-01:
Actually, I'll have to correct myself yet again-  I was using the dsnoop interface and it's still there.  However, seems that no matter what I use, it still gives errors.  The thing that puzzles me, though, is why it's hung up on this one file from Tuesday that doesn't exist?  Presumably there's a reference to that file in a script somewhere and it can't recover?

`birdnetpi:~/BirdNET-Pi$ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default
    Playback/recording through the PulseAudio sound server
output
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front output / input
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
usbstream:CARD=Device
    USB Audio Device
    USB Stream Output
usbstream:CARD=vc4hdmi
    vc4-hdmi
    USB Stream Output
`

mcguirepr89, 2022-09-01:
>The thing that puzzles me, though, is why it's hung up on this one file from Tuesday that doesn't exist? Presumably there's a reference to that file in a script somewhere and it can't recover?

I think you're referring to these lines, here:
```
Sep 01 11:07:12 spectrogram.sh[1828]: sox FAIL formats: can't open input file `/BirdSongs/August-2022/30-Tuesday/2022-08-30-birdnet-12:59:34.wav': No such file or directory
Sep 01 11:07:12 systemd[1]: spectrogram_viewer.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Sep 01 11:07:12 systemd[1]: spectrogram_viewer.service: Failed with result 'exit-code'.
```

If so, you can disregard those errors. That complaint will occur anytime the main `birdnet_analysis.service` cannot run. (The error is from the `spectrogram` service that creates a spectrogram for each currently analyzed file. While the `birdnet_analysis.service` is failing, there is no spectrogram to create, so the spectrogram service will also fail. This indication shows us that the last file properly processed by `birdnet_analysis.service` was the file just before `August-2022/30-Tuesday/2022-08-30-birdnet-12:59:34.wav`.)

pentode1, 2022-09-01:
after changing the sound car back to default, it seems to have dropped that file it was stuck on... and is now stuck on a file from Wed.  Meanwhile the log (from the log viewer) is back to that python error from my first post.  Index error: list index out of range

mcguirepr89, 2022-09-01:
Let's see some info about the files the recording services is producing. Could you please provide the output for:
```
source /etc/birdnet/birdnet.conf
ls -lR $RECS_DIR | sed "s|$HOME||g"
```

pentode1, 2022-09-01:
Ok, for the first one, there are thousands of lines, so here are the last several hundred:

`/BirdSongs/May-2022:
total 232
drwxrwxr-x 2 sierra sierra  4096 May  5 00:00 04-Wednesday
drwxrwxr-x 2 sierra sierra  4096 May  5 00:00 04-Wednesday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May  6 00:00 05-Thursday
drwxrwxr-x 2 sierra sierra  4096 May  6 00:00 05-Thursday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May  7 00:00 06-Friday
drwxrwxr-x 2 sierra sierra  4096 May  7 00:00 06-Friday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May  8 00:00 07-Saturday
drwxrwxr-x 2 sierra sierra  4096 May  8 00:00 07-Saturday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 12 00:00 11-Wednesday
drwxrwxr-x 2 sierra sierra  4096 May 12 00:00 11-Wednesday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 13 00:00 12-Thursday
drwxrwxr-x 2 sierra sierra  4096 May 13 00:00 12-Thursday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 14 00:00 13-Friday
drwxrwxr-x 2 sierra sierra  4096 May 14 00:01 13-Friday-Analyzed
drwxrwxr-x 2 sierra sierra 61440 May 15 00:00 14-Saturday
drwxrwxr-x 2 sierra sierra 45056 May 15 00:00 14-Saturday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 19 00:00 18-Wednesday
drwxrwxr-x 2 sierra sierra  4096 May 19 00:00 18-Wednesday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 20 00:00 19-Thursday
drwxrwxr-x 2 sierra sierra  4096 May 20 00:00 19-Thursday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 21 00:00 20-Friday
drwxrwxr-x 2 sierra sierra  4096 May 21 00:00 20-Friday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 22 00:00 21-Saturday
drwxrwxr-x 2 sierra sierra  4096 May 22 00:00 21-Saturday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 26 00:00 25-Wednesday
drwxrwxr-x 2 sierra sierra  4096 May 26 00:00 25-Wednesday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 27 00:00 26-Thursday
drwxrwxr-x 2 sierra sierra  4096 May 27 00:00 26-Thursday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 28 00:00 27-Friday
drwxrwxr-x 2 sierra sierra  4096 May 28 00:00 27-Friday-Analyzed
drwxrwxr-x 2 sierra sierra  4096 May 29 00:00 28-Saturday
drwxrwxr-x 2 sierra sierra  4096 May 29 00:00 28-Saturday-Analyzed

/BirdSongs/May-2022/04-Wednesday:
total 0

/BirdSongs/May-2022/04-Wednesday-Analyzed:
total 0

/BirdSongs/May-2022/05-Thursday:
total 0

/BirdSongs/May-2022/05-Thursday-Analyzed:
total 0

/BirdSongs/May-2022/06-Friday:
total 0

/BirdSongs/May-2022/06-Friday-Analyzed:
total 0

/BirdSongs/May-2022/07-Saturday:
total 0

/BirdSongs/May-2022/07-Saturday-Analyzed:
total 0

/BirdSongs/May-2022/11-Wednesday:
total 0

/BirdSongs/May-2022/11-Wednesday-Analyzed:
total 0

/BirdSongs/May-2022/12-Thursday:
total 0

[... 283 lines cut ...]
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:35 2022-09-01-birdnet-11:35:18.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:35 2022-09-01-birdnet-11:35:33.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:36 2022-09-01-birdnet-11:35:48.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:36 2022-09-01-birdnet-11:36:03.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:36 2022-09-01-birdnet-11:36:18.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:36 2022-09-01-birdnet-11:36:33.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:37 2022-09-01-birdnet-11:36:48.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:37 2022-09-01-birdnet-11:37:03.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:37 2022-09-01-birdnet-11:37:18.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:37 2022-09-01-birdnet-11:37:33.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:38 2022-09-01-birdnet-11:37:48.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:38 2022-09-01-birdnet-11:38:03.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:38 2022-09-01-birdnet-11:38:18.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:38 2022-09-01-birdnet-11:38:33.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:39 2022-09-01-birdnet-11:38:48.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:39 2022-09-01-birdnet-11:39:03.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:39 2022-09-01-birdnet-11:39:18.wav
-rw-r--r-- 1 sierra sierra 1440044 Sep  1 11:39 2022-09-01-birdnet-11:39:33.wav
-rw-r--r-- 1 sierra sierra  660044 Sep  1 11:39 2022-09-01-birdnet-11:39:48.wav
`
and

`birdnetpi:~/BirdNET-Pi$ ls -l ~/BirdSongs/August-2022/31-Wednesday/2022-08-31-birdnet-21:34:47.wav | sed "s|$HOME||g"
-rw-r--r-- 1 sierra sierra 1439788 Aug 31 21:35 /BirdSongs/August-2022/31-Wednesday/2022-08-31-birdnet-21:34:47.wav
`

mcguirepr89, 2022-09-01:
The file `~/BirdSongs/August-2022/31-Wednesday/2022-08-31-birdnet-21:34:47.wav` seems slightly smaller than the rest of the audio files. I wonder if it is corrupted? Try deleting it and restarting services:
```
rm -f ~/BirdSongs/August-2022/31-Wednesday/2022-08-31-birdnet-21:34:47.wav && restart_services.sh
```

pentode1, 2022-09-01:
still that python error

pentode1, 2022-09-01:
If I continue to delete files as with the previous command, it'll go to the next file, then hang forever.

pentode1, 2022-09-01:
is there a way to reinstall from scratch and keep the database(s) that have been created?

mcguirepr89, 2022-09-02:
@pentode1 it _should_ be possible to reinstall from scratch, but if you are at all amenable to the idea, it is very valuable to me to find a root cause for this issue and would really like the chance to see what is the matter, if you'd be willing to let me take a peek under the hood, so to speak. A clean re-install (_if_ that fixes things) wouldn't help me ever prevent this from happening to others in the future. If you are willing to let me take a look, please [email me directly](mailto:mcguirepr89@gmail.com) and we will arrange a remote session ~~

Otherwise:
1. Backup your database:
   ```
   mv ~/BirdNET-Pi/birds.db ~
   ```
2. Backup your files:
   ```
   mv ~/BirdSongs/Extracted ~
   ```
3. Uninstall the old:
   ```
   uninstall.sh && rm -drfv ~/BirdNET-Pi && rm -drfv ~/BirdSongs
   ```
4. Run another installation:
   ```
   curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash
   ```
5. Replace the new with the old stuff:
   ```
   mv -f ~/birds.db ~/BirdNET-Pi/birds.db
   mv -f ~/Extracted ~/BirdSongs/Extracted
   ```
6. Reboot for good measure:
   ```
   sudo reboot
   ```

Please let me know which course of action you decide to take.

My very best regards as always,
Patrick

macmat76, 2022-09-02:
Hi, I'm facing similar issues. About 26th Aug my birdnetpi crashed, today I did some investigation:
- disk was full of wav files, I've decided to delete all of them
- /usr/local/bin/livestream.sh script failed to start due to lack of REC_CARD variable 
- I was trying to set some card name in this REC_CARD variable, but none of them is working
-  ffmpeg -sources alsa returns:
```
ffmpeg version 4.3.4-0+deb11u1+rpt1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --extra-version=0+deb11u1+rpt1 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-mmal --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-sand --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
Auto-detected sources for alsa:
  null [Discard all samples (playback) or generate zero samples (capture)]
* default [Playback/recording through the PulseAudio sound server]
  lavrate [Rate Converter Plugin Using Libav/FFmpeg Library]
  samplerate [Rate Converter Plugin Using Samplerate Library]
  speexrate [Rate Converter Plugin Using Speex Resampler]
  jack [JACK Audio Connection Kit]
  oss [Open Sound System]
  pulse [PulseAudio Sound Server]
  upmix [Plugin for channel upmix (4,6,8)]
  vdownmix [Plugin for channel downmix (stereo) with a simple spacialization]
  usbstream:CARD=Headphones [USB Stream Output]
  hw:CARD=Device,DEV=0 [Direct hardware device without any conversions]
  plughw:CARD=Device,DEV=0 [Hardware device with all software conversions]
  sysdefault:CARD=Device [Default Audio Device]
  front:CARD=Device,DEV=0 [Front output / input]
  dsnoop:CARD=Device,DEV=0 [Direct sample snooping device]
  usbstream:CARD=Device [USB Stream Output]
  usbstream:CARD=vc4hdmi0 [USB Stream Output]
  usbstream:CARD=vc4hdmi1 [USB Stream Output]
```
- but I'm not able to define any card name to make ffmpeg works again , for example when I set in /etc/birdnet/birdnet.conf
  ```
  REC_CARD="hw:CARD=Device,DEV=0"
  ```
the output is as below:
```
Sep  2 21:47:44 birdnetpi systemd[1]: Started BirdNET-Pi Live Stream.
Sep  2 21:47:44 birdnetpi birdnet_recording.sh[390]: + grep 5050 /dev/fd/63
Sep  2 21:47:44 birdnetpi birdnet_recording.sh[88851]: ++ netstat -tulpn
Sep  2 21:47:44 birdnetpi birdnet_recording.sh[390]: + sleep 1
Sep  2 21:47:44 birdnetpi livestream.sh[88850]: ffmpeg version 4.3.4-0+deb11u1+rpt1 Copyright (c) 2000-2021 the FFmpeg developers
Sep  2 21:47:44 birdnetpi livestream.sh[88850]:   built with gcc 10 (Debian 10.2.1-6)
Sep  2 21:47:44 birdnetpi livestream.sh[88850]:   configuration: --prefix=/usr --extra-version=0+deb11u1+rpt1 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-mmal --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-sand --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
[... cut ...]
Sep  2 21:47:44 birdnetpi livestream.sh[88850]:   libswscale      5.  7.100 /  5.  7.100
Sep  2 21:47:44 birdnetpi livestream.sh[88850]:   libswresample   3.  7.100 /  3.  7.100
Sep  2 21:47:44 birdnetpi livestream.sh[88850]:   libpostproc    55.  7.100 / 55.  7.100
Sep  2 21:47:44 birdnetpi livestream.sh[88850]: hw:CARD=Device,DEV=0: Protocol not found
Sep  2 21:47:44 birdnetpi livestream.sh[88850]: Did you mean file:hw:CARD=Device,DEV=0?
Sep  2 21:47:44 birdnetpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
```

Any hint appreciated ;)

mcguirepr89, 2022-09-02:
Hi @macmat76 I'm sorry you're also having trouble. I don't know how similar your problem is to @pentode1's since we haven't yet see anything in logs that show they are the same problem. 

With that said, we can certainly look at your issue, but if there are similarities between what you are able to observe in your logs when compared to the logs I've solicited from @pentode1, please include those here as that will get us much farther in our troubleshooting.

The issue you are seeing when trying to specify the `REC_CARD` manually is that the `hw:` version of _any_ ALSA device is unable to be used in another process. I assume this would mean that your `birdnet_recording.service` is using the `hw:` device. This is why I recommend always using `default`, which is the PulseAudio server. Pulse will allow you to share the hardware among several processes.

Can you confirm whether or not your `birdnet_analysis.service` or `birdnet_recording.service` are running properly? I.e., can you please provide some log output from your "View Logs" page _or_ the output from `journalctl --no-hostname -n50 -ubirdnet_analysis.service -ubirdnet_recording | sed "s|$HOME||g"`

macmat76, 2022-09-05:
Hello,
I think I've found what is wrong on my instance. Somehow my file /etc/birdnet/birdnet.conf was containing only a part of required variables, because of missing for example CHANNELS, REC_CARD, LATITUDE the analysis couldn't start properly. After restoring them manually the system is back a bit, however I still can't go to Settings due to 401 error, even if I provide the default credentials.

closed 2022-09-06 by pentode1