mcguirepr89/BirdNET-Pi #532: Anyone know how to restore old birds.db to a new installation?

pentode1, 2022-09-06:
After reinstalling birdnetpi, I would like to reintegrate my old database into a fresh installation but the system doesn't seem to recognize the db file for some reason.  I can open the db with a sqlite viewer on another machine so I'm pretty confident the database is valid and intact, but birdnetpi just doesn't appear to see it.

Any insight into the best way to restore the old db would be much appreciated.

ehpersonal38, 2022-09-06:
You have the right idea - simply moving a `birds.db` backup file to ~/BirdNET-Pi/scripts/ and then rebooting should have everything sorted. I'm somewhat confident the issue you're facing is a permissions one. So maybe try changing the permissions of the file and rebooting: `sudo chmod 777 ~/BirdNET-Pi/scripts/birds.db`

pentode1, 2022-09-06:
Ah, rookie mistake.  I just blindly followed the instructions from my previous issue which had me moving the file into the BirdNET-Pi folder itself.  Thanks very much ehpersonal38!

closed 2022-09-06 by pentode1