mcguirepr89/BirdNET-Pi #533: Analysis stopped overnight

Vertikar, 2022-09-07:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
Web interface working, livestream and spectrogram were fine.
No new birds detected overnight or this morning, and log viewer wasn't showing analysis running.

Suspect listening to livestream or looking at spectrogram may have caused it based on some of the logs.

Could be related to https://github.com/mcguirepr89/BirdNET-Pi/issues/393 but thought I'd open a new issue in case it's different

**To Reproduce**
Leave device running for a period of time
Resolved by restart of services.

**Screenshots**
See logs below

**Additional context**

Logs I've found so far:
<pre>
Sep 07 23:02:46 birdnet-pi birdnet_analysis.sh[1856994]: Files loaded
Sep 07 23:02:46 birdnet-pi birdnet_analysis.sh[1856994]: RECORDING_LENGTH set to 15
Sep 07 23:02:48 birdnet-pi env[1857057]: Line = 2022-09-07-birdnet-23:02:24.wav.csv
Sep 07 23:02:54 birdnet-pi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 45898.
Sep 07 23:02:54 birdnet-pi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
Sep 07 23:02:54 birdnet-pi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
Sep 07 23:02:54 birdnet-pi systemd[1]: spectrogram_viewer.service: Succeeded.
Sep 07 23:02:55 birdnet-pi birdnet_analysis.sh[1856994]: /home/rpi/BirdNET-Pi/birdnet/bin/python3 /home/rpi/BirdNET-Pi/scripts/analyze.py --i /home/rpi/BirdSongs/September-2022/07-Wednesday/2022-09-07-birdnet-23:02:39.wav --o /home/rpi/BirdSongs/September-2022/07-Wednesday/2022-09-07-birdnet-23:02:39.wav.csv --lat >
Sep 07 23:02:55 birdnet-pi python3[1769575]: ANALYZING AUDIO... DONE! Time 1.6 SECONDS
Sep 07 23:02:55 birdnet-pi python3[1769575]: WRITING RESULTS TO /home/rpi/BirdSongs/September-2022/07-Wednesday/2022-09-07-birdnet-23:02:24.wav.csv ... DONE! WROTE 0 RESULTS.
Sep 07 23:02:55 birdnet-pi python3[1769575]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
Sep 07 23:02:55 birdnet-pi python3[1769575]: Date-time: 2022-09-07 23:02:39
Sep 07 23:02:56 birdnet-pi birdnet_analysis.sh[1857156]: 0.0;3.0-('Petroica goodenovii_Red-capped Robin', 0.087012656)
Sep 07 23:02:56 birdnet-pi birdnet_analysis.sh[1857156]: 3.0;6.0-('Malurus leucopterus_White-winged Fairywren', 0.060569905)
Sep 07 23:02:56 birdnet-pi birdnet_analysis.sh[1857156]: 6.0;9.0-('Malurus leucopterus_White-winged Fairywren', 0.10356414)
Sep 07 23:02:56 birdnet-pi birdnet_analysis.sh[1857156]: 9.0;12.0-('Ninox strenua_Powerful Owl', 0.07109845)
Sep 07 23:02:56 birdnet-pi birdnet_analysis.sh[1857156]: 12.0;15.0-('Callocephalon fimbriatum_Gang-gang Cockatoo', 0.069369435)
Sep 07 23:02:56 birdnet-pi systemd[1]: birdnet_analysis.service: Succeeded.
Sep 07 23:02:59 birdnet-pi systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 449.
Sep 07 23:02:59 birdnet-pi systemd[1]: Stopped BirdNET Analysis.
Sep 07 23:02:59 birdnet-pi systemd[1]: Started BirdNET Analysis.
Sep 07 23:03:00 birdnet-pi birdnet_analysis.sh[1857180]: Files loaded
Sep 07 23:03:00 birdnet-pi env[1857216]: Line = 2022-09-07-birdnet-23:02:39.wav.csv
Sep 07 23:03:00 birdnet-pi birdnet_analysis.sh[1857180]: RECORDING_LENGTH set to 15
Sep 07 23:03:01 birdnet-pi CRON[1857257]: pam_unix(cron:session): session opened for user rpi(uid=1000) by (uid=0)
Sep 07 23:03:01 birdnet-pi CRON[1857258]: (rpi) CMD (/usr/local/bin/cleanup.sh >/dev/null 2>&1)
Sep 07 23:03:01 birdnet-pi CRON[1857257]: pam_unix(cron:session): session closed for user rpi
Sep 07 23:03:03 birdnet-pi sudo[1857307]:    caddy : PWD=/home/rpi/BirdSongs/Extracted ; USER=root ; COMMAND=/usr/bin/timedatectl
Sep 07 23:03:03 birdnet-pi sudo[1857307]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=999)
Sep 07 23:03:03 birdnet-pi dbus-daemon[394]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.251' (uid=0 pid=1857309 comm="timedatectl ")
Sep 07 23:03:03 birdnet-pi systemd[1]: Starting Time & Date Service...
Sep 07 23:03:03 birdnet-pi dbus-daemon[394]: [system] Successfully activated service 'org.freedesktop.timedate1'
Sep 07 23:03:03 birdnet-pi systemd[1]: Started Time & Date Service.
Sep 07 23:03:03 birdnet-pi sudo[1857307]: pam_unix(sudo:session): session closed for user root
Sep 07 23:03:04 birdnet-pi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 45899.
Sep 07 23:03:04 birdnet-pi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
Sep 07 23:03:04 birdnet-pi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
Sep 07 23:03:04 birdnet-pi systemd[1]: spectrogram_viewer.service: Succeeded.
Sep 07 23:03:09 birdnet-pi sudo[1857373]:    caddy : PWD=/home/rpi/BirdSongs/Extracted ; USER=root ; COMMAND=/usr/bin/timedatectl set-timezone Australia/Melbourne
Sep 07 23:03:09 birdnet-pi sudo[1857373]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=999)
Sep 07 23:03:09 birdnet-pi sudo[1857373]: pam_unix(sudo:session): session closed for user root
Sep 07 23:03:09 birdnet-pi sudo[1857376]:    caddy : PWD=/home/rpi/BirdSongs/Extracted ; USER=root ; COMMAND=/usr/bin/timedatectl
Sep 07 23:03:09 birdnet-pi sudo[1857376]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=999)
Sep 07 23:03:09 birdnet-pi sudo[1857376]: pam_unix(sudo:session): session closed for user root
Sep 07 23:03:09 birdnet-pi php[1769745]: Restarting Services
Sep 07 23:03:09 birdnet-pi sudo[1857384]:    caddy : PWD=/home/rpi/BirdSongs/Extracted ; USER=root ; COMMAND=/usr/local/bin/restart_services.sh
Sep 07 23:03:09 birdnet-pi sudo[1857384]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=999)
Sep 07 23:03:09 birdnet-pi sudo[1857386]:     root : PWD=/home/rpi/BirdSongs/Extracted ; USER=root ; COMMAND=/usr/bin/systemctl stop birdnet_server.service
Sep 07 23:03:09 birdnet-pi sudo[1857386]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Sep 07 23:03:09 birdnet-pi systemd[1]: Stopping BirdNET Analysis...
[... 211 lines cut ...]
Sep 07 23:03:20 birdnet-pi python3[1857482]: WRITING RESULTS TO /home/rpi/BirdSongs/September-2022/07-Wednesday/2022-09-07-birdnet-23:02:54.wav.csv ... DONE! WROTE 0 RESULTS.
Sep 07 23:03:20 birdnet-pi python3[1857482]: READING AUDIO DATA... DONE! READ 0 CHUNKS.
Sep 07 23:03:20 birdnet-pi python3[1857482]: Date-time: 2022-09-07 23:03:09
Sep 07 23:03:21 birdnet-pi systemd[1]: chart_viewer.service: Succeeded.
Sep 07 23:03:21 birdnet-pi systemd[1]: chart_viewer.service: Consumed 12.196s CPU time.
Sep 07 23:03:30 birdnet-pi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 45901.
Sep 07 23:03:30 birdnet-pi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
Sep 07 23:03:30 birdnet-pi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
Sep 07 23:03:30 birdnet-pi systemd[1]: spectrogram_viewer.service: Succeeded.
Sep 07 23:03:40 birdnet-pi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 45902.
Sep 07 23:03:40 birdnet-pi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
Sep 07 23:03:40 birdnet-pi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
Sep 07 23:03:40 birdnet-pi systemd[1]: spectrogram_viewer.service: Succeeded.
Sep 07 23:03:45 birdnet-pi systemd[1]: systemd-timedated.service: Succeeded.
Sep 07 23:03:50 birdnet-pi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 45903.
Sep 07 23:03:50 birdnet-pi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
Sep 07 23:03:50 birdnet-pi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
Sep 07 23:03:50 birdnet-pi systemd[1]: spectrogram_viewer.service: Succeeded.
Sep 07 23:04:01 birdnet-pi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 45904.
Sep 07 23:04:01 birdnet-pi systemd[1]: Stopped BirdNET-Pi Spectrogram Viewer.
Sep 07 23:04:01 birdnet-pi systemd[1]: Started BirdNET-Pi Spectrogram Viewer.
Sep 07 23:04:01 birdnet-pi systemd[1]: spectrogram_viewer.service: Succeeded.
Sep 07 23:04:11 birdnet-pi systemd[1]: spectrogram_viewer.service: Scheduled restart job, restart counter is at 45905.

</pre>

Vertikar, 2022-09-09:
I've since updated to the latest version and restarted a few times, and haven't seen the issue again. Will report back if it shows up in the next week or so.
I've only been running it for 5 days when this occurred.

mcguirepr89, 2022-09-09:
Hi, @Vertikar -- I'm sorry you have encountered an issue so soon. Thank you for providing logging, it is very helpful. I hope I can be that helpful in return!!

There are a few things in the logs that you'll want to change (go ahead and change your BirdNET-Pi Web Password since it is revealed in the log as well as requesting a new BirdWeather ID as a courtesy to Tim to keep those confidential.)

I'm curious, though, was there some kind of glitch in your `birdnet.conf` file that wiped your BirdWeather ID? Or did you manually remove it at this time:
```
Sep 07 23:03:16 birdnet-pi birdnet_analysis.sh[1857580]: < BIRDWEATHER_ID=xXB5WB79r17M2hXwjytVephe
Sep 07 23:03:16 birdnet-pi birdnet_analysis.sh[1857580]: ---
Sep 07 23:03:16 birdnet-pi birdnet_analysis.sh[1857580]: > BIRDWEATHER_ID=
Sep 07 23:03:16 birdnet-pi birdnet_analysis.sh[1857562]: The birdnet.conf file has changed
```
:thinking:

Please do keep an eye on things and let me know if this happens to you again. If/when it does, I would like to solicit some logging from you to dig deeper, if you don't mind.

My best regards,
Patrick

Vertikar, 2022-09-12:
Thanks Patrick, I'll go ahead and change what you recommended.

> I'm curious, though, was there some kind of glitch in your `birdnet.conf` file that wiped your BirdWeather ID? Or did you manually remove it at this time

I manually removed it at the that time, as I'm this is isn't the final location and am picking up some inaccurate detections overnight, that I didn't want submitted to BirdWeather.

> Please do keep an eye on things and let me know if this happens to you again. If/when it does, I would like to solicit some logging from you to dig deeper, if you don't mind.

Definitely will, happy to leave it in a error condition for longer this time around too.

Vertikar, 2022-09-27:
Haven't seen this happen again, so will close this off for now

closed 2022-09-27 by Vertikar