mcguirepr89/BirdNET-Pi #542: Failing to connect

dbrianhendrix, 2022-09-12:
This is new. I woke up this morning and started seeing these errors and EVERY recording -- still investigating:

07:53:45 birdnet_analysis.sh[3416745]: analyze.py --i /home/birder/BirdSongs/September-2022/11-Sunday/2022-09-11-birdnet-12:42:15.wav --o /home/birder/BirdSongs/September-2022/11-Sunday/2022-09-11-birdnet-12:42:15.wav.csv --lat 35.9870 --lon -87.1451 --week 34 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --birdweather_id "IN_USE"
07:53:45 birdnet_analysis.sh[3437694]: Traceback (most recent call last):
07:53:45 birdnet_analysis.sh[3437694]: File "/usr/local/bin/analyze.py", line 13, in
07:53:45 birdnet_analysis.sh[3437694]: client.connect(ADDR)
07:53:45 birdnet_analysis.sh[3437694]: ConnectionRefusedError: [Errno 111] Connection refused
07:53:45 birdnet_analysis.sh[3416745]: RECORDING_LENGTH set to 15
07:53:45 birdnet_analysis.sh[3416745]: analyze.py --i /home/birder/BirdSongs/September-2022/11-Sunday/2022-09-11-birdnet-12:42:30.wav --o /home/birder/BirdSongs/September-2022/11-Sunday/2022-09-11-birdnet-12:42:30.wav.csv --lat 35.9870 --lon -87.1451 --week 34 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --birdweather_id "IN_USE"
07:53:45 birdnet_analysis.sh[3437700]: Traceback (most recent call last):
07:53:45 birdnet_analysis.sh[3437700]: File "/usr/local/bin/analyze.py", line 13, in
07:53:45 birdnet_analysis.sh[3437700]: client.connect(ADDR)
07:53:45 birdnet_analysis.sh[3437700]: ConnectionRefusedError: [Errno 111] Connection refused
07:53:45 birdnet_analysis.sh[3416745]: RECORDING_LENGTH set to 15
07:53:46 birdnet_analysis.sh[3416745]: analyze.py --i /home/birder/BirdSongs/September-2022/11-Sunday/2022-09-11-birdnet-12:42:45.wav --o /home/birder/BirdSongs/September-2022/11-Sunday/2022-09-11-birdnet-12:42:45.wav.csv --lat 35.9870 --lon -87.1451 --week 34 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --birdweather_id "IN_USE"
07:53:46 birdnet_analysis.sh[3437706]: Traceback (most recent call last):
07:53:46 birdnet_analysis.sh[3437706]: File "/usr/local/bin/analyze.py", line 13, in
07:53:46 birdnet_analysis.sh[3437706]: client.connect(ADDR)
07:53:46 birdnet_analysis.sh[3437706]: ConnectionRefusedError: [Errno 111] Connection refused
07:53:46 birdnet_analysis.sh[3416745]: RECORDING_LENGTH set to 15
07:53:46 birdnet_analysis.sh[3416745]: analyze.py --i /home/birder/BirdSongs/September-2022/11-Sunday/

mcguirepr89, 2022-09-12:
Hi, @dbrianhendrix --

So it seems your loopback interface is not available. Could you please try to ping your loopback interface to see what that response looks like? It should be a quick ICMP response, but if you loopback is unavailable, it probably won't respond. If it does respond, we'll need to figure out why your `server.py` is not able to connect to the loopback socket.

dbrianhendrix, 2022-09-12:
Appears to ping just fine . . .

birdnetpi:~/BirdNET-Pi$ ping 127.0.0.1
PING 127.0.0.1 (127.0.0.1) 56(84) bytes of data.
64 bytes from 127.0.0.1: icmp_seq=1 ttl=64 time=0.070 ms
64 bytes from 127.0.0.1: icmp_seq=2 ttl=64 time=0.076 ms
64 bytes from 127.0.0.1: icmp_seq=3 ttl=64 time=0.077 ms
64 bytes from 127.0.0.1: icmp_seq=4 ttl=64 time=0.106 ms
64 bytes from 127.0.0.1: icmp_seq=5 ttl=64 time=0.138 ms
64 bytes from 127.0.0.1: icmp_seq=6 ttl=64 time=0.098 ms

mcguirepr89, 2022-09-12:
Could you please also try `localhost` to be sure that is resolving to the loopback interface?

dbrianhendrix, 2022-09-12:
birdnetpi:~/BirdNET-Pi$ ping localhost
PING localhost(localhost (::1)) 56 data bytes
64 bytes from localhost (::1): icmp_seq=1 ttl=64 time=0.091 ms
64 bytes from localhost (::1): icmp_seq=2 ttl=64 time=0.136 ms
64 bytes from localhost (::1): icmp_seq=3 ttl=64 time=0.121 ms
64 bytes from localhost (::1): icmp_seq=4 ttl=64 time=0.092 ms
64 bytes from localhost (::1): icmp_seq=5 ttl=64 time=0.151 ms
64 bytes from localhost (::1): icmp_seq=6 ttl=64 time=0.106 ms
64 bytes from localhost (::1): icmp_seq=7 ttl=64 time=0.083 ms
64 bytes from localhost (::1): icmp_seq=8 ttl=64 time=0.090 ms

mcguirepr89, 2022-09-12:
It appears that your localhost is resolving to IPv6 -- could you please show your `/etc/hosts` file?

dbrianhendrix, 2022-09-12:
birdnetpi:~/BirdNET-Pi$ cat /etc/hosts
127.0.0.1       localhost
192.168.1.7     birdnetpi

mcguirepr89, 2022-09-12:
I'm sure you've tried restarting services to see if the socket is locked? Would you mind doing that again and sharing the log output so I might take a look?

dbrianhendrix, 2022-09-12:
I've rebooted which will restart services.

dbrianhendrix, 2022-09-12:
Lot of services -- any specific ones you want me to restart?

mcguirepr89, 2022-09-12:
> Would you mind doing that again and sharing the log output so I might take a look?
> I've rebooted which will restart services.

Thanks, I'll look forward to the feedback.

> Lot of services -- any specific ones you want me to restart?

"Tools" > "Services" > "Restart Core Services"

dbrianhendrix, 2022-09-12:
Well, wonderful. Now my password doesn't work . . . when it rains, it pours.

dbrianhendrix, 2022-09-12:
I log into the pi with account called "birder" for access and for sudo to root. Does the birdnet-pi use another account for services? Let me check.

dbrianhendrix, 2022-09-12:
Looks like everything runs under "birder" -- password for birder is allowing me to login via SSH but not tools in webui.

mcguirepr89, 2022-09-12:
The system has a username and password, then there is basic authentication from the web browser. For you, you chose for the system user to be `birder`, which will give you shell access.

The basic authenticatio credentials are User: `birdnet` and the password is whatever you set in "Advanced Settings"

You cannot use the operating system credentials to pass the web server basic authentication --they are distinct.

dbrianhendrix, 2022-09-12:
If I remember right, I made them have the same password so it'd be easy to remember. And I tried "birdnet" and password still failed. Is there a way to change "birdnet" pw from command line?

mcguirepr89, 2022-09-12:
Yes -- the web server's basic authentication password can be set in `birdnet.conf` as the `CADDY_PWD`, but you have a customized Caddyfile, so you will need to make those changes manually.

To change the basic authentication password, you will need to generate a new, hashed caddy password and replace the old one from your Caddyfile.

To generate a new hashed Caddy password, issue:
```
caddy hash-password
```
That will print your hashed password after you create it. Copy and paste it to replace the hashed password that is currently in your `/etc/caddy/Caddyfile`

dbrianhendrix, 2022-09-12:
I have now tried that procedure 3 times with 3 different passwords without it working:

[1] Edited "/home/birder/BirdNET-Pi/birdnet.conf with plaintext password,
[2] Generated hash with "caddy hash-password",
[3] Copied hash from step [2] above and pasted to "/etc/caddy/Caddyfile" (5 lines), and
[4] Ran "sudo systemctl reload caddy"

Webui still did not accept new password.

dbrianhendrix, 2022-09-12:
Okay, weird. I used a password that was ONLY alphanumeric text and it worked.

mcguirepr89, 2022-09-12:
https://github.com/mcguirepr89/BirdNET-Pi/issues/276

dbrianhendrix, 2022-09-12:
However, I'm still getting connection refused in log.

mcguirepr89, 2022-09-12:
Didn't mean to close the issue, meant to just post that bug to keep things connected--

so, back to the original problem

Could you please tell me any customizations that are also going on for the system as well? I meant to add that to the issue template

dbrianhendrix, 2022-09-12:
None. I haven't customized anything.

dbrianhendrix, 2022-09-12:
I can send a Zoom link if needed!

mcguirepr89, 2022-09-12:
How about an [AnyDesk](https://anydesk.com/en/downloads) session instead?

Email me the desktop it provides and I will be happy to take a look :)

dbrianhendrix, 2022-09-12:
Sent to your gmail account.

mcguirepr89, 2022-09-12:
**Update** -- worked with Brian in remote session and found old bug -- issue resolving the hostname from the socket server -- and update was able to address this :)

closed 2022-09-12 by mcguirepr89

ehpersonal38, 2022-09-12:
Great work!

dbrianhendrix, 2022-10-11:
body{font-family:Helvetica,Arial;font-size:13px}Are we still doing AnyDesk?  D. Brian Hendrixveteran  |  mac guy  |  birder  |  coffee snobTennessee Naturalist, firstSenior Linux Engineer, ***@***.***