mcguirepr89/BirdNET-Pi #544: No access after 'no detections today' issue

tomwhite101, 2022-09-12:
Hi all,
Last week my pi was reporting 'no detections today' error and the log in to the tools section seemed to have failed too, so was unable to run reboot/update from the system controls. Power off and on did not resolve.

Ran reboot and update via SSL as can access/log in pi that way, but now have 'page not found' when I try to access.

Is there anything I can do in SSL to recover? can I recover db from the memory card if not?

Very best,

mcguirepr89, 2022-09-12:
Fixed this with Tom -- disk got full, chatted about bats, ran an update, double-checked cleanup.sh and /etc/crontab

closed 2022-09-12 by mcguirepr89

ehpersonal38, 2022-09-12:
Pretty bumbed I missed that bat convo - maybe next time.