mcguirepr89/BirdNET-Pi #548: Webpage completely inaccessible, can still SSH

tianello, 2022-09-15:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
My local webpage is no longer accessible as of today. Only the top three elements will "load", and only in broken CSS form. This is true on multiple devices and browsers tested. I can still SSH in and control things that way, but no amount of updating/troubleshooting has been successful. I'm still getting a flood of email notifications about birds in my yard though, so everything else seems to be working, just not the webpage.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to my RPi's IP address
2. Watch as webpage fails to load

**Expected behavior**
The webpage should load

**Screenshots**
[image]

**Additional context**
I was attempting two things at the same time today, setting up a public installation, and setting up email notifications via Apprise. I don't know which of the two things had an impact on this issue. I also disabled the RPi's wifi adapter this morning, as it kept defaulting to wifi instead of ethernet, and defeating the whole point of me building it as a PoE setup, and potentially causing an issue with my attempts to set up port forwarding.

mcguirepr89, 2022-09-15:
Hi, @tianello -- first of all, thank you so much for filling out the issue template fully -- it is extremely helpful and seems to have already pointed us to the issue.

I believe the issue is going to be with whatever changes you attempted when reconfiguring the web server (setting up a public installation).

To get started, would you mind terribly please sharing the output from this command:
```
grep URL ~/birdnet.conf
```

I have a suspicion that there may be a trailing slash on the URL, which will keep Caddy from running properly.

If that setting looks fine, however, I will solicite a different set of commands to keep looking at this. For what it is worth, this is definitely just an issue with the Caddyfile and we should be able to fix this with a little more information.

tianello, 2022-09-16:
Hi @mcguirepr89, thanks for the fast response! I tried the command you suggested but it says "No such file or directory".

tianello, 2022-09-16:
Also, while I've got your attention, I had configured the Apprise notifications to email me only when there was an infrequent visitor or for the first visit each day, but my inbox is getting absolutely flooded with notifications, is there a way to disable this via SSH while we troubleshoot the other issue?
[image]

tianello, 2022-09-16:
Hi @mcguirepr89, I just decided to wipe everything and start with a fresh installation, something I probably would have done regardless of this issue now that I've done some tinkering and troubleshooting of various things. I know you've expressed elsewhere that you appreciate when people sit tight while you debug things, but I'll be away from home this weekend and am eager to get this set up for public access! Thanks for the guidance and for the amazing work you've put into this project!

closed 2022-09-27 by mcguirepr89

axelsoundingwild, 2023-02-14:
hi @mcguirepr89 , I have the same issue now but I don't want to reinstall everything as I'm in my 4x4 vehicle and I don't have that bandwidth. I tried to make it public and this is what happened, I see the screenshot tianello had at first.

I run grep: /home/pi/birdnet.conf
No such file or directory

Any help? :)
Thanks
Axel

mcguirepr89, 2023-02-14:
Hi, Axel -- please pardon my typo. The file you should be `grep`ing for is actually `~/BirdNET-Pi/birdnet.conf`

Let me know those results when you can

axelsoundingwild, 2023-02-14:
Nothing is happening, no response. Would you like me to copy the file content here?

axelsoundingwild, 2023-02-14:
sorry my bad, i've run 

grep URL ~/BirdNET-Pi/birdnet.conf

and this is the result

## BIRDNETPI_URL is the URL where the extractions, data-set, and live-stream
BIRDNETPI_URL=http://birdnetdaphne.soundingwild.com
#----------------------------  RTSP Stream URL  -------------------------------#
## HEARTBEAT_URL is a location to ping every time some analysis is done
## no information is sent to the the URL, its a heart beat to show that the
HEARTBEAT_URL=

mcguirepr89, 2023-02-14:
I have taken a look at the server that your URL resolves to and it does not show any open ports:
```
mclaptop➜  ~ sudo nmap --open birdnetdaphne.soundingwild.com 
Starting Nmap 7.80 ( https://nmap.org ) at 2023-02-14 08:53 EST
Note: Host seems down. If it is really up, but blocking our ping probes, try -Pn
Nmap done: 1 IP address (0 hosts up) scanned in 3.35 seconds
```

You'll need to recheck your router settings to be sure the appropriate ports (80 and 443) are open and directed to the BirdNET-Pi.

axelsoundingwild, 2023-02-14:
is this correct? just checking the port forwarding on my router

[image]

mcguirepr89, 2023-02-14:
>is this correct? just checking the port forwarding on my router

That CNAME seems to be properly registered with your DNS, though that does not have any affect on which ports are opened on the router. It also does not confirm anything about whether soundingwild.ddns.net is resolving properly. Can you confirm that the public IP address for the installation is still `31.221.189.72`? That is the IP address that `soundingwild.ddns.net` resolves to, so that means `birdnetdaphne.soundingwild.com` will also resolve to that IP address. Once you confirm that the DNS entries are correct, you'll still need to look into your router settings to see which ports are opened and where on the LAN they point.

axelsoundingwild, 2023-02-14:
ye the public IP is still that one.

I'm looking at the router. I'm running a SIM so that may have some
limitations.

axelsoundingwild, 2023-02-14:
OK, I give up for now. I'm on a spanish SIM card and I believe there is the issue about port forwarding. If anyone has another solution please let me know.

The problem now is that I still can't access the GUI, even tho I have removed the URL. How to solve this?

## BIRDNETPI_URL is the URL where the extractions, data-set, and live-stream
BIRDNETPI_URL=
#----------------------------  RTSP Stream URL  -------------------------------#
## HEARTBEAT_URL is a location to ping every time some analysis is done
## no information is sent to the the URL, its a heart beat to show that the
HEARTBEAT_URL=

axelsoundingwild, 2023-02-15:
I've restarted Caddy with this command and the GUI came back. I had a similar issue last year on a previous install. thanks P!

sed 's/reload/restart/g' ~/BirdNET-Pi/scripts/update_caddyfile.sh | sudo bash