mcguirepr89/BirdNET-Pi #549: Analysis stops without warning an requires a system reboot to restart.

stranda, 2022-09-16:
Analysis stops without warning an requires a system reboot to restart.  

When this happens: 

*   previous recordings are accessible and the website works as expected with the exception that the View Log page is blank and not updating.
*   the currently analyzing spectrogram image is not updated

This may be another version of https://github.com/mcguirepr89/BirdNET-Pi/issues/533, but I am not sure.

Based on the log snippet included below (pulled from the /var/log/daemon.log file this looks like a problem with recording puts some part of the BirdNetPi system in a unstable state.

Finally, I am running this system on the Le Potato device with bullseye Armbian.  It installed flawlessly, though I cannot assess whether this is the source of the issue.

I should add that this is a remarkable effort and *thanks for making it available to the public*

**To Reproduce**
Steps to reproduce the behavior:
1. run BirdnetPi system for some period of time (2-48 hours)

**Expected behavior**
Continued analysis and reporting

**Screenshots**
I don't have relevant screen shots, but see log entries below

**Additional context**
The attached log is not the entire logfile, but includes a period before and during the time when the software stops working as expected.
[problem_log.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/9584804/problem_log.txt)

mcguirepr89, 2022-09-16:
Hi, @stranda -- thanks for filling out the issue template completely -- it is super helpful and gets us off to the right start.

Thank you, also, for creating a new issue, since this is not related to any others that have been opened previously.

Your assessment is spot on:
>Based on the log snippet included below (pulled from the `/var/log/daemon.log` file this looks like a problem with recording puts some part of the BirdNetPi system in a unstable state.

Without recordings, the system has no idea what to do except try again and again to make the recordings. If it fails trying to create them, no other services have data on which to operate and the rest of the system will experience a bit of a domino effect wherein each service comes to a halt (or keeps running without any new data, so is spitting out nonsense)

The errors I'm seeing from the log unfortunately are coming straight from ALSA (from both services that rely on ALSA: `birdnet_recording` and `livestream`), not from BirdNET-Pi. I've done some googling for these log excerpts, but nothing helpful has come up yet:

```
Sep 16 09:04:36 BirdNetPiWSB birdnet_recording.sh[5022]: overrun!!! (at least 60107773.553 ms long)
Sep 16 09:04:36 BirdNetPiWSB birdnet_recording.sh[5022]: arecord: xrun:1650: xrun: prepare error: Input/output error
...
Sep 16 09:04:36 BirdNetPiWSB livestream.sh[4875]: [alsa @ 0xaaab07b9e380] cannot recover from underrun (snd_pcm_prepare failed: Input/output error)
Sep 16 09:04:36 BirdNetPiWSB livestream.sh[4875]: [alsa @ 0xaaab07b9e380] ALSA read error: Broken pipe
```

If I were to bet/guess, however, I would suspect this has something to do with your mic's sample-rate and the sample-rate you're using for recording (which needs to be 48000Hz for the BirdNET-Lite model, so don't go altering that). It is my (cursory and amateur) understanding that these `overrun` and `underrun` issues are related to the sample-rate the sound card uses not corresponding with the sample-rate used for recording, and that ALSA is panicking about what to do with the extra (or lack of) audio data, unable to "recover from underrun."

That is a guess based on some basic googling of the errors in the log -- I would suggest poking around some Le Potato forums as well to see if there is any chance this error might be related to firmware/hardware/custom kernel stuff for that SBC.

May I ask if you have set anything other than "default" as the "REC_CARD" (sound card) setting for the system?

stranda, 2022-09-16:
Thanks for the quick response.  
Ok, I'll poke around on the Le Potato forums. 
And, yep I'm using the 'default' device.  

Incidentally, I'm using a Ugreen USB/Audio card which I thought was a pretty good choice.  I have another no-name brand I could try to see if that makes a difference.  Will update if I learn anything

mcguirepr89, 2022-09-16:
@stranda as mentioned, most of my feedback was guesswork, for which I do apologize -- at the bare minimum I hope it is not misleading!

For what it is worth, the `birdnet_recording` service is _supposed_ to restart under any scenario that makes it stop (except an explicit `systemctl stop` request). (This is configured in the service file for `birdnet_recording` as the `Restart=` line:

```
[Unit]
Description=BirdNET Recording
[Service]
Environment=XDG_RUNTIME_DIR=/run/user/1000
Restart=always
Type=simple
RestartSec=3
User=${USER}
ExecStart=/usr/local/bin/birdnet_recording.sh
[Install]
WantedBy=multi-user.target
``` 
)

Since it seems we were able to spot portions of the log that document ALSA's recording pipe breaking, I would like to see what types of errors it reports when trying to _restart_, which I don't see in the logs. That might give us more information about why the recording service is unable to recover itself as it should. :thinking:

stranda, 2022-09-17:
Thanks for the details, @mcguirepr89 . After some more investigation, I  think the issue operates like this:

A hardware or system issue causes pulseaudio to terminate and BirdNetPi to stop processing.  

BirdNetPi recording processing doesn't start back up.
This happens because Pulseaudio doesn't start back up.  I noticed that after a failure like you identified above, recordings were not processed until I opened a shell on the device. It looks like that interactive shell kicks off another pulseaudio daemon, but somehow the BirdNetPi process running alone cannot.  This behavior seems partially consistent with this description of pulseaudio's deamon from wiki.debian.org

> *Restarting the Pulseaudio Daemon*
> Since Debian 9, Pulseaudio is managed by the per-user instance of [systemd](https://wiki.debian.org/systemd). It uses socket activation, which means Pulseaudio will start automatically when an application tries to use it.
> 

However I don't understand why a routine in BirdNetPi doesn't set off the socket activation.  The quote above suggests it would happen automatically.

I've swapped out the power supply and usb-audio adapter on my setup because I had them and hope that this might resolve the issue by addressing a/the root cause.  If not, I'll keep looking at it, and I'm open to suggestions on how to periodically fire off a pulseaudio daemon in a sensible fashion

Here is the output from journalctl, but I only captured it after I opened a shell and processes started up, so it may not be useful.

===============

-- Journal begins at Fri 2022-09-16 18:57:14 EDT, ends at Fri 2022-09-16 21:55:34 EDT. --
Sep 16 21:55:07 systemd[1]: Started BirdNET Recording.
Sep 16 21:55:07 birdnet_recording.sh[129643]: + source /etc/birdnet/birdnet.conf
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ LATITUDE=32.7606
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ LONGITUDE=-79.9877
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ BIRDWEATHER_ID=
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ CADDY_PWD=
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ ICE_PWD=birdnetpi
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ BIRDNETPI_URL=
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ RTSP_STREAM=
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ APPRISE_NOTIFICATION_TITLE='A WSB detection of: '
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ APPRISE_NOTIFICATION_BODY='A $sciname $comname was just detected with
 a confidence of $confidence'
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ APPRISE_NOTIFY_EACH_DETECTION=0
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ APPRISE_NOTIFY_NEW_SPECIES=1
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ APPRISE_WEEKLY_REPORT=1
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ FLICKR_API_KEY=
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ FLICKR_FILTER_EMAIL=
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ RECS_DIR=/home/astrand/BirdSongs
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ REC_CARD=default
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ PROCESSED=/home/astrand/BirdSongs/Processed
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ EXTRACTED=/home/astrand/BirdSongs/Extracted
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ OVERLAP=0.0
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ CONFIDENCE=0.7
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ SENSITIVITY=1.25
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ CHANNELS=1
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ FULL_DISK=purge
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ PRIVACY_THRESHOLD=0
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ RECORDING_LENGTH=15
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ EXTRACTION_LENGTH=
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ AUDIOFMT=mp3
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ DATABASE_LANG=en
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ SILENCE_UPDATE_INDICATOR=0
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ LAST_RUN=
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ THIS_RUN=
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ IDFILE=/home/astrand/BirdNET-Pi/IdentifiedSoFar.txt
Sep 16 21:55:07 birdnet_recording.sh[129643]: ++ APPRISE_NOTIFY_NEW_SPECIES_EACH_DAY=0
Sep 16 21:55:07 birdnet_recording.sh[129643]: + '[' -z 15 ']'
Sep 16 21:55:07 birdnet_recording.sh[129643]: + '[' '!' -z ']'
Sep 16 21:55:07 birdnet_recording.sh[129643]: + pulseaudio --check
Sep 16 21:55:07 birdnet_recording.sh[129644]: E: [pulseaudio] core-util.c: Failed to create secure directory (/run/use
r/1000/pulse): No such file or directory
Sep 16 21:55:07 birdnet_recording.sh[129643]: + pulseaudio --start
Sep 16 21:55:07 birdnet_recording.sh[129645]: E: [autospawn] core-util.c: Failed to create secure directory (/run/user
/1000/pulse): No such file or directory
Sep 16 21:55:07 birdnet_recording.sh[129645]: W: [autospawn] lock-autospawn.c: Cannot access autospawn lock.
Sep 16 21:55:07 birdnet_recording.sh[129645]: E: [pulseaudio] main.c: Failed to acquire autospawn lock
Sep 16 21:55:07 birdnet_recording.sh[129643]: + pgrep arecord
Sep 16 21:55:07 birdnet_recording.sh[129643]: + grep 5050 /dev/fd/63
Sep 16 21:55:07 birdnet_recording.sh[129648]: ++ netstat -tulpn
Sep 16 21:55:07 birdnet_recording.sh[129649]: tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LIST
[... 37 lines cut ...]
Sep 16 21:55:10 birdnet_recording.sh[129863]: ++ CHANNELS=1
Sep 16 21:55:10 birdnet_recording.sh[129863]: ++ FULL_DISK=purge
[... cut ...]
Sep 16 21:55:10 birdnet_recording.sh[129873]: tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LIST
EN      2487/python3
Sep 16 21:55:10 birdnet_recording.sh[129863]: + '[' -z default ']'
Sep 16 21:55:10 birdnet_recording.sh[129863]: + arecord -f S16_LE -c1 -r48000 -t wav --max-file-time 15 -D default --u
se-strftime /home/astrand/BirdSongs/%B-%Y/%d-%A/%F-birdnet-%H:%M:%S.wav
Sep 16 21:55:10 birdnet_recording.sh[129875]: Recording WAVE '/home/astrand/BirdSongs/%B-%Y/%d-%A/%F-birdnet-%H:%M:%S.
wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

mcguirepr89, 2022-09-17:
Howdy, @stranda -- you've homed in on another very important piece of the puzzle: if the sound card is set to `default`, this is going to be PulseAudio on BirdNET-Pi. You're also spot on that pulseaudio must run in user-space in user runtime directory, which is started on boot via the `/etc/systemd/system/getty@tty1.service.d/autologin.conf` service, starting the shell for the userspace.

You are also correct that when that pulse service stops, everything else is taken to a hault.

The kicker is that the birdnet_recording.sh service _also_ restarts pulseaudio on [line 15](https://github.com/mcguirepr89/BirdNET-Pi/blob/fe25aec571d8409384774d455aff29347c281d0d/scripts/birdnet_recording.sh#L15) -- so this sucks :/

Be sure that the Armbian has the `getty@tty1.service.d/autologin.conf` configured and enabled (it could be a RaspiOSism)

stranda, 2022-09-18:
Hi @mcguirepr89 
Looks like Armbian doesn't have a getty@tty1.service.d/autologin.conf and uses a different approach to perform autologins.  I could probably add the directory and conf file, but I notice that the 'armbian-config' utility provides an option to have auto login.  So I configured through the utility and am going to wait to see what happens.

I've had around 36 hours of uptime now that I updated the usb/audio interface and power supply, so that's good.

thanks much

mcguirepr89, 2022-09-18:
@stranda I'll bet we found the culprit -- I'll keep this open until we feel sure that the autologin is the solution.

At that point, I'll reference this discussion (and possibly ask for any additional input from you if you'd be willing) for tweaks needed for installing on Le Potato and as a note for general Armbian tweaks :+1:

matthew73210, 2022-09-25:
> Hi @mcguirepr89 Looks like Armbian doesn't have a [getty@tty1.service.d](mailto:getty@tty1.service.d)/autologin.conf and uses a different approach to perform autologins. I could probably add the directory and conf file, but I notice that the 'armbian-config' utility provides an option to have auto login. So I configured through the utility and am going to wait to see what happens.
> 
> I've had around 36 hours of uptime now that I updated the usb/audio interface and power supply, so that's good.
> 
> thanks much

Hey @stranda,

Has this sorted the issue?

Cheers

stranda, 2022-09-25:
Hi @matthew73210,  So far, I haven't had a hang (maybe 6 days).  I'm still not saying that I have a bulletproof system, however. 
cheers

matthew73210, 2022-09-26:
> Hi @matthew73210, So far, I haven't had a hang (maybe 6 days). I'm still not saying that I have a bulletproof system, however. cheers

Hello,

I also has issues with the mic and autologin so I just set up an rstp stream. Sometimes the whole system would hang just like you mentioned. After checking system info there was no memory left so the system stopped. After rooting around in the terminal I found that for some reason the analysis would stop, and the folder holding the recordings for analysis would fill up to the brim. Removing the contents of the whole folder sorted it out. 
The initial issue which you have sorted is why I moved to an rstp stream, the end goal being to set it up as a docker app for home assistant. Haven't got around to that yet.

Cheers