mcguirepr89/BirdNET-Pi #555: Error in Species Stats for brand new install, 1 species detected.

Astrobirder, 2022-09-18:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
In a fresh birdnet install, I connected a microphone and did a single playback of Carolina Chickadee and then went to the Species Stats page of my Birdnet-Pi web interface.

There below the "Resample Resolution" in the left most column (set to 15 minutes in the radio buttons), I got this message in a pink box:

RangeError: min(1) is equal/bigger than max(1)

Inside the box to the right of that error was the following text:
at t.checkBoundaries (http://birdnetpi.local/stats/static/js/16.08b2e4d3.chunk.js:1:5346)
at http://birdnetpi.local/stats/static/js/16.08b2e4d3.chunk.js:1:18498
at Array.forEach (<anonymous>)
at t.componentDidMount (http://birdnetpi.local/stats/static/js/16.08b2e4d3.chunk.js:1:18468)
at hu (http://birdnetpi.local/stats/static/js/5.3a0bc4b5.chunk.js:2:1474112)
at ks (http://birdnetpi.local/stats/static/js/5.3a0bc4b5.chunk.js:2:1491047)
at t.unstable_runWithPriority (http://birdnetpi.local/stats/static/js/5.3a0bc4b5.chunk.js:2:1510441)
at Wi (http://birdnetpi.local/stats/static/js/5.3a0bc4b5.chunk.js:2:1432799)
at Ds (http://birdnetpi.local/stats/static/js/5.3a0bc4b5.chunk.js:2:1487782)
at bs (http://birdnetpi.local/stats/static/js/5.3a0bc4b5.chunk.js:2:1483868)

Though this text wasn't readable on the webpage as the box was too small, so I did a copy and paste above.

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh Birdnet-Pi install
2. Playback a single bird species
3. Click on Species Stats
4. See error

**Expected behavior**
No Error :-)  May only be an edge case where this doesn't really matter, but might also occur if there literally is only one species detected on a specific day.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
[image]

mcguirepr89, 2022-09-18:
Hi, @Astrobirder -- thanks for spotting this -- we'll take a closer look and make some adjustments for those one-word-a-day birds :)

CaiusX, 2022-09-18:
Hi. This seems to be an edge case, I'll test & fix with a single reading, but would guess if you had 2 readings it would have worked.

ehpersonal38, 2022-09-18:
Heh, pretty good testament to BirdNET-Pi's stability when recent bug reports have become relatively minor like this. 😄

Astrobirder, 2022-09-18:
Yes this is totally an edge case, and not a high priority.

ehpersonal38, 2022-09-19:
Oh absolutely haha, we always want to make things better. Just after months of quite major bug reports (usually a small typo that breaks literally everything), it's nice to have these!

Astrobirder, 2022-09-19:
@ehpersonal38 Well. I have only played with it for about 5 minutes;-).

CaiusX, 2022-09-19:
Hi @Astrobirder 

Agreed that bugs are worth reporting, this particular one crops up when you only have one day of results, and the date range selector gets into a fitz - could do an error check and say "only one day of results, wait another day and a slider will appear" rather than the error box (the actual output in the charts is correct).

I will include this in the next major push, but wont touch it now.

Best
AP

closed 2022-09-19 by CaiusX

Astrobirder, 2022-09-19:
No problem.  I knew it was an edge / error checking case.

Astrobirder, 2022-09-19:
I also have to quickly comment that I rather like the issue number that I got.  I thought it was rather TIMEly. ;-) ;-).  @DD4WH