mcguirepr89/BirdNET-Pi #566: SD Card Full

reefwing, 2022-09-24:
After a couple of months of use my SD card is full and BirdNET-Pi stops analysing.

What files can I safely delete? I'd like to keep the species history but not necessarily the recordings of the songs.

I'm no Linux expert but I can probably work out a script to clear out these files on a regular basis, unless something like this already exists?

reefwing, 2022-09-24:
I should note that I did see the nuclear option in the scripts file "clear_all_data.sh", but I was hoping to keep the species detected data by date.

lloydbayley, 2022-09-24:
Have you set the behaviour in the Settings -> Advanced -> Full Disk Behaviour? You can set it to delete old recordings.

ehpersonal38, 2022-09-24:
^ do double check that this setting is set to "Purge," but since it's the default I will assume that it already is on your system.

Did you by chance install BirdNET-Pi from the Core Electronics article? That version uses a separate GitHub repo which means your system wouldn't have been getting updates since like April. That could definitely be the culprit here since a ton of changes have taken place and the disk "Purge" service got a code overhaul since then.

reefwing, 2022-09-25:
Thanks for the quick response.

I did see the Core Electronics article but used the instructions from GitHub I think. I installed around Aug 10th using:

`curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash
`
When I go to System Controls it says there are 16 updates. I'll try running those. I don't remember seeing a purge option in settings and for some reason I can't get back to the settings tab.

[image]

reefwing, 2022-09-25:
I attempted to update but I think it failed because the SD card was full. After that I couldn't get to the web page anymore, so I decided to bite the bullet and just reinstall the whole thing. I've checked under Settings -> Advanced -> Full Disk Behaviour and Purge is selected. I'll let you know if the problem reoccurs.

closed 2022-09-25 by reefwing