mcguirepr89/BirdNET-Pi #575: BirdNET-Pi log looks like it is a day behind

trophyguy, 2022-09-25:
This morning (Sunday) I was looking at my log and it's Sunday morning with the correct time listed on the log but looks like everything in my log is being analysed as Saturday?

[image]

I haven't rebooted in a few days. Last time was when the PI went unresponsive. Could this be related to another issue I noticed Saturday also. In the screenshot, the chart at the top of the page shows today's top 10 detections and has a timestamp of 18:16. The most recent detection shown below this has a timestamp of 09:12 for Saturday.

[image]

[image]

I was looking in System Info to see if it shows the current date and time and don't see that in there. Just the last reboot and uptime.

Thanks,
Tony

lloydbayley, 2022-09-30:
Just had a look at your site and it seems that everything has caught up now. Is it still going ok? Was it just a glitch in the matrix? :)

trophyguy, 2022-09-30:
Yes, all is well now. I kept an eye on the log and it took a few hours but it analyzed and processed everything from the previous day and then continued on to the next days things. An odd hiccup but all is well now.

lloydbayley, 2022-09-30:
Glad to hear it. It's happened to me before. It seems your message was overlooked for some time. I usually try to help where I can sooner but there's quite a few issues that I'm not sure why they are still open. Anyway, all is well for you now and that's what matters!

[ Mods: Could someone have a look at this and close some of these stale issues? ]

closed 2022-09-30 by ehpersonal38