mcguirepr89/BirdNET-Pi #579: ffmpeg never stops on rtsp stream

BastLeblanc, 2022-09-28:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**

using rtsp stream (from a reolink camera)
was working well for a day. but this morning the ffmpeg process seems to hang

**To Reproduce**

```
rasp64:~/BirdNET-Pi/scripts$ ps -ef|grep ffmp
pi           805     798  5 13:51 ?        00:03:00 ffmpeg -loglevel 32 -ac 2 -i rtsp://admin:PASSWORDHIDDEN@192.168.1.152:554/h264Preview_01_sub -acodec libmp3lame -b:a 320k -ac 2 -content_type audio/mpeg -f mp3 icecast://source:birdnetpi@localhost:8000/stream -re
pi         42171   42167  0 14:34 ?        00:00:02 ffmpeg -i rtsp://admin:PASSWORDHIDDEN@192.168.1.152:554/h264Preview_01_sub -t 15 -vn -acodec pcm_s16le -ac 2 -ar 48000 file:/home/pi/BirdSongs/StreamData/2022-09-28-birdnet-14:34:58.wav
pi         51001   41408  0 14:44 pts/1    00:00:00 grep ffmp
```

the ffmpeg is hanging there (should be 15sec at the -t 15 suggest)

logs from the recording service
```
$ journalctl -u birdnet_recording.service -f
-- Journal begins at Thu 2022-09-22 04:06:10 BST. --
Sep 28 14:35:01 rasp64 birdnet_recording.sh[42171]:   configuration: --prefix=/usr --extra-version=0+deb11u1+rpt3 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-mmal --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-sand --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
Sep 28 14:35:01 rasp64 birdnet_recording.sh[42171]:   libavutil      56. 51.100 / 56. 51.100
Sep 28 14:35:01 rasp64 birdnet_recording.sh[42171]:   libavcodec     58. 91.100 / 58. 91.100
Sep 28 14:35:01 rasp64 birdnet_recording.sh[42171]:   libavformat    58. 45.100 / 58. 45.100
Sep 28 14:35:01 rasp64 birdnet_recording.sh[42171]:   libavdevice    58. 10.100 / 58. 10.100
Sep 28 14:35:01 rasp64 birdnet_recording.sh[42171]:   libavfilter     7. 85.100 /  7. 85.100
Sep 28 14:35:01 rasp64 birdnet_recording.sh[42171]:   libavresample   4.  0.  0 /  4.  0.  0
Sep 28 14:35:01 rasp64 birdnet_recording.sh[42171]:   libswscale      5.  7.100 /  5.  7.100
Sep 28 14:35:01 rasp64 birdnet_recording.sh[42171]:   libswresample   3.  7.100 /  3.  7.100
Sep 28 14:35:01 rasp64 birdnet_recording.sh[42171]:   libpostproc    55.  7.100 / 55.  7.100

```

**Expected behavior**
The ffmpeg should last 15sec.. 

**Screenshots**

**Additional context**
Add any other context about the problem here.

BastLeblanc, 2022-09-28:
If I try to run the exact same command manually it works
```

$ ffmpeg -i rtsp://admin:XXXX@192.168.1.152:554/h264Preview_01_sub -t 15 -vn -acodec pcm_s16le -ac 2 -ar 48000 file:/home/pi/BirdSongs/StreamData/2022-09-28-birdnet-14:34:58.wav
ffmpeg version 4.3.4-0+deb11u1+rpt3 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --extra-version=0+deb11u1+rpt3 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-mmal --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-sand --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
Input #0, rtsp, from 'rtsp://admin:Delphes77@192.168.1.152:554/h264Preview_01_sub':
  Metadata:
    title           : Session streamed by "preview"
    comment         : h264Preview_01_sub
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 640x352, 30 tbr, 90k tbn, 180k tbc
    Stream #0:1: Audio: aac (LC), 16000 Hz, mono, fltp
Stream mapping:
  Stream #0:1 -> #0:0 (aac (native) -> pcm_s16le (native))
Press [q] to stop, [?] for help
Output #0, wav, to 'file:/home/pi/BirdSongs/StreamData/2022-09-28-birdnet-14:34:58.wav':
  Metadata:
    INAM            : Session streamed by "preview"
    ICMT            : h264Preview_01_sub
    ISFT            : Lavf58.45.100
    Stream #0:0: Audio: pcm_s16le ([1][0][0][0] / 0x0001), 48000 Hz, stereo, s16, 1536 kb/s
    Metadata:
      encoder         : Lavc58.91.100 pcm_s16le
size=    2813kB time=00:00:15.00 bitrate=1536.1kbits/s speed=1.43x
video:0kB audio:2812kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.005000%
```

BastLeblanc, 2022-09-28:
anybody has seen this before? any clue how to debug?

ehpersonal38, 2022-09-28:
Hi there! I'm unable to replicate this issue on my machine, although it may only occur after FFMPEG working for a day or more like you said.

I did some searching for issues related to this and came across a [post on Stackoverflow](https://stackoverflow.com/a/16527559/19115745) which advocates for redirecting all FFMPEG output to /dev/null to prevent it from hanging. 

So if you'd like to give that a try for a few days and see if it works: you'll have to append the following to line 11 in ~/BirdNET-Pi/scripts/birdnet_recording.sh 
`> /dev/null 2>&1 < /dev/null`

So the entire line will look like
`ffmpeg -i  ${i} -t ${RECORDING_LENGTH} -vn -acodec pcm_s16le -ac 2 -ar 48000 file:${RECS_DIR}/StreamData/$(date "+%F")-birdnet-$(date "+%H:%M:%S").wav > /dev/null 2>&1 < /dev/null`

Then just do a restart of services or a reboot, and let me know how it goes.

BastLeblanc, 2022-09-28:
after a reboot it seems to be back in working, I'll change to >/dev/null for the ffmpeg script and will report back ! thanks

BastLeblanc, 2022-09-30:
Haven't seen any more of this issue since adding /dev/null.
But I'm not sure that would be something to merge in the main branch.

I'm closing this issue as solved with the workaround if someone else has the issue

closed 2022-09-30 by BastLeblanc

cgomesu, 2022-12-12:
I can reproduce this issue as well and the suggested redirection seems to fix it.  (ffmpeg's `-nostdin` option also works, btw, as mentioned in the stackoverflow post.)  will take a closer look at it over the weekend.