mcguirepr89/BirdNET-Pi #593: Can't connect , "ERR_CONNECTION_REFUSED"

fredfred41, 2022-10-04:
Hi,

I tried to install Birdnet to my Pi 3 Model B Plus Rev 1.3.

I followed the specific pre-install tutorial for pi 3 model B+, then the regular one.

Installation seems to be completed, but when trying to connect to birdnet via webpage, either birdnet.local (that's the right adress for me) or local ip, I have this chrome error : "ERR_CONNECTION_REFUSED".

I tried from another computer and with firefox, edge and opera.

Is there a way to solve this? What did I done wrong?

Thanks

ehpersonal38, 2022-10-05:
Hi there,

Can you please provide the output of the following command to help diagnose this issue?:
`journalctl --no-hostname --no-pager -n100 -ucaddy`

ehpersonal38, 2022-10-05:
Could you try the steps below and see if they help?

----------------------------------------------------------

Let's please try increasing the systemd timeout for the certificate maintenance:
```
sudo systemctl edit caddy
```
Within the blank space provided at the top of the document, type or paste the following:
```
[Service]
TimeoutSec=300s
```
Then reload and restart the service
```
sudo systemctl daemon-reload && sudo systemctl restart caddy
```

fredfred41, 2022-10-05:
It worked! thanks a lot

I tried what you did tell me, but I had an error : 
```
birdnet:~/BirdNET-Pi$ sudo systemctl daemon-reload && sudo systemctl restart caddy
Failed to reload daemon: Failed to activate service 'org.freedesktop.systemd1': timed out (service_start_timeout=25000ms)
```

So I manually restarted the Pi, the problem is solved. Some errors seems to be still there, but I can access the homepage via chrome or firefox, direct ip or .local adress .

```
birdnet:~/BirdNET-Pi$ journalctl --no-hostname --no-pager -n100 -ucaddy
-- Journal begins at Thu 2022-09-22 05:09:01 CEST, ends at Thu 2022-10-06 00:43:41 CEST. --
Oct 02 21:30:42 caddy[870]: caddy.Version=v2.6.1 h1:EDqo59TyYWhXQnfde93Mmv4FJfYe00dO60zMiEt+pzo=
Oct 02 21:30:42 caddy[870]: runtime.GOOS=linux
Oct 02 21:30:42 caddy[870]: runtime.GOARCH=arm64
Oct 02 21:30:42 caddy[870]: runtime.Compiler=gc
Oct 02 21:30:42 caddy[870]: runtime.NumCPU=4
Oct 02 21:30:42 caddy[870]: runtime.GOMAXPROCS=4
Oct 02 21:30:42 caddy[870]: runtime.Version=go1.19.1
Oct 02 21:30:42 caddy[870]: os.Getwd=/
Oct 02 21:30:42 caddy[870]: LANG=en_GB.UTF-8
Oct 02 21:30:42 caddy[870]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
Oct 02 21:30:42 caddy[870]: NOTIFY_SOCKET=/run/systemd/notify
Oct 02 21:30:42 caddy[870]: HOME=/var/lib/caddy
Oct 02 21:30:42 caddy[870]: LOGNAME=caddy
Oct 02 21:30:42 caddy[870]: USER=caddy
Oct 02 21:30:42 caddy[870]: INVOCATION_ID=e4e3a659d06b467091c3ab4a9aeb9b3b
Oct 02 21:30:42 caddy[870]: JOURNAL_STREAM=8:13216
Oct 02 21:30:42 caddy[870]: {"level":"info","ts":1664739042.8568838,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
Oct 02 21:30:42 caddy[870]: {"level":"warn","ts":1664739042.8662252,"msg":"Caddyfile input is not formatted; run the 'caddy fmt' command to fix inconsistencies","adapter":"caddyfile","file":"/etc/caddy/Caddyfile","line":1}
Oct 02 21:30:42 caddy[870]: {"level":"info","ts":1664739042.9147992,"logger":"admin","msg":"admin endpoint started","address":"localhost:2019","enforce_origin":false,"origins":["//localhost:2019","//[::1]:2019","//127.0.0.1:2019"]}
Oct 02 21:30:42 caddy[870]: {"level":"info","ts":1664739042.918439,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x400026aa80"}
Oct 02 21:30:42 caddy[870]: {"level":"warn","ts":1664739042.9189775,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv0","http_port":80}
Oct 02 21:30:42 caddy[870]: {"level":"info","ts":1664739042.9275308,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}
Oct 02 21:30:42 caddy[870]: {"level":"info","ts":1664739042.9277563,"logger":"tls","msg":"finished cleaning storage units"}
Oct 02 21:30:42 caddy[870]: {"level":"info","ts":1664739042.9280093,"logger":"http.log","msg":"server running","name":"srv0","protocols":["h1","h2","h3"]}
Oct 02 21:30:42 caddy[870]: {"level":"info","ts":1664739042.932032,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
Oct 02 21:30:42 caddy[870]: {"level":"info","ts":1664739042.9328938,"msg":"serving initial configuration"}
Oct 02 21:30:42 systemd[1]: Started Caddy.
-- Boot 2a145c7fcc204f938da26fb4a6cbf988 --
Oct 02 23:46:39 systemd[1]: Starting Caddy...
Oct 04 01:57:53 systemd[1]: caddy.service: start operation timed out. Terminating.
Oct 04 01:57:53 systemd[1]: caddy.service: Failed with result 'timeout'.
Oct 04 01:57:53 systemd[1]: Failed to start Caddy.
Oct 04 02:04:50 systemd[1]: caddy.service: Unit cannot be reloaded because it is inactive.
-- Boot 74300d88299e44e3a26ab3fa69083b5e --
Oct 04 03:04:27 systemd[1]: Starting Caddy...
Oct 05 21:55:06 caddy[832]: caddy.HomeDir=/var/lib/caddy
Oct 05 21:55:06 caddy[832]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
Oct 05 21:55:06 caddy[832]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
Oct 05 21:55:06 caddy[832]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
Oct 05 21:55:06 caddy[832]: caddy.Version=v2.6.1 h1:EDqo59TyYWhXQnfde93Mmv4FJfYe00dO60zMiEt+pzo=
Oct 05 21:55:06 caddy[832]: runtime.GOOS=linux
Oct 05 21:55:06 caddy[832]: runtime.GOARCH=arm64
Oct 05 21:55:07 caddy[832]: runtime.Compiler=gc
Oct 05 21:55:07 caddy[832]: runtime.NumCPU=4
Oct 05 21:55:07 caddy[832]: runtime.GOMAXPROCS=4
Oct 05 21:55:07 caddy[832]: runtime.Version=go1.19.1
Oct 05 21:55:07 caddy[832]: os.Getwd=/
Oct 05 21:55:07 caddy[832]: LANG=en_GB.UTF-8
Oct 05 21:55:07 caddy[832]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
Oct 05 21:55:07 caddy[832]: NOTIFY_SOCKET=/run/systemd/notify
Oct 05 21:55:07 caddy[832]: HOME=/var/lib/caddy
Oct 05 21:55:07 caddy[832]: LOGNAME=caddy
Oct 05 21:55:07 caddy[832]: USER=caddy
Oct 05 21:55:07 caddy[832]: INVOCATION_ID=a731e2d50d2b46e7a5378eb50c6e1a8a
Oct 05 21:55:07 caddy[832]: JOURNAL_STREAM=8:14094
Oct 05 21:55:07 caddy[832]: {"level":"info","ts":1664999707.4159298,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
Oct 05 21:55:07 caddy[832]: {"level":"info","ts":1664999707.4287057,"logger":"admin","msg":"admin endpoint started","address":"localhost:2019","enforce_origin":false,"origins":["//localhost:2019","//[::1]:2019","//127.0.0.1:2019"]}
[... 27 lines cut ...]
Oct 06 00:43:04 caddy[943]: runtime.GOMAXPROCS=4
Oct 06 00:43:04 caddy[943]: runtime.Version=go1.19.1
[... cut ...]
Oct 06 00:43:04 caddy[943]: {"level":"info","ts":1665009784.173161,"logger":"http.log","msg":"server running","name":"srv0","protocols":["h1","h2","h3"]}
Oct 06 00:43:04 caddy[943]: {"level":"info","ts":1665009784.1942272,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
Oct 06 00:43:04 caddy[943]: {"level":"info","ts":1665009784.194694,"msg":"serving initial configuration"}
Oct 06 00:43:04 systemd[1]: Started Caddy.
birdnet:~/BirdNET-Pi$
```

Thanks again, I will have fun, right now !

ehpersonal38, 2022-10-05:
That's great!! I hope you enjoy your BirdNET-Pi :)

@mcguirepr89 I used your code to help solve this issue, so thanks! _(sorry for pinging while you're away, just thought you should get credit)_

closed 2022-10-05 by ehpersonal38

fredfred41, 2022-10-06:
H again, sorry but, maybe because I changed web password, I can't acess the website anymore, same error : ERR_CONNECTION_REFUSED

```
birdnet:~/BirdNET-Pi$ journalctl --no-hostname --no-pager -n100 -ucaddy
-- Journal begins at Thu 2022-09-22 05:09:01 CEST, ends at Thu 2022-10-06 02:47:14 CEST. --
Oct 06 02:23:36 caddy[989]: {"level":"error","ts":1665015816.0730696,"logger":"http.log.error","msg":"EOF","request":{"remote_ip":"192.168.1.3","remote_port":"54078","proto":"HTTP/1.1","method":"GET","host":"192.168.1.117","uri":"/todays_detections.php?ajax_detections=true&display_limit=undefined&hard_limit=5","headers":{"Connection":["keep-alive"],"Referer":["http://192.168.1.117/views.php"],"Accept-Encoding":["gzip, deflate"],"Accept-Language":["fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7"],"Cookie":[],"User-Agent":["Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/106.0.0.0 Safari/537.36"],"Accept":["*/*"]}},"duration":1.729136813,"status":502,"err_id":"saz9kyzhe","err_trace":"reverseproxy.statusError (reverseproxy.go:1271)"}
Oct 06 02:24:34 caddy[989]: {"level":"error","ts":1665015874.2961721,"logger":"http.log.error","msg":"EOF","request":{"remote_ip":"192.168.1.3","remote_port":"54082","proto":"HTTP/1.1","method":"GET","host":"192.168.1.117","uri":"/overview.php?ajax_detections=true&previous_detection_identifier=undefined","headers":{"Connection":["keep-alive"],"User-Agent":["Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/106.0.0.0 Safari/537.36"],"Accept":["*/*"],"Referer":["http://192.168.1.117/views.php"],"Accept-Encoding":["gzip, deflate"],"Accept-Language":["fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7"],"Cookie":[]}},"duration":0.006275637,"status":502,"err_id":"wmh3sj9v4","err_trace":"reverseproxy.statusError (reverseproxy.go:1271)"}
Oct 06 02:24:34 caddy[989]: {"level":"error","ts":1665015874.3149714,"logger":"http.log.error","msg":"EOF","request":{"remote_ip":"192.168.1.3","remote_port":"54076","proto":"HTTP/1.1","method":"GET","host":"192.168.1.117","uri":"/overview.php?fetch_chart_string=true","headers":{"Connection":["keep-alive"],"User-Agent":["Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/106.0.0.0 Safari/537.36"],"Referer":["http://192.168.1.117/views.php"],"Accept-Language":["fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7"],"Accept":["*/*"],"Accept-Encoding":["gzip, deflate"],"Cookie":[]}},"duration":0.007681984,"status":502,"err_id":"qepctfaec","err_trace":"reverseproxy.statusError (reverseproxy.go:1271)"}
Oct 06 02:24:34 caddy[989]: {"level":"error","ts":1665015874.31738,"logger":"http.log.error","msg":"EOF","request":{"remote_ip":"192.168.1.3","remote_port":"54077","proto":"HTTP/1.1","method":"GET","host":"192.168.1.117","uri":"/todays_detections.php?ajax_detections=true&display_limit=undefined&hard_limit=5","headers":{"Accept-Encoding":["gzip, deflate"],"Accept-Language":["fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7"],"Cookie":[],"Referer":["http://192.168.1.117/views.php"],"Accept":["*/*"],"Connection":["keep-alive"],"User-Agent":["Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/106.0.0.0 Safari/537.36"]}},"duration":0.010140822,"status":502,"err_id":"j7v0ztqzd","err_trace":"reverseproxy.statusError (reverseproxy.go:1271)"}
Oct 06 02:24:34 caddy[989]: {"level":"error","ts":1665015874.3189363,"logger":"http.log.error","msg":"EOF","request":{"remote_ip":"192.168.1.3","remote_port":"54082","proto":"HTTP/1.1","method":"GET","host":"192.168.1.117","uri":"/overview.php?ajax_left_chart=true","headers":{"Connection":["keep-alive"],"Referer":["http://192.168.1.117/views.php"],"Accept-Language":["fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7"],"User-Agent":["Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/106.0.0.0 Safari/537.36"],"Accept":["*/*"],"Accept-Encoding":["gzip, deflate"],"Cookie":[]}},"duration":0.012861222,"status":502,"err_id":"2ew4xa3wt","err_trace":"reverseproxy.statusError (reverseproxy.go:1271)"}
Oct 06 02:25:34 caddy[989]: {"level":"error","ts":1665015934.265049,"logger":"http.log.error","msg":"EOF","request":{"remote_ip":"192.168.1.3","remote_port":"54082","proto":"HTTP/1.1","method":"GET","host":"192.168.1.117","uri":"/overview.php?ajax_detections=true&previous_detection_identifier=undefined","headers":{"Referer":["http://192.168.1.117/views.php"],"Accept-Language":["fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7"],"Connection":["keep-alive"],"User-Agent":["Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/106.0.0.0 Safari/537.36"],"Accept":["*/*"],"Accept-Encoding":["gzip, deflate"],"Cookie":[]}},"duration":0.008431823,"status":502,"err_id":"qx7wvtqje","err_trace":"reverseproxy.statusError (reverseproxy.go:1271)"}
Oct 06 02:25:34 caddy[989]: {"level":"error","ts":1665015934.2830908,"logger":"http.log.error","msg":"EOF","request":{"remote_ip":"192.168.1.3","remote_port":"54076","proto":"HTTP/1.1","method":"GET","host":"192.168.1.117","uri":"/overview.php?fetch_chart_string=true","headers":{"Connection":["keep-alive"],"Referer":["http://192.168.1.117/views.php"],"Cookie":[],"User-Agent":["Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/106.0.0.0 Safari/537.36"],"Accept":["*/*"],"Accept-Encoding":["gzip, deflate"],"Accept-Language":["fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7"]}},"duration":0.005474497,"status":502,"err_id":"rmest17qs","err_trace":"reversepr [... cut ...]
Oct 06 02:42:23 caddy[934]: {"level":"info","ts":1665016943.4669607,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
Oct 06 02:42:23 caddy[934]: Error: adapting config using caddyfile: parsing caddyfile tokens for 'basicauth': /etc/caddy/Caddyfile:11 - Error during parsing: username and password cannot be empty or missing
Oct 06 02:42:23 systemd[1]: caddy.service: Main process exited, code=exited, status=1/FAILURE
Oct 06 02:42:23 systemd[1]: caddy.service: Failed with result 'exit-code'.
Oct 06 02:42:23 systemd[1]: Failed to start Caddy.
```

ehpersonal38, 2022-10-06:
Maybe try running `bash ~/BirdNET-Pi/scripts/update_caddyfile.sh`?

fredfred41, 2022-10-06:
I censored my password :

```
birdnet:~/BirdNET-Pi$ bash ~/BirdNET-Pi/scripts/update_caddyfile.sh
+ '[' -d /etc/caddy ']'
+ '[' -f /etc/caddy/Caddyfile ']'
+ cp /etc/caddy/Caddyfile /etc/caddy/Caddyfile.original
cp: cannot create regular file '/etc/caddy/Caddyfile.original': Permission denied
+ '[' -z '...........censored.............' ']'
++ caddy hash-password -plaintext '..........censored............'
Error: unknown shorthand flag: 'p' in -plaintext
+ HASHWORD=
+ cat
/home/birdnet/BirdNET-Pi/scripts/update_caddyfile.sh: line 11: /etc/caddy/Caddyfile: Permission denied
+ sudo caddy fmt --overwrite /etc/caddy/Caddyfile
+ sudo systemctl reload caddy
caddy.service is not active, cannot reload.
```

fredfred41, 2022-10-06:
Do you think I should open a new "issue" or maybe post in "discussions" section? I m' not used to github etiquette...

ehpersonal38, 2022-10-06:
We can discuss this here, that's fine!

Could I please see the output of 
`ls -lh | grep caddy`

fredfred41, 2022-10-07:
```
birdnet:~/BirdNET-Pi$ ls -lh | grep caddy
-rw-r--r-- 1 caddy   caddy    140 Oct  6 01:37 apprise.txt
birdnet:~/BirdNET-Pi$
```

ehpersonal38, 2022-10-07:
Caddy should belong to root. Hopefully these commands will fix it:
```
sudo chown -R root:root /etc/caddy
sudo chmod 755 /etc/caddy
```

fredfred41, 2022-10-07:
```
birdnet:~/BirdNET-Pi$ ls -lh | grep caddy
-rw-r--r-- 1 caddy   caddy    140 Oct  6 01:37 apprise.txt
birdnet:~/BirdNET-Pi$ sudo chown -R root:root /etc/caddy
birdnet:~/BirdNET-Pi$ sudo chmod 755 /etc/caddy
birdnet:~/BirdNET-Pi$ ls -lh | grep caddy
-rw-r--r-- 1 caddy   caddy    140 Oct  6 01:37 apprise.txt
```

I still can't access via web, maybe should I simply reinstall everything...

ehpersonal38, 2022-10-07:
Oops, my `ls` commands were wrong above, meant to `ls -lh /etc/`. Anyways, it doesn't matter, if after the chown commands above and a reboot don't work, then I am out of ideas unfortunately.

Your system seems to have other issues like the systemd one above, so I'd agree that a OS reinstall is probably in your best bets.

fredfred41, 2022-10-07:
ok, is there a command to use  to reinstall birdnet without wiping raspian system and start evrything from the begining?

ehpersonal38, 2022-10-08:
Sure, ubinstall instructions here https://github.com/mcguirepr89/BirdNET-Pi#uninstallation