mcguirepr89/BirdNET-Pi #595: Regular daily freezing of services

cornerkilling, 2022-10-07:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
A clear and concise description of what the bug is.

Analyzing stops after several hours of running. This issue has happened several days in a row. A reboot or restarting all services prompts the analyzer to starting working through the backlog of recordings. 

**To Reproduce**
Steps to reproduce the behavior:
Issue happens after several hours. No specific actions cause it.

**Expected behavior**

**Screenshots**
[image]

**Additional context**
Started recently using Apprise to notify Home Assistant via mqtt. No other config changes have been made recently. No updates are pending.

lloydbayley, 2022-10-07:
I know it's a simple thing and not meaning to be rude at all, but have you checked on your available disk space? It's happened to me before when it's gone awry and used up all the disk space with unprocessed recordings. I didn't have mine set to purge and it would sometimes do this.

cornerkilling, 2022-10-07:
Not a bad idea, I hadn't checked. I have an old 120GB ssd plugged in for storage and didn't think it would fill up any time soon. It looks fine right now.
[image]

jds3d, 2022-10-07:
I have also recently had this issue. But mine isn't lasting several hours,
it just processed a couple more files and then dies. Still looking into
clues.

ehpersonal38, 2022-10-07:
Can you try paging through this command and see what happened the moment the analysis stopped? Maybe an error message?
```
journalctl --no-hostname -ubirdnet_analysis -ubirdnet_server -uextraction -ubirdnet_recording
```

cornerkilling, 2022-10-07:
Here is the output from yesterday when I restarted services and rebooted. 
>
Oct 06 12:16:33 birdnet_analysis.sh[1633319]: RECORDING_LENGTH set to 30
Oct 06 12:16:33 birdnet_analysis.sh[1633319]: /home/pi/BirdNET-Pi/birdnet/bin/p>
Oct 06 12:16:33 python3[1633318]: ANALYZING AUDIO... DONE! Time 3.1 SECONDS
Oct 06 12:16:33 python3[1633318]: WRITING RESULTS TO /home/pi/BirdSongs/October>
Oct 06 12:16:33 python3[1633318]: READING AUDIO DATA... DONE! READ 10 CHUNKS.
Oct 06 12:16:33 python3[1633318]: Date-time: 2022-10-06 12:03:33
Oct 06 12:16:37 birdnet_analysis.sh[1633730]: 0.0;3.0-('Helmitheros vermivorum_>
Oct 06 12:16:37 birdnet_analysis.sh[1633730]: 3.0;6.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:37 birdnet_analysis.sh[1633730]: 6.0;9.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:37 birdnet_analysis.sh[1633730]: 9.0;12.0-('Chaetura pelagica_Chim>
Oct 06 12:16:37 birdnet_analysis.sh[1633730]: 12.0;15.0-('Chaetura pelagica_Chi>
Oct 06 12:16:37 birdnet_analysis.sh[1633730]: 15.0;18.0-('Chaetura pelagica_Chi>
Oct 06 12:16:37 birdnet_analysis.sh[1633730]: 18.0;21.0-('Chaetura pelagica_Chi>
Oct 06 12:16:37 birdnet_analysis.sh[1633730]: 21.0;24.0-('Chaetura pelagica_Chi>
Oct 06 12:16:37 birdnet_analysis.sh[1633730]: 24.0;27.0-('Chaetura pelagica_Chi>
Oct 06 12:16:37 birdnet_analysis.sh[1633730]: 27.0;30.0-('Chaetura pelagica_Chi>
Oct 06 12:16:37 birdnet_analysis.sh[1633319]: RECORDING_LENGTH set to 30
Oct 06 12:16:37 birdnet_analysis.sh[1633319]: /home/pi/BirdNET-Pi/birdnet/bin/p>
Oct 06 12:16:37 python3[1633318]: ANALYZING AUDIO... DONE! Time 3.0 SECONDS
Oct 06 12:16:37 python3[1633318]: WRITING RESULTS TO /home/pi/BirdSongs/October>
Oct 06 12:16:37 python3[1633318]: READING AUDIO DATA... DONE! READ 10 CHUNKS.
Oct 06 12:16:37 python3[1633318]: Date-time: 2022-10-06 12:03:59
Oct 06 12:16:40 birdnet_analysis.sh[1633764]: 0.0;3.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:40 birdnet_analysis.sh[1633764]: 3.0;6.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:40 birdnet_analysis.sh[1633764]: 6.0;9.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:40 birdnet_analysis.sh[1633764]: 9.0;12.0-('Helmitheros vermivorum>
Oct 06 12:16:40 birdnet_analysis.sh[1633764]: 12.0;15.0-('Helmitheros vermivoru>
Oct 06 12:16:40 birdnet_analysis.sh[1633764]: 15.0;18.0-('Chaetura pelagica_Chi>
Oct 06 12:16:40 birdnet_analysis.sh[1633764]: 18.0;21.0-('Chaetura pelagica_Chi>
Oct 06 12:16:40 birdnet_analysis.sh[1633764]: 21.0;24.0-('Chaetura pelagica_Chi>
Oct 06 12:16:40 birdnet_analysis.sh[1633764]: 24.0;27.0-('Chaetura pelagica_Chi>
Oct 06 12:16:40 birdnet_analysis.sh[1633764]: 27.0;30.0-('Chaetura pelagica_Chi>
Oct 06 12:16:40 birdnet_analysis.sh[1633319]: RECORDING_LENGTH set to 30
Oct 06 12:16:40 birdnet_analysis.sh[1633319]: /home/pi/BirdNET-Pi/birdnet/bin/p>
Oct 06 12:16:40 python3[1633318]: ANALYZING AUDIO... DONE! Time 3.1 SECONDS
Oct 06 12:16:40 python3[1633318]: WRITING RESULTS TO /home/pi/BirdSongs/October>
Oct 06 12:16:41 python3[1633318]: READING AUDIO DATA... DONE! READ 10 CHUNKS.
Oct 06 12:16:41 python3[1633318]: Date-time: 2022-10-06 12:04:29
Oct 06 12:16:49 birdnet_analysis.sh[1633801]: 0.0;3.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:49 birdnet_analysis.sh[1633801]: 3.0;6.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:49 birdnet_analysis.sh[1633801]: 6.0;9.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:49 birdnet_analysis.sh[1633801]: 9.0;12.0-('Chaetura pelagica_Chim>
Oct 06 12:16:49 birdnet_analysis.sh[1633801]: 12.0;15.0-('Chaetura pelagica_Chi>
Oct 06 12:16:49 birdnet_analysis.sh[1633801]: 15.0;18.0-('Chaetura pelagica_Chi>
Oct 06 12:16:49 birdnet_analysis.sh[1633801]: 18.0;21.0-('Chaetura pelagica_Chi>
Oct 06 12:16:49 birdnet_analysis.sh[1633801]: 21.0;24.0-('Chaetura pelagica_Chi>
Oct 06 12:16:49 birdnet_analysis.sh[1633801]: 24.0;27.0-('Chaetura pelagica_Chi>
Oct 06 12:16:49 birdnet_analysis.sh[1633801]: 27.0;30.0-('Chaetura pelagica_Chi>
Oct 06 12:16:49 birdnet_analysis.sh[1633319]: RECORDING_LENGTH set to 30
Oct 06 12:16:49 birdnet_analysis.sh[1633319]: /home/pi/BirdNET-Pi/birdnet/bin/p>
Oct 06 12:16:49 python3[1633318]: ANALYZING AUDIO... DONE! Time 3.0 SECONDS
Oct 06 12:16:49 python3[1633318]: WRITING RESULTS TO /home/pi/BirdSongs/October>
Oct 06 12:16:49 python3[1633318]: 2022-10-06;12:04:29;Chaetura pelagica;Chimney>
Oct 06 12:16:49 python3[1633318]: Soundscape POST Response Status -  201
Oct 06 12:16:49 python3[1633318]: { "timestamp": "2022-10-06T12:04:38-04:00","l>
Oct 06 12:16:49 python3[1633318]: Detection POST Response Status -  201
Oct 06 12:16:50 python3[1633318]: READING AUDIO DATA... DONE! READ 10 CHUNKS.
Oct 06 12:16:50 python3[1633318]: Date-time: 2022-10-06 12:04:59
Oct 06 12:16:53 birdnet_analysis.sh[1633867]: 0.0;3.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:53 birdnet_analysis.sh[1633867]: 3.0;6.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:53 birdnet_analysis.sh[1633867]: 6.0;9.0-('Chaetura pelagica_Chimn>
Oct 06 12:16:53 birdnet_analysis.sh[1633867]: 9.0;12.0-('Megaceryle alcyon_Belt>
Oct 06 12:16:53 birdnet_analysis.sh[1633867]: 12.0;15.0-('Helmitheros vermivoru>
Oct 06 12:16:53 birdnet_analysis.sh[1633867]: 15.0;18.0-('Helmitheros vermivoru>
Oct 06 12:16:53 birdnet_analysis.sh[1633867]: 18.0;21.0-('Chaetura pelagica_Chi>
Oct 06 12:16:53 birdnet_analysis.sh[1633867]: 21.0;24.0-('Helmitheros vermivoru>
Oct 06 12:16:53 birdnet_analysis.sh[1633867]: 24.0;27.0-('Chaetura pelagica_Chi>
Oct 06 12:16:53 birdnet_analysis.sh[1633867]: 27.0;30.0-('Chaetura pelagica_Chi>
[... 172 lines cut ...]
[... cut ...]
Oct 06 13:55:51 env[1067]: Line = 2022-10-06-birdnet-12:02:59.wav.csv
Oct 06 13:55:51 env[1067]: Line = 2022-10-06-birdnet-12:03:33.wav.csv
Oct 06 13:55:51 env[1067]: Line = 2022-10-06-birdnet-12:03:59.wav.csv
Oct 06 13:55:51 env[1067]: Line = 2022-10-06-birdnet-12:04:29.wav.csv
Oct 06 13:55:51 env[1067]: Line = 9.0;12.0;Chaetura pelagica;Chimney Swift;0.74>
Oct 06 13:55:51 env[1067]: NEWFILE=Chimney_Swift-74-2022-10-06-birdnet-12:04:29>
Oct 06 13:55:52 env[1067]: Line = 2022-10-06-birdnet-12:04:59.wav.csv
Oct 07 00:00:02 env[238780]: Removing old directories

ehpersonal38, 2022-10-07:
```
Oct 06 13:55:15 birdnet_recording.sh[445]: E: [pulseaudio] core-util.c: Failed >
Oct 06 13:55:15 birdnet_recording.sh[396]: + pulseaudio --start
Oct 06 13:55:15 birdnet_recording.sh[543]: E: [autospawn] core-util.c: Failed t>
Oct 06 13:55:15 birdnet_recording.sh[543]: W: [autospawn] lock-autospawn.c: Can>
Oct 06 13:55:15 birdnet_recording.sh[543]: E: [pulseaudio] main.c: Failed to ac>
```

These lines stand out to me as relevant. Can you paste the full lines though? It cuts off halfway through

cornerkilling, 2022-10-07:
Oct 06 13:55:15 birdnet_recording.sh[445]: E: [pulseaudio] core-util.c: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Oct 06 13:55:15 birdnet_recording.sh[396]: + pulseaudio --start
Oct 06 13:55:15 birdnet_recording.sh[543]: E: [autospawn] core-util.c: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Oct 06 13:55:15 birdnet_recording.sh[543]: W: [autospawn] lock-autospawn.c: Cannot access autospawn lock.
Oct 06 13:55:15 birdnet_recording.sh[543]: E: [pulseaudio] main.c: Failed to acquire autospawn lock

ehpersonal38, 2022-10-07:
Thanks for that - I edited my comment above with something to try

cornerkilling, 2022-10-07:
Thanks, I gave that a try and will let it run for a bit.

matthew73210, 2022-10-09:
@cornerkilling this may help in the futur when debugging but use [pastebin](https://pastebin.com/) when posting long logs.
Out of curiosity, what platform are you using, I may have missed it?

mcguirepr89, 2022-10-11:
@owai1980 Did you follow the installation instructions for the RPi3B? (Did you add swap space and increase the recording length)?

Have you tried the solution yet, or is this just to add to a comment to the issue before trying the solution?

jds3d, 2022-10-11:
I wanted to add that my issue looks fixed (for now) (though it was likely a different issue) as I have an RPi4.  In case this helps anyone else, I had some orphaned folders from an old install that were very large.  I cleared them but the file system was mostly unresponsive.  I couldn't do an ls on one folder with 3000 wav files, around 6GB.  I tried update through the website to get latest version, i tried restarting core services through the website, and i tried a sudo reboot.  None of that helped, and I was going to look into whether my disk was corrupt.  But then I unplugged it and replugged it in 20s later and it started chugging away on the old files.  We'll see how long it goes.

cornerkilling, 2022-10-11:
> ```
> Oct 06 13:55:15 birdnet_recording.sh[445]: E: [pulseaudio] core-util.c: Failed >
> Oct 06 13:55:15 birdnet_recording.sh[396]: + pulseaudio --start
> Oct 06 13:55:15 birdnet_recording.sh[543]: E: [autospawn] core-util.c: Failed t>
> Oct 06 13:55:15 birdnet_recording.sh[543]: W: [autospawn] lock-autospawn.c: Can>
> Oct 06 13:55:15 birdnet_recording.sh[543]: E: [pulseaudio] main.c: Failed to ac>
> ```
> 
> These lines look similar to another issue. Could you try this solution and see if it helps?: [#302 (comment)](https://github.com/mcguirepr89/BirdNET-Pi/issues/302#issuecomment-1172912042)

Sorry about the formatting. I'll keep that in mind for the future. I'm on the Pi4. 
I let everything run over the weekend and did not have any freezes or need to restart. This issue is solved for me. Thanks!

closed 2022-10-11 by cornerkilling