mcguirepr89/BirdNET-Pi #601: Apprise MQTT stops

lloydbayley, 2022-10-10:
Having a weird thing happen the last week or so. I thought it was me as I was redoing some of my home automation system.
For no apparent reason, the MQTT notifications via apprise seem to stop. Can't find any trace of why, everything seems ok.
If I use the test button on the Settings page, the test comes through. It seems I need to reboot the Pi before I can get it working again. Core services restart doesn't seem to work. Anyone else?

mcguirepr89, 2022-10-11:
Hi, Lloyd -- I've seen this and am hoping someone else might echo the issue or you might see it clear up after your HA system is settled so we can have some information to work with, since you're saying nothing has changed

closed 2022-10-11 by mcguirepr89

lloydbayley, 2022-10-11:
I won't trouble you with any more issues.