mcguirepr89/BirdNET-Pi #603: audio files put in the wrong directory, then not analysed

owai1980, 2022-10-11:
Hello,
My Rpi3 for some reasons stops the analyse. this bug is already reported. if i restart it, for example, the next day, it will continue to analyse the 15s wav files that were recorded and saved.

the bug: when a bird is identified in yesterday's recording, the mp3 file is saved in today's directory.

[image]

and i think this makes problems later, and the mp3 file is not possible to be found.

suggested solution, when saving the mp3 file, don't use "today date", but the date of the recording.

mcguirepr89, 2022-10-11:
This is not an accurate understanding of what is happening. The file name of the recording contains the time at which the sounds occurred -- the file modification date show the time at which the recording was analyzed.

owai1980, 2022-10-11:
you are right about the file name (2022-10-08).
you are right about the modification date (2022-10-09). (date of analysis)

BUT the directory is WRONG (2022-10-09). it should be 2022-10-08 because the audio was recorded on 2022-10-08. 

another example story to explain more clearly: if birdnet records a bird during the night, at 23:59, it will be analysed the next day, and saved in the wrong directory. later the webUI cannot find it.

owai1980, 2022-10-11:
ok, so... if you don't believe me, how do you explain my screenshot? 
why are the files named "xxx-2022-10-08-xxx.mp3" saved in the directory "2022-10-09" ??

mcguirepr89, 2022-10-11:
I certainly can't say at this point-- I guess I just don't understand and cannot help -- sorry for being an arrogant dick

>suggested solution, when saving the mp3 file, don't use "today date", but the date of the recording.

My response:
```
DATE="$(echo "${line}" \
      | awk -F- '/birdnet/{print $1"-"$2"-"$3}')"
    if [ ! -z ${DATE} ];then
      OLDFILE="$(echo "${line}" | awk -F. '/birdnet/{print $1"."$2}')" ; continue
    fi

    if [ -z ${DATE} ];then
      DATE=$(date "+%F")
    fi
```

That is the part of the code that sets the date. As you can see, the date is taken from BirdNET.csv results -- if no date can be parsed from the file, then "today's date" is used.

Why your system is not able to parse dates correctly at the end of the night but _can_ during the rest of the day is beyond me -- best of luck!

mcguirepr89, 2022-10-11:
I apologize if the locale setting is the issue -- it is my intent that users for any language can use the software, but I've found that my efforts in having the system properly handle the slight punctuation and date formatting changes for each locale have been mostly in vain, though the BirdNET-Pi locale settings do work well in translating the species names.

mcguirepr89, 2022-10-11:
I've added a little bit to the issue title and am re-opening it in case another user _whose criteria match_ might weigh in or provide additional observations

owai1980, 2022-10-11:
no need to apologize! i'm a programmer too and i know how difficult it can be to work on various environements!!

mcguirepr89, 2022-10-11:
Also, since you're using the RPi3B, I wanted to ensure you followed the [pre-install instructions](https://github.com/mcguirepr89/BirdNET-Pi/wiki/RPi0W2-Installation-Guide) (they work just as well post-install, though running them pre-install tends to speed up the installation). The additional swap space, increasing the recording length to 30 seconds, and ensuring the WiFi network card has no power-saving settings enabled on the RPi3B address many of the hardware's shortcomings. Additionally, there are occasional issues with the Caddy's timeout settings on the less-resourceful hardware options: [see this comment on how to tweak the timeout to add more time for Caddy to complete its certificate maintenance](https://github.com/mcguirepr89/BirdNET-Pi/issues/593#issuecomment-1269047353).