mcguirepr89/BirdNET-Pi #619: Fatal Error after a few days of operation

ejoselow, 2022-10-24:
Hi, I've done several installs of Birdnet-pi on a Pi 3b and after a few days, I get these error messages and it stops recording and storing new sounds. When this happens, I have to wipe the micro sd card and start over. 

Here's a screenshot:

[image]
 
In text, the errors are: 

Fatal error: Uncaught Error: Call to a member function fetchArray() on bool in /home/birder/BirdNET-Pi/scripts/overview.php:47 Stack trace: #0 {main} thrown in /home/birder/BirdNET-Pi/scripts/overview.php on line 47

Fatal error: Uncaught Error: Call to a member function fetchArray() on bool in /home/birder/BirdNET-Pi/scripts/todays_detections.php:28 Stack trace: #0 {main} thrown in /home/birder/BirdNET-Pi/scripts/todays_detections.php on line 28

Thanks!

mcguirepr89, 2022-10-24:
Hi, @ejoselow -- thanks for installing BirdNET-Pi. Sorry you are experiencing what seems to be some really unusual behavior already. Before we get too far, can you confirm for me that your "Full Disk Behavior" is set to "purge"?

ejoselow, 2022-10-24:
Hi Patrick-- I didn't change that setting from the default. The settings
I've changed are:

Time zone
Birdweather ID
Mic channels to 1
Privacy Threshold to 1%
Recording Length to 18
Extraction Length to 6
Sigmoid sensitivity to 1.1

I've got it now running on a new SD and am hoping that helps. If needed for
replication, I still have the image with the error.

--Ethan

mcguirepr89, 2022-10-24:
The troubleshooting method for this type of error will involve keeping a close watch on your used disk space (using the "Tools" > "System Info" tool, or `df -h /` at the commandline) as it approaches 95% full. That is the point at which the system's disk maintenance _ought_ to kick in. Once it approaches 95% full, let me know and we will grab some more information.

If you are comfortable at the commandline and would like to start tests now, you can tinker with the `disk_check.sh` utility/script.

1. First, let's ensure the crontab got installed properly: `cat /etc/crontab`
   Expected output should contain these lines where `$USER` is your regular user:
   ```
   */5 * * * * $USER /usr/local/bin/disk_check.sh >/dev/null 2>&1
   */3 * * * * $USER /usr/local/bin/cleanup.sh >/dev/null 2>&1
   @reboot $USER /usr/local/bin/cleanup.sh >/dev/null 2>&1
   ```
1. Next we'll test the `disk_check.sh` script in a few steps:
   1. Check your current disk usage: `df -h /`
      Example output:
      ```
      Filesystem      Size  Used Avail Use% Mounted on
      /dev/root        100G  68G   32G  68% /
      ```
   1. Using the text editor of your choice, go to line [5](https://github.com/mcguirepr89/BirdNET-Pi/blob/de87bca48cb3fe43cd8cc607f8334ff66d73472d/scripts/disk_check.sh#L5) AND line [32](https://github.com/mcguirepr89/BirdNET-Pi/blob/de87bca48cb3fe43cd8cc607f8334ff66d73472d/scripts/disk_check.sh#L32): `vim ~/BirdNET-Pi/scripts/disk_check.sh`
      ```
      if [ "${used//%}" -ge 95 ]; then
      ```
      and change `95` to one percentage less than your current disk usage (using the example output above):
      ```
      if [ "${used//%}" -ge 67 ]; then
      ```
   1. Run the `disk_check.sh` script manually and ensure it is working: `disk_check.sh`
1. Ensure the crontab is being triggered properly:
   1. Reconfigure the output redirection in `/etc/crontab` where `$USER` is your regular user and `$HOME` is its corresponding home directory; also, trigger it right away: `sudo vim /etc/crontab`
      ```
      #*/5 * * * * $USER /usr/local/bin/disk_check.sh >/dev/null 2>&1
      * * * * * $USER /usr/local/bin/disk_check.sh > $HOME/mycronoutput.log
      ```
   1. `tail` the log: `until tail -f $HOME/mycronoutput.log;do sleep 1;done`
      - **If the log does show up at the next minute, you can check the journal for why cron wasn't able to make the log file**
1. Check the `cleanup.sh` script in case it is misbehaving: `cleanup.sh`

As you see, there are a few "points of failure" where the proper disk maintenance automation might fail, though we'll just have to poke and prod until one of the logs points to the culprit.

Let me know how you'd like to proceed with troubleshooting and I will do my best to help out and get to the bottom of this unusual error.

My best regards,
Patrick

ejoselow, 2022-10-28:
>        2. Run the disk_check.sh script manually and ensure it is working:
>    2. Ensure the crontab is being triggered properly:
>       2. tail the log: until tail -f $HOME/mycronoutput.log;do sleep
>       3. Check the cleanup.sh script in case it is misbehaving: cleanup.sh

Hi-- My comments are inline below. I'm stuck with the text editor steps
because I've never done that before. Can you spell out exactly what to do?
Thanks, --Ethan

> The troubleshooting method for this type of error will involve keeping a
> close watch on your used disk space (using the "Tools" > "System Info"
> tool, or df -h / at the commandline) as it approaches 95% full. That is
> the point at which the system's disk maintenance *ought* to kick in. Once
> it approaches 95% full, let me know and we will grab some more information.
>
> If you are comfortable at the commandline and would like to start tests
> now, you can tinker with the disk_check.sh utility/script.
>
>    1. First, let's ensure the crontab got installed properly: cat
>    /etc/crontab
>    Expected output should contain these lines where $USER is your regular
>    user:
>
>    */5 * * * * $USER /usr/local/bin/disk_check.sh >/dev/null 2>&1
>    */3 * * * * $USER /usr/local/bin/cleanup.sh >/dev/null 2>&1
>    @reboot $USER /usr/local/bin/cleanup.sh >/dev/null 2>&1
>
>
>  The step above checks out. All looks right.

>
>    1. Next we'll test the disk_check.sh script in a few steps:
>       1. Check your current disk usage: df -h /
>       Example output:
>
>       Filesystem      Size  Used Avail Use% Mounted on
>       /dev/root        100G  68G   32G  68% /
>
>
> This looks right. Used disk space is at 47%.

>
>    1.
>       1.
>
>        2. Using the text editor of your choice, go to line 5
>       AND line 32
>       <https://github.com/mcguirepr89/BirdNET-Pi/blob/de87bca48cb3fe43cd8cc607f8334ff66d73472d/scripts/disk_check.sh#L32>:
>       vim ~/BirdNET-Pi/scripts/disk_check.sh
>
>       if [ "${used//%}" -ge 95 ]; then
>
>       and change 95 to one percentage less than your current disk usage
>       (using the example output above):
>
>       if [ "${used//%}" -ge 67 ]; then
>
>
> Wasn't sure what to do here-- never done this sort of thing before. So
far, I understand that I'm supposed to change the script at lines 5 and 32
of the disk_check.sh utility. I now have a text editor app and understand
roughly what they're for. How do I actually edit the script on my SD card?
Do I mount it on my PC, open disk_check.sh, and make the changes you
describe above? Looking at the directory, I don't see "disk_check.sh" as a
possible file to open and modify.

ehpersonal38, 2022-10-29:
To make it easier, you can just use the File Manager to edit the file in Tools -> File Manager. You'll want to navigate to `/home/pi/BirdNET-Pi/scripts/` and then click `disk_check.sh`, then click "Edit".

ejoselow, 2022-10-31:
Hi again, I'd had the pi turned off for a few days, but when I restarted I
ran the disk space usage and saw it was at 95%. It's a 32 gb card, so I
don't see how that's happening, but it seems like an important clue.
Disk_check.sh appears to work. Here's a partial screenshot of the end of
the output. Going to try the last steps you laid out now as well.

[image: image.png]

ejoselow, 2022-10-31:
Hi, cleanup.sh seems to give me normal output (i.e., no errors). On below,
I'm trying to follow along on this, but am not clear. Where's the crontab
file? Can you give the full path, or otherwise explain what to do? Thanks,
--Ethan

- Ensure the crontab is being triggered properly:

   1. Reconfigure the output redirection in /etc/crontab where $USER is
   your regular user and $HOME is its corresponding home directory; also,
   trigger it right away: sudo vim /etc/crontab

   #*/5 * * * * $USER /usr/local/bin/disk_check.sh >/dev/null 2>&1
   * * * * * $USER /usr/local/bin/disk_check.sh > $HOME/mycronoutput.log

   2. tail the log: until tail -f $HOME/mycronoutput.log;do sleep 1;done
      - *If the log does show up at the next minute, you can check the
      journal for why cron wasn't able to make the log file*

- Check the cleanup.sh script in case it is misbehaving: cleanup.sh

DJM0, 2022-12-02:
Don't want to open a new issue, but think I'm having a similar issue on an install that's been working fine since August.

I do regularly keep it up to date with the repo, so guessing something has been introduced in the last month or so.

Crons are running fine, but cleanup doesn't seem to be touching most of the recordings last month:

```
birdpi:~/BirdNET-Pi/scripts$ du -sh /home/pi/BirdSongs/November-2022
13G	/home/pi/BirdSongs/November-2022
```

After a few days it's just filling up again, don't really want to reinstall and loose months of detections and recordings. 77,000 according to the DB.

closed 2022-12-13 by mcguirepr89

DJM0, 2022-12-13:
@mcguirepr89 what was the fix? Worried this will bite again and will loose data

mcguirepr89, 2022-12-15:
Hi, @DJM0 -- no fix in place for this because we never saw the root cause. I closed this because the troubleshooting was going nowhere (being "comfortable at the commandline" looks much different than what we see above) and could not have been fruitful without lots of unnecessary work and broken-record repetition.

You are welcome to start a new issue, but please thoroughly provide as much information as possible when creating the issue and do not overlook any solicited information needed to help.

DJM0, 2022-12-15:
I didn't open a new issue as this one seemed very similar, I know how frustrating it can be to have multiple issues open for the same thing. Happy to help troubleshoot but hasn't occurred again since manually deleting backlog of unprocessed recording. No errors in logs, all processes were running fine and I could run the cleanup scripts without error, they just weren't doing anything. And as it just happened days after updating after months of no issues and nothing else on the Pi I assumed it was a bad commit. Will open an issue if I run into it again.

dwd1, 2024-01-13:
Database is corrupted
Solution:
sqlite3 birds.db ".recover" | sqlite3 birds2.db; mv birds2.db birds.db