mcguirepr89/BirdNET-Pi #622: Set web GUI password.  Caddy failed to start on reboot

gerdesj, 2022-10-25:
I installed BNPi on a clean new RPi OS on a Pi 3B.  I used the curl|bash install.

I set a password via the web GUI and it worked during the duration of that boot time.  After a reboot, caddy did not start and I got this error in the journal:

`Oct 25 21:42:23 birdnetpi caddy[815]: Error: adapting config using caddyfile: parsing caddyfile tokens for 'basicauth': /etc/caddy/Caddyfile:11 - Error during parsing: username and password cannot be empty or missing
`
I used _caddy hash-password_ to generate a hash and entered it into _/etc/caddy/Caddyfile_ against each basicauth stanza after the word "birdnet".  _systemctl start caddy_ then started the caddy web server OK.

I'll dig a bit further - I did use the latest RaspiOS which I did see somewhere as untested.  I'm posting this bug in case someone else hits this snag.

closed 2022-10-25 by mcguirepr89

mcguirepr89, 2022-10-25:
If this is a bug, it is certainly a duplicate of #276

gerdesj, 2022-10-25:
The password I set only used alpha-num so no special chars, which #276 specifically mentions.