mcguirepr89/BirdNET-Pi #623: Streamdata filling up

matthew73210, 2022-10-26:
**Describe the bug**
This may have already been posted as an issue.
On random occasions the StreamData folder will fill up with recordings. No analysis happens on these files and the whole system stops. Today 12 gb of data. Has happened on random times throughout the past weeks. In systeminfo only arecord is down (using rtsp). The only way I know that its happened is when I take a peek at the live log and nothing happens.

Is there anyway to kickstart the analysis script to churn out data that has not been analysed? The system that i'm on shouldn't have an issue.

**To Reproduce**
Unknown

**Expected behavior**
After analysis the file should be purged.

**Screenshots**
[image]

**Additional context**
On orange pi using rtsp.

kinteki, 2022-11-14:
Same problem here.
I have to do this manually every day:

rm /home/YOURFOLDER/BirdSongs/StreamData/*

closed 2022-12-13 by mcguirepr89

matthew73210, 2022-12-14:
@mcguirepr89 may I inquire as to why this was closed? It's still an ongoing issue that I have to resolve every week or so.

mcguirepr89, 2022-12-14:
It's closed as both a duplicate and "wont fix" because I won't be implementing a fix for it -- it is still drastically under-documented and on non-standard hardware. Troubleshooting this would require that you have another system that can test the RTSP connection in tandem with `ffmpeg`'s `debug` `loglevel`, monitor your BirdNET-Pi system logs, go through them when we see processing has stopped, test each file to be sure that the RTSP connection was writing usable data, then what? Disable the analysis if there are more than 30 files? What happens at that point? Would you care to script the solution? Have you tried anything other than a drastic wipe of your SD card? You say with confidence that "The system that i'm on shouldn't have an issue," which is a dubious thing to say when you're not using a Raspberry Pi (the basis of this repo). If you would like my help, please start on some of the steps above and provide information that I can use and we'll see if we can get to the root of the issue. Otherwise, it is non-standard and should not stay open indefinitely.

matthew73210, 2022-12-15:
> It's closed as both a duplicate and "wont fix" because I won't be implementing a fix for it -- it is still drastically under-documented and on non-standard hardware. Troubleshooting this would [...]

Thats fine be me. It was more so that if future people have the same issue they'll know that as of now there's no known fix for this setup.

kinteki, 2023-01-05:
Hi,

RTSP connection is OK, tested.

Birdsong catch was yesterday, too.

Everything works well until the disk fills up. It seems the Full Disk 
Behavior:PurgeKeep   purge not works well.

When I do this manually:

/rm /home/zoli/BirdSongs/StreamData/*/

and restarting all services, the system works well. Reboot not needed.

What can I do? Can u help me?

2022. 12. 15. 14:45 keltezéssel, Matthew Burton írta:

> It's closed as both a duplicate and "wont fix" because I won't be
>     implementing a fix for it -- it is still drastically
>     under-documented and on non-standard hardware. Troubleshooting
>     this [...]

matthew73210, 2023-01-08:
Maybe a cron to remove data from said file and restart at midnight? The issue here is that it'll be difficult to find the cause as it's so random.

mcguirepr89, 2023-01-08:
I think I finally found the root cause, which I believe to be a result of a race among the threads. Disabling threading, (according to this working theory), would prevent this from happening going forward. 

I haven't tested that theory yet, but I strongly suspect that has been what is behind all of these impossible-to-document issues of disk space filling up. 

Essentially, what is happening is that there are 2 threads for the analysis. 
- Thread A opens file 1
- While it is open, Thread 2 opens the same file
- Thread A finishes analysis, so closes the file
- Thread B hangs indefinitely because it is a thread whose processing file has closed

There is no way to gracefully exit a hung thread -- and it doesn't provide any tracebacks -- and the system doesn't know there is a failure, so it just keeps recording. All the while, the clean up tasks having nothing to clean up, because it waits for _analyzed data_ to purge, but the system is filled with _un_-analyzed files.

#### To disable threading (to test this theory):
In the `server.py` file, found in `~/BirdNET-Pi/scripts`, change line 506 from `thread = threading.Thread(target=handle_client, args=(conn, addr))` to simply `handle_client(conn, addr)` and remove line 507. You can also remove the `import threading` line near the top of the file.

matthew73210, 2023-01-08:
I've given you recommendation a wir. At first nothing happened but after restarting all the services the thing is on a roll. I'll report back on stability.
Out of curiosity why do 2 threads open the same file?

mcguirepr89, 2023-01-08:
https://towardsdatascience.com/unraveling-pythons-threading-mysteries-e79e001ab4c#:~:text=The%20threading%20library%20makes%20the%20child%20processes%20able%20to%20access%20and%20modify%20the%20same%20variables%2C%20however%20the%20GIL%20does%20not%20prevent%20race%20conditions%20and%20we%20have%20to%20use%20lock%20to%20prevent%20this%20from%20happening. That whole write up is great, but I've highlighted the most relevant portions. In essence, threading was included without locking things down properly. If the CPU is fast enough to create the race conditions, then we see the problem.

matthew73210, 2023-01-08:
> [Threads can race if you don't lock things down properly -- which I did not](https://towardsdatascience.com/unraveling-pythons-threading-mysteries-e79e001ab4c#:~:text=The%20threading%20library%20makes%20the%20child%20processes%20able%20to%20access%20and%20modify%20the%20same%20variables%2C%20however%20the%20GIL%20does%20not%20prevent%20race%20conditions%20and%20we%20have%20to%20use%20lock%20to%20prevent%20this%20from%20happening.) That whole write up is great, but I've highlighted the most relevant portions. In essence, threading was included without locking things down properly. If the CPU is fast enough to create the race conditions, then we see the problem.

Nice read, I had tried to implement threading while I was at uni studying loadflow on HV grids. You either had an equation that would always find an answer but sometimes in an infinite time or one the would only find an answer (would still put out a number but very false) if the initial condition was close. You could do multiple variables at the same time and kill the ones when finished.
What is the use here of reading the file twice through two threads? I'm on an orange pi which speeds through a 30s file in under a second. Maybe another option than cutting MT would be to restart the service if it detects say 10 files waiting for analysis?