mcguirepr89/BirdNET-Pi #631: Add optional system name to web page header

Clumpton, 2022-10-30:
**Is your feature request related to a problem? Please describe.**
I have two systems running here (front and back gardens). In my browser both tabs just say "Birdnet-Pi" and so does the page header. I have to look at the url in the address bar to see which one is which.

**Describe the solution you'd like**
Add an optional "System Name" in the settings so that tabs & pages indicate which system you are viewing.

**Describe alternatives you've considered**
Thought of hacking the html/css myself, but that would probably not survive any updates.

**Additional context**
Being able to changing page background colour (or just toolbar colour) would be a luxury addition to helping identify the system.

lloydbayley, 2022-11-08:
I like this idea too as I am thinking about deploying a second installation at the front of the house. Believe it or not, there are quite a few more in the trees at the front that the back mic doesn't pick up.

I'm not quite sure if it's being looked at at the moment by anyone but I can't see why it can't be done at some stage. If this isn't being looked at, I might investigate to see about it if that is ok with the others. I really don't know what the etiquette is with contributing and don't want to step on any toes.

lloydbayley, 2022-11-09:
Right-oh! Will have a crack at it. Just was worried that someone else might have been working on it.
BTW: Ethan, I've spoken to Patrick and have said it's ok to give you my email address which will make contact easier as github doesn't allow direct messages which I think is WEIRD! With a capital WE.

lloydbayley, 2022-11-09:
Bah, I think I'm almost there but having great strife dealing with not having special, code-breaking, characters in the name. preg_replaces abound! Still working on it.

Clumpton, 2022-11-09:
> Bah, I think I'm almost there but having great strife dealing with not having special, code-breaking, characters in the name. preg_replaces abound! Still working on it.

Just a thought... you can propose the hostname as the default, it should be different if the user has more than one system.

lloydbayley, 2022-11-09:
Good idea. Will give that some thought!

ehpersonal38, 2022-11-09:
> Right-oh! Will have a crack at it. Just was worried that someone else might have been working on it. BTW: Ethan, I've spoken to Patrick and have said it's ok to give you my email address which will [...]

Sure!! My email is my username at gmail . com, if you ever need to reach me for something.

lloydbayley, 2022-11-16:
I've completed this one now, the pull request was merged into the system so it now supports a site name. I think we can close this issue now!

lloydbayley, 2022-11-16:
Actually, don't close it yet please, I've identified a bug causing it not to show after an update. Will work on it.

ehpersonal38, 2022-11-18:
No worries, I'm not either! Thanks for your help in getting this feature request implemented! 🐤

closed 2022-11-18 by ehpersonal38