mcguirepr89/BirdNET-Pi #640: Apprise MQTT Notifications Eventually Stop

ahstevens, 2022-11-03:
I have set up an Apprise MQTT notification for new detections, and it tends to work for a few hours before ceasing to send new messages to the MQTT broker. Sending a test message from the Birdnet-Pi interface reactivates it for another few hours, but it inevitably fails to send notifications after some time.

My Apprise configuration is structured as follows:

```
mqtt://username:password@myMQTTbrokerIP:port/birdnet/detection
```

Notification Title is blank

Notification Body is
```
{"Common_Name": "$comname", "Scientific_Name": "$sciname", "Confidence_Score": $confidence, "link": "$listenurl", "Date": "$date", "Time": "$time", "Week": "$week", "Latitude": $latitude, "Longitude": $longitude, "Minimum_Confidence": $cutoff, "Sigmoid_Sensitivity": $sens, "Overlap": $overlap, "Image": "$flickrimage"}
```

Only "Notify each new detection" is checked.

Minimum time between notifications of the same species is set to 0.

Any thoughts on what might be causing this or how to further diagnose the issue would be very welcome.

Thank you for your work on this project!

lloydbayley, 2022-11-03:
Fancy this being here just as I re-enter the forums after a break. I have this problem too - EXACTLY the same behaviour and setup! I have checked it also by sending the notification via Prowl so it's not just MQTT. I couldn't find any solution so I've had to hack together some nasty code in a separate python program to get around this and turn off the notifications within the system. It sort of works for now but it's not a solution. Just letting you know that you're not the only one.

ehpersonal38, 2022-11-03:
Honestly pretty stumped on this one! However, I've just created a branch that adds a ton more debug information to the Apprise notification process, so if either of you want to switch to this branch and test things out I would greatly appreciate it!:

```
cd ~/BirdNET-Pi/scripts/
update_birdnet.sh -b apprise-things
restart_services.sh
```

You will then see a bunch of new debug information being printed in your "View Log." Hopefully once the issue occurs again on your machine, we'll have some more info to work with.

Thanks!

lloydbayley, 2022-11-04:
Ok Ethan, have made the required changes  but as it's nighttime, I may not get any notifications.
Also, there is no rhyme or reason as to when it will stop.
As soon as I notice it has stopped, is there a particular journalctl command you would like run?
Or should I see if I can hunt back for the stop point?

ehpersonal38, 2022-11-04:
Yes if you're familiar with using journalctl you can try hunting back to when it stopped and then sending a snippet of those logs so I can see the output both before and after the moment the bug occurs. Thanks!

lloydbayley, 2022-11-06:
Well Ethan, this is interesting. Since I switched to your branch, it has been working....I tried the secondary install that I use for testing and it stops but your branch doesn't (so-far). Will keep an eye on it but I'm starting to wonder how a try/except loop makes a difference. Will give it a few more days to see. Just keeping you in the loop.

ahstevens, 2022-11-06:
Just chiming in to say that I also switched to the debug branch provided, and haven’t seen the Approse notifications stop like they had been. I did make the switch in the late afternoon, so I’ll know for sure tomorrow morning once the birds are vocalizing again.

lloydbayley, 2022-11-06:
Ok, now that's REALLY weird! Will keep you all updated too.

ehpersonal38, 2022-11-06:
That's great to hear!

I did happen to make one other change on the slightest whim that it may be the problem. Previously the Apprise object would be reinitialized with every single notification sent, whereas the change I made to this new branch only initializes it once at the very top of the script. I thought that would help in case of a memory leak or something. But it's also very plausible that it's the try/catch like you suggested. Either way it doesn't really matter as long one of those fix it, so do keep me updated! Thanks.

lloydbayley, 2022-11-06:
That would drive me nuts not knowing which it is. However, so-far, happy to report it's STILL going. 
I'll keep you abreast of things (hehe, I said breast) - I'm off to bed now. Overtired and silly.

ehpersonal38, 2022-11-06:
Hehe, well you're free to remove the try/except block or revert the Apprise object initialization and see if the issue still occurs.

Have a good night!

lloydbayley, 2022-11-06:
Bit scared to mess with it. If I do (for purely testing purposes), I shall keep you informed. My usual M.O. is to do these things one at a time over a few days, so as we don't introduce more variables. G'night!

lloydbayley, 2022-11-07:
Ok. Update. Still running fine here. As first debug, am commenting out the enclosing try.

```def notify(body, title, attached=""):
#    try:
        print("def notify - called")
        if attached != "":
            print("def notify - 1")
            print(apobj.notify(
                body=body,
                title=title,
                attach=attached,
            ))
        else:
            print("def notify - 2")
            print(apobj.notify(
                body=body,
                title=title,
            ))
#    except Exception as e:
#        print("=============== EXCEPTION: "+str(e))
```

Will see how that goes. 
I think you might be onto something with the singular init, Ethan.

Will report in after this has run for a day or two.

ehpersonal38, 2022-11-07:
Hooray! I'm definitely interested in seeing what the culprit ends up being. Thanks for testing this. Also if you haven't already, make sure you unindent the code you showed above.

lloydbayley, 2022-11-07:
That's a good point. Thanks for the reminder.
It probably would have wet its nest on the next reminder! Nice spotting!

lloydbayley, 2022-11-08:
Ok. Well, I'm pleased to report that after removing the try enclosure, notifications remained up overnight and now into the afternoon here in sunny (that's a novelty lately) Sydney.

Ethan, I think we can say that the fix was the singular init change.
Unless @ahstevens has any input to the contrary after testing, I'll leave it to you to make the changes in the main repo and we can call this fixed!

After you make the change, we can switch back to the main repo for final monitoring.

Nice work all!

ahstevens, 2022-11-08:
I made the change to test reverting just the single init this afternoon, but because of daylight saving time ending here in New Hampshire, it got dark so early that the birds went to bed before there was adequate time to verify that this modification is the fix. Given the report from @lloydbayley that the try/catch removal doesn't break it; I am comfortable calling this one solved.

I'll make a quick comment in the morning confirming that the removal of the singular init stopped the Apprise notifications, just for closure's sake.

But, I think this issue has been tackled. Thank you to all involved; this is a great project and I appreciate the hard work that's been put into it!

lloydbayley, 2022-11-08:
Ok. Still going strong here. All chirps, peeps, twitters and tweets accounted for! Happy with everything here.

(When this gets closed, perhaps a few more issues could be tidied up as well if they are finished and I will try my best to assist with more!)

ehpersonal38, 2022-11-08:
I'm glad we were able to resolve this! I've just gone and pushed the singular init change to the main repo, you're both free to update and make sure all is well there. Thanks for being awesome testers for this issue, @ahstevens and Lloyd  !!

Closing this now but @ahstevens do let us know if your system went back to being broken again after reverting the singular init change. Might be worth mentioning to the Apprise folks to add to their wiki, or maybe it should've been common sense to implement it like that in the first place? Not sure.

🐤

closed 2022-11-08 by ehpersonal38

ahstevens, 2022-11-08:
Here's my promised quick update: reverting the singular init change did indeed break it after a while. Good sleuthing, everyone! Will update my installation from the main branch later today.