mcguirepr89/BirdNET-Pi #653: Birdnet-Pi won't update or record any calls

Scraggarax, 2022-11-16:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
A few days ago, my cat pulled down the microphone that I had sitting out an open window. He went for the cattail sock on it and pulled the thing off. Since then Birdnet-Pi stopped audio recording after a few minutes. I replaced the microphone with a new one and it started working fine.

Up until I ran an update on the 11th, and since then Birdnet-Pi hasn't recorded anything at all. When I go to run the updater, I get the below as an output.

`+ sudo -u pi git -C /home/pi/BirdNET-Pi reset --hard
error: unable to write file birdnet.conf-defaults
error: unable to write file homepage/index.php
error: unable to write file scripts/config.php
error: unable to write file scripts/install_config.sh
error: unable to write file scripts/update_birdnet_snippets.sh
fatal: Could not reset index file to revision 'HEAD'.
+ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi fetch origin main
From https://github.com/mcguirepr89/BirdNET-Pi
 * branch            main       -> FETCH_HEAD
+ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi switch -C main --track origin/main
error: Your local changes to the following files would be overwritten by checkout:
	birdnet.conf-defaults
	homepage/index.php
	scripts/config.php
	scripts/install_config.sh
	scripts/update_birdnet_snippets.sh
Please commit your changes or stash them before you switch branches.
Aborting
+ set +x
`

lloydbayley, 2022-11-16:
Nice followup! I think you've worked it out. Sounds like a corruption in there somewhere with the OS. I also note from the line: /usr/bin/xauth: unable to write authority file /home/pi/.Xauthority-n that it was having problems writing to the Xwindows file. If you don't need Xwindows and are happy running in CLI mode, things might be easier. I initially installed in desktop mode and then re-did mine with the lite version without GUI. Less stuff to go wrong. :)

Out of interest, when 'el cato' pulled the thing down, did it lose power/unplug at the time? (The Pi, not the cat).
It might just be a simple OS corruption error.

Scraggarax, 2022-11-16:
@lloydbayley The Pi didnt't lose power, but the microphone was unplugged from the USB sound card.

Turns out it didn't record anything after coming back up.  every time I ran `df -h /` it would tell me that my card was 100% full. I tried deleting songbird data, but that didn't help. Every time I ran `disk_check.sh` it would error out:

> pi@rpi4-8gb:~/BirdNET-Pi$ sudo disk_check.sh
> + df -h /
> + tail -n1
> + awk {print $5}
> + used=100%
> /usr/local/bin/disk_check.sh: 6: Bad substitution

I ended up running  bash ~/BirdNET-Pi/scripts/update_birdnet.sh and that seemed to have cleared up about 5% of the space... now to see if it records anything.  I do have it set to purge, so I'm not sure what happened.

Scraggarax, 2022-11-16:
@ehpersonal38 
[image]

ehpersonal38, 2022-11-16:
That indicates to me that your system is processing a huge backlog of audio recordings, probably because your analysis stopped for a few days while your microphone kept recording. Hence why your disk got filled up, too.

So at this point I would first check your disk space usage again. It should be dropping dramatically, but if it's not and your disk is still basically full then I would go to Tools -> Services -> Recording Service, and stop that service for a few hours while the analysis does its thing and clears the backlog.

lloydbayley, 2022-11-17:
I have this problem sometimes too because something (usually me tinkering or because I have auto-purge off - don't ask) caused a problem and I ran out of disk space so everything stops processing and the mic records up until the last puffabyte (not a real size) just to hammer it home that the disk is full! :)
I agree, let it run for a few hours and see if it sorts itself out.

Scraggarax, 2022-11-17:
Bummer. Looks like it filled up again, I'm gonna try to clean up again and shut down recording and hope that fixes... Will update. 

[image]
[image]

lloydbayley, 2022-11-17:
Oh dear! Yes, I'd go that route. Look forward to hearing how you went!

Scraggarax, 2022-11-17:
[image]

Deleted recordings from the 13th and 14th (damn, that's a lot of recording)

lloydbayley, 2022-11-17:
Quite the backlog! I got sick of the 32GB SD card I was using and today, I migrated the RPi4 to use a 256GB Flash drive instead. No more MicroSD. Boy is it fast!!!

Scraggarax, 2022-11-17:
So I went into the file manager and moved everything out of the `/pi/birdsongs/2022-november/15-Tuesday-analyzed` folder, then restarted the analysis service and that seemed to fix it. It cleared everything before that so the only folder left was `/pi/birdsongs/2022-november/16-wednesday` and now we've got a working output!

[image]

lloydbayley, 2022-11-17:
That look good. Are you going to try drip-feeding the other files in to see if it processes them? (I have no idea if it would now it's moved past that date...)

Scraggarax, 2022-11-17:
So I had to keep restarting the coreservices as it would hang after some time. But hey! It's good to go now.

I'm gonna upgrade my storage soon and do a fresh install. But until then, thanks for the help!

[image]

[image]

lloydbayley, 2022-11-17:
Great news! I didn't really do much but sometimes the benefit of someones' opinion and reassurance is of great help.
Glad you got it sorted!

Scraggarax, 2022-11-17:
Looks like it's now stopping again.

Im not sure how to proceed now

[image]

ehpersonal38, 2022-11-17:
I would try rebooting and letting it run again. I've noticed that it will stop randomly once or twice after mucking around with it (deleting files, stopping services, etc).

8f it still stops after a while then I'd assume it's a microphone issue, but journalctl will tell us more.

gubbins-inc, 2022-11-20:
Another reason to limit the number of unprocessed recordings.
[Issue 605](https://github.com/mcguirepr89/BirdNET-Pi/issues/605#issue-1410622831)

CaiusX, 2022-12-07:
Hi @gubbins-inc -  any update on this?  If solved please close the issue.

tigerwee, 2024-06-26:
### [`_****_`](url)