mcguirepr89/BirdNET-Pi #662: Missing detections

ccorcos, 2022-11-21:
I just started using BirdNetPi and I'm eagerly looking for detections but I'm a little disappointed not to be seeing many. Yet when I listen to live audio, there are some quite obvious bird chirps I can hear... 

The only thing I can think of is that there is a fair amount of static noise from the microphone. I tried, turning the input volume down, but I'm not sure if that helped or made it worse.

Any thoughts on what I should try?

Thanks!

ehpersonal38, 2022-11-21:
Can we see your "View Logs"?

ccorcos, 2022-11-22:
```
16:07:40---9.0;12.0-('Salpinctes obsoletus_Rock Wren', 0.060166493)
16:07:40---12.0;15.0-('Toxostoma redivivum_California Thrasher', 0.055574145)
16:07:44---Files loaded
16:07:44---RECORDING_LENGTH set to 15
16:07:53---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:07:38.wav --o BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:07:38.wav.csv --lat 39.XX --lon -120.XX --week 43 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:07:53---ANALYZING AUDIO... DONE! Time 1.5 SECONDS
16:07:53---WRITING RESULTS TO BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:07:23.wav.csv ... DONE! WROTE 0 RESULTS.
16:07:53---READING AUDIO DATA... DONE! READ 5 CHUNKS.
16:07:53---Date-time: 2022-11-21 16:07:38
16:07:55---0.0;3.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.048968785)
16:07:55---3.0;6.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.056508917)
16:07:55---6.0;9.0-('Salpinctes obsoletus_Rock Wren', 0.057425633)
16:07:55---9.0;12.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.06725109)
16:07:55---12.0;15.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.0521769)
16:07:58---Files loaded
16:07:58---RECORDING_LENGTH set to 15
16:08:09---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:07:53.wav --o BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:07:53.wav.csv --lat 39.XX --lon -120.XX --week 43 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:08:09---ANALYZING AUDIO... DONE! Time 1.5 SECONDS
16:08:09---WRITING RESULTS TO BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:07:38.wav.csv ... DONE! WROTE 0 RESULTS.
16:08:09---READING AUDIO DATA... DONE! READ 5 CHUNKS.
16:08:09---Date-time: 2022-11-21 16:07:53
16:08:11---0.0;3.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.060148086)
16:08:11---3.0;6.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.06640538)
16:08:11---6.0;9.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.053263515)
16:08:11---9.0;12.0-('Setophaga nigrescens_Black-throated Gray Warbler', 0.049038723)
16:08:11---12.0;15.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.06094395)
16:08:14---Files loaded
16:08:14---RECORDING_LENGTH set to 15
16:08:23---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:08:08.wav --o BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:08:08.wav.csv --lat 39.XX --lon -120.XX --week 43 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:08:23---ANALYZING AUDIO... DONE! Time 1.6 SECONDS
16:08:23---WRITING RESULTS TO BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:07:53.wav.csv ... DONE! WROTE 0 RESULTS.
16:08:23---READING AUDIO DATA... DONE! READ 5 CHUNKS.
16:08:23---Date-time: 2022-11-21 16:08:08
16:08:24---0.0;3.0-('Melanerpes formicivorus_Acorn Woodpecker', 0.12086501)
16:08:24---3.0;6.0-("Cyanocitta stelleri_Steller's Jay", 0.086017534)
16:08:24---6.0;9.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.06323477)
16:08:24---9.0;12.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.05321262)
16:08:24---12.0;15.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.049733374)
16:08:28---Files loaded
16:08:28---RECORDING_LENGTH set to 15
16:08:39---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:08:23.wav --o BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:08:23.wav.csv --lat 39.XX --lon -120.XX --week 43 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:08:39---ANALYZING AUDIO... DONE! Time 1.5 SECONDS
16:08:39---WRITING RESULTS TO BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:08:08.wav.csv ... DONE! WROTE 0 RESULTS.
16:08:39---READING AUDIO DATA... DONE! READ 5 CHUNKS.
16:08:39---Date-time: 2022-11-21 16:08:23
16:08:41---0.0;3.0-('Melanerpes formicivorus_Acorn Woodpecker', 0.088022836)
16:08:41---3.0;6.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.074701175)
16:08:41---6.0;9.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.061155558)
16:08:41---9.0;12.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.0688896)
16:08:41---12.0;15.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.074991256)
16:08:44---Files loaded
16:08:45---RECORDING_LENGTH set to 15
16:08:53---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:08:38.wav --o BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:08:38.wav.csv --lat 39.XX --lon -120.XX --week 43 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:08:53---ANALYZING AUDIO... DONE! Time 1.6 SECONDS
16:08:53---WRITING RESULTS TO BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:08:23.wav.csv ... DONE! WROTE 0 RESULTS.
16:08:53---READING AUDIO DATA... DONE! READ 5 CHUNKS.
16:08:53---Date-time: 2022-11-21 16:08:38
16:08:55---0.0;3.0-('Setophaga occidentalis_Hermit Warbler', 0.059289105)
16:08:55---3.0;6.0-('Psaltriparus minimus_Bushtit', 0.11178646)
16:08:55---6.0;9.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.06924075)
16:08:55---9.0;12.0-('Aimophila ruficeps_Rufous-crowned Sparrow', 0.0843348)
16:08:55---12.0;15.0-('Empidonax difficilis_Pacific-slope Flycatcher', 0.09381067)
16:08:58---Files loaded
16:08:58---RECORDING_LENGTH set to 15
[... cut ...]
16:10:09---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:09:53.wav --o BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:09:53.wav.csv --lat 39.XX --lon -120.XX --week 43 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:10:09---ANALYZING AUDIO... DONE! Time 1.5 SECONDS
16:10:09---WRITING RESULTS TO BirdSongs/November-2022/21-Monday/2022-11-21-birdnet-16:09:38.wav.csv ... DONE! WROTE 0 RESULTS.
16:10:09---READING AUDIO DATA... DONE! READ 5 CHUNKS.
16:10:09---Date-time: 2022-11-21 16:09:53
```

ccorcos, 2022-11-22:
Here's a video with Live Audio during those logs and you can hear all the bird noises going on...

https://www.dropbox.com/s/ome0zytv1lzw2es/Screen%20Recording%202022-11-21%20at%204.08.35%20PM.mov?dl=0

ccorcos, 2022-11-22:
Hmm. I'm using a [Movo M1 USB Lavalier Lapel Clip-on Omnidirectional Computer Microphone](https://www.amazon.com/gp/product/B0176NRE1G/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) I could have sworn I saw it in [this thread](https://github.com/mcguirepr89/BirdNET-Pi/discussions/39) but I'm not seeing it now...

I'll try [this one](https://www.amazon.com/gp/product/B016C4ZG74?psc=1) and report back!

lloydbayley, 2022-11-27:
Any updates on your tests with the new mic?

ccorcos, 2022-11-27:
It arrives tomorrow. I'll let you know!

ccorcos, 2022-12-04:
Hey! This mic has an audio jack so I had to order an audio jack → USB cable. And then it was raining / stormy here. But I tried it out and it's looking much better! Here's a before / after comparison for you.

[image]
[image]

Just to summarize for those who want a similar setup, this is what's I'm using:

- [PoP voice Professional Lavalier Lapel Microphone Omnidirectional Condenser Mic](https://www.amazon.com/gp/product/B016C4ZG74?psc=1)
- [Audio Jack → USB cable](https://www.amazon.com/gp/product/B09FZFRC54/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

closed 2022-12-04 by ccorcos