mcguirepr89/BirdNET-Pi #664: Compatibility with moving audio data to an external drive

simon-ball, 2022-11-26:
**Is your feature request related to a problem? Please describe.**
Related to both #430, and several issues about SD cards filling up.

A consistent weakness with (any) Raspberry Pi based project is that SD cards suck - they easily get corrupted by minor power transients, and in a data-heavy project like this, even medium sized cards reach end-of-life write activity relatively fast. 

That's unavoidable to an extent. SD cards a) tend to have bottom tier storage chips to start with and b) just aren't intended for this kind of write-heavy usage cycle; but they're what we have to start with, because that's the only built in interface a raspberry pi has. 

It would be ideal to be able to mount some external file system to provide bulk storage for both the extracted bird songs, and the unprocessed .wav data. 

**Describe the solution you'd like**
I have tried various solutions for doing this, but none of them have worked. I don't have a precise understanding of exactly _why_ those attempts haven't worked, but my best _guess_ so far (and I want to emphasise, it is a _guess_ - if I'm just missing something obvious, I'd love to hear what it is!) is that the webserver does not honour symlinks that cross devices, presumably for security reasons.

The other most likely possibility is that I've screwed up the permissions when moving data to an external drive. It's Linux and I'm not an expert, this is _always_ a possibility.

Note: I, personally, would not have an issue with something like this remaining as a strictly "power user" approach, i.e. no gui required, just definite support that it is _possible_. 

**Describe alternatives you've considered**
My current workaround is to mount my NAS under `/mnt/backup`, and run a daily cron job that copies the data from `/home/<user>/BirdSongs/Extracted` to that external. This helps avoid purging data I don't want purged, but it doesn't solve the issue that the SD card is being hammered by heavy write activity; and the recordings still become unavailable via the web interface. 

**Additional context**
Approaches I have tried so far, all of which have failed. In all cases, multiple services fail to start until I restore normal behaviour, pointing all birdnet-specific locations back to the µSD card
* Mount a network drive via CIFS under `/mnt/external`. Create a symlink that redirects `/home/<user>/BirdSongs` to `/mnt/external/BirdSongs`
```bash
mkdir /mnt/external
mount //nas.lan/birdsong /mnt/external
rsync -ravh ~/BirdSongs /mnt/external
rm -r ~/Birdsongs
ln -s /mnt/external/BirdSongs ~/BirdSongs
```

* Replace the network mount with a USB hard drive (ext4 formatted), otherwise the same (`/mnt/external` symlinked to `/home/user/BirdSongs`)
* Mount a USB hard drive (ext4 formatted, so symlink support is definite) under `/home/<user>/BirdSongs`, in order to avoid symlinking
```bash
mkdir /mnt/temp
mount /dev/sda1 /mnt/temp
rsync -ravh ~/BirdSongs/* /mnt/temp
umount /mnt/temp
rm -r ~/BirdSongs
mount /dev/sda1 ~/BirdSongs
```

* Mount a USB hard drive as `/home`, such that `/home/user/BirdNet-Pi` is on the same block device as `/home/user/BirdSongs`
```bash
mkdir /mnt/temp
mount /dev/sda1 /mnt/temp
rsync -ravh /home/* /mnt/temp
umount /mnt/tmp
mount /dev/sda1 /home
```

As noted, in all cases, services subsequently fail to start (although I admit that I haave failed to note _which_ services, so as noted, I am not 100% sure exactly _what_ is failing and whether it's an issue with my implementation, or the architecture of this project being hostile to external devices).

Approaches I have NOT YET TRIED (although I am planning to as time allows - first on the list is replacing the now-read-only 8gb µSD card in my current installation :D )
* moving `/var` to the USB drive as well as `/home`, i.e. try and relocate the entire webserver to be local to the USB drive, and therefore not have to worry about crossing boundaries
* USB booting and thus moving all of `/` to the external drive. This seems to be the most likely to work, but also the most annoying.

blogdemoi, 2022-11-26:
I think you can boot the Pi from a USB device, though I never tried.

lloydbayley, 2022-11-27:
What model Raspberry Pi do you have? If it's a 4, I have just moved mine over to completely use a 256GB USB Flash drive and can tell you how.

simon-ball, 2022-11-27:
3B+. As I noted, USB booting is the next step, if I can figure out how to make it work. I'm fairly sure older models support USB booting _somehow_, I just need to figure it out.

It's not my preferred approach - it would be a lot simpler to just mount and symlink to a separate drive if I can figure out what's actually stopping me (much easier to tie into my ansible setup scripts, for instance); but if it works, it'll do.

lloydbayley, 2022-11-28:
Fair-enough. I did see your note about the next steps but was just telling you that I'd done the whole thing that way. As I can't be of more assistance, I'll bow out.

ExperiMentor, 2022-12-05:
As others have at least partially written, the way to go is to run the entire system from a USB-connected SSD (Solid Sate Drive). These have the highest number of write-cycles permited (USB sticks are better than SD cards but still limited).
Raspberry Pi 4 will boot from a USB SSD out of the box.
Raspberry Pi 3B+ will boot from USB SSD after one-time setting of a firmware bit (see https://www.instructables.com/Booting-Raspberry-Pi-3-B-With-a-USB-Drive/)
Raspberry Pi Zero 2W will boot from USB SSD (though instructions vary and there seem to be a lot of reports of 'not working for me'.

To make a bootable SSD drive, simply write the operating system to it using Raspberry Pi Imager (exactly same process as writing the operating system to an SD card, except that it writes blisteringlky quickly!)

CaiusX, 2022-12-07:
@ExperiMentor 

Thank you for the thorough explorations - having as large a verifiable record of identification sound samples as possible is key to improving the validity of the system .  Keep us posted on progress.

closed 2022-12-13 by mcguirepr89