mcguirepr89/BirdNET-Pi #670: Not sure if it's using the "week" or "lat/lon" feature correctly.

jds3d, 2022-12-08:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
My BirdNETPi in Northern Virginia just gave me a 78% confidence Mississippi Kite observation.  In December, week 46.  There has never been a Mississippi Kite sighting north of Southern Florida in December-February.  They winter in Paraguay mostly.  

Here is a snippet of my log file showing that it's using lat, lon, and week in the input to analyze.py.  

16:44:52---RECORDING_LENGTH set to 3
16:44:52---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/December-2022/08-Thursday/2022-12-08-birdnet-16:44:47.wav --o BirdSongs/December-2022/08-Thursday/2022-12-08-birdnet-16:44:47.wav.csv --lat 39.XX --lon -76.XX --week 46 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --exclude_list BirdNET-Pi/exclude_species_list.txt --birdweather_id "IN_USE"
16:44:52---ANALYZING AUDIO... DONE! Time 0.2 SECONDS
16:44:52---WRITING RESULTS TO BirdSongs/December-2022/08-Thursday/2022-12-08-birdnet-16:44:44.wav.csv ... DONE! WROTE 0 RESULTS.
16:44:53---READING AUDIO DATA... DONE! READ 1 CHUNKS.
16:44:53---Date-time: 2022-12-08 16:44:47
16:44:53---0.0;3.0-('Scolopax minor_American Woodcock', 0.06156967)
16:44:53---RECORDING_LENGTH set to 3

**Expected behavior**
I expect Mississippi Kite to NOT even be in the list of possible birds in Week 46 at this Lat/Lon.  

**Screen shot**
[image]

jds3d, 2022-12-08:
Additionally, I just ran the following locally on my desktop (not on rpi):
python species.py --o example/species_list.txt --lat 38.xx --lon -77.xx --week 46

With the real lat/lon and got 79 species, and MIKI is NOT in the list.  So again, how is it getting MIKI at 78% confidence?

ehpersonal38, 2022-12-09:
I've also noticed that the analyze.py doesn't seem to care about the lat/lon you give it. You can try following my hacky little guide for getting around this issue if you'd like:

https://github.com/mcguirepr89/BirdNET-Pi/discussions/374#discussioncomment-2940777

jds3d, 2022-12-09:
Hmm. After more consideration, I don't think it's the lat/lon that's not being considered. I think it's the week. MIKI is seen in my area during the spring/summer. And I don't get any false positives on birds that are never here, like no European or Australian birds for example.

closed 2022-12-13 by mcguirepr89