mcguirepr89/BirdNET-Pi #673: Unable to serve up HTTP web

dbrianhendrix, 2022-12-15:
My site has stopped serving up web services and now simply says:

This site can’t be reached
192.168.1.7 refused to connect.
Try:
    Checking the connection
    [Checking the proxy and the firewall](chrome-error://chromewebdata/#buttons)

ERR_CONNECTION_REFUSED

It has been rebooted but same issue. Didn't want to change anything until discussed for troubleshooting steps.

mcguirepr89, 2022-12-15:
Hi, @dbrianhendrix 

Let's see exactly why caddy is refusing your connection.

Please issue this command and provide its output (review the text and redact and information you want to keep private):
1. `sudo systemctl restart caddy & journalctl -n100 --no-pager --no-hostname -ucaddy`

That command will attempt to restart `caddy` and then will spit out the logs from that attempt wherein we should see the exact reasons why you cannot connect.

dbrianhendrix, 2022-12-15:
Well, I won't bore you with the output. The command fixed the issue with the restart of caddy. I will definitely make a note of the command and put it in my birdnetpi toolbox.

However, now the password doesn't work for logging into Tools. I'll look through my notes and see if I can find how to do that pw reset. I seem to recall it had something to do with a caddyfile or something.

Thanks again.

dbrianhendrix, 2022-12-15:
Well, it also appears I've lost my ability to login using HTTPS.

dbrianhendrix, 2022-12-15:
It appears that a power loss or reboot does NOT return the Pi to a working website to connect to.

I've rerun the command above and here's the output:

birdnetpi:~/BirdNET-Pi$ sudo systemctl restart caddy & journalctl -n100 --no-pager --no-hostname -ucaddy
[1] 17710
-- Journal begins at Sun 2022-12-04 04:04:47 CST, ends at Thu 2022-12-15 10:27:49 CST. --
Dec 15 09:32:00 caddy[421511]: {"level":"error","ts":1671118320.16382,"logger":"pki.ca.local","msg":"failed to install root certificate","error":"failed to execute sudo: exit status 1","certificate_file":"storage:pki/authorities/local/root.crt"}
Dec 15 09:32:00 caddy[421511]: {"level":"info","ts":1671118320.1644487,"logger":"http","msg":"enabling automatic TLS certificate management","domains":["hendrixhabitatbirds.birdnetpi.com","fairviewtennessee.birdnetpi.com","birdnetpi.local"]}
Dec 15 09:32:00 caddy[421511]: {"level":"warn","ts":1671118320.165939,"logger":"tls","msg":"stapling OCSP","error":"no OCSP stapling for [birdnetpi.local]: no OCSP server specified in certificate","identifiers":["birdnetpi.local"]}
Dec 15 09:32:00 caddy[421511]: {"level":"info","ts":1671118320.1660235,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}
Dec 15 09:32:00 caddy[421511]: {"level":"info","ts":1671118320.1688714,"logger":"tls","msg":"finished cleaning storage units"}
Dec 15 09:32:00 caddy[421511]: {"level":"info","ts":1671118320.1706088,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
Dec 15 09:32:00 systemd[1]: Started Caddy.
Dec 15 09:32:00 caddy[421511]: {"level":"info","ts":1671118320.1752973,"msg":"serving initial configuration"}
Dec 15 09:33:38 systemd[1]: Stopping Caddy...
Dec 15 09:33:38 caddy[421511]: {"level":"info","ts":1671118418.9568684,"msg":"shutting down apps, then terminating","signal":"SIGTERM"}
Dec 15 09:33:38 caddy[421511]: {"level":"warn","ts":1671118418.9569645,"msg":"exiting; byeee!! 👋","signal":"SIGTERM"}
Dec 15 09:33:39 caddy[421511]: {"level":"info","ts":1671118419.093734,"logger":"tls.cache.maintenance","msg":"stopped background certificate maintenance","cache":"0x4000490ee0"}
Dec 15 09:33:39 caddy[421511]: {"level":"info","ts":1671118419.0958104,"logger":"admin","msg":"stopped previous server","address":"tcp/localhost:2019"}
Dec 15 09:33:39 caddy[421511]: {"level":"info","ts":1671118419.0961342,"msg":"shutdown complete","signal":"SIGTERM","exit_code":0}
Dec 15 09:33:39 systemd[1]: caddy.service: Succeeded.
Dec 15 09:33:39 systemd[1]: Stopped Caddy.
Dec 15 09:33:39 systemd[1]: caddy.service: Consumed 24.287s CPU time.
Dec 15 09:33:39 systemd[1]: Starting Caddy...
Dec 15 09:33:39 caddy[423143]: caddy.HomeDir=/var/lib/caddy
Dec 15 09:33:39 caddy[423143]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
Dec 15 09:33:39 caddy[423143]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
Dec 15 09:33:39 caddy[423143]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
Dec 15 09:33:39 caddy[423143]: caddy.Version=v2.5.0 h1:eRHzZ4l3X6Ag3kUt8nj5IxATprhqKq/wToP7OHlXWA0=
Dec 15 09:33:39 caddy[423143]: runtime.GOOS=linux
Dec 15 09:33:39 caddy[423143]: runtime.GOARCH=arm64
Dec 15 09:33:39 caddy[423143]: runtime.Compiler=gc
Dec 15 09:33:39 caddy[423143]: runtime.NumCPU=4
Dec 15 09:33:39 caddy[423143]: runtime.GOMAXPROCS=4
Dec 15 09:33:39 caddy[423143]: runtime.Version=go1.18.1
Dec 15 09:33:39 caddy[423143]: os.Getwd=/
Dec 15 09:33:39 caddy[423143]: LANG=en_GB.UTF-8
Dec 15 09:33:39 caddy[423143]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
Dec 15 09:33:39 caddy[423143]: NOTIFY_SOCKET=/run/systemd/notify
Dec 15 09:33:39 caddy[423143]: HOME=/var/lib/caddy
Dec 15 09:33:39 caddy[423143]: LOGNAME=caddy
Dec 15 09:33:39 caddy[423143]: USER=caddy
Dec 15 09:33:39 caddy[423143]: INVOCATION_ID=6a14ee8ef50b4791a2460e18dbefceb2
Dec 15 09:33:39 caddy[423143]: JOURNAL_STREAM=8:1719545
Dec 15 09:33:39 caddy[423143]: {"level":"info","ts":1671118419.301013,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
Dec 15 09:33:39 caddy[423143]: {"level":"warn","ts":1671118419.319563,"msg":"Caddyfile input is not formatted; run the 'caddy fmt' command to fix inconsistencies","adapter":"caddyfile","file":"/etc/caddy/Caddyfile","line":2}
Dec 15 09:33:39 caddy[423143]: {"level":"info","ts":1671118419.332122,"logger":"admin","msg":"admin endpoint started","address":"tcp/localhost:2019","enforce_origin":false,"origins":["//127.0.0.1:2019","//localhost:2019","//[::1]:2019"]}
Dec 15 09:33:39 caddy[423143]: {"level":"info","ts":1671118419.3372326,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x40004ce8c0"}
Dec 15 09:33:39 caddy[423143]: {"level":"info","ts":1671118419.3400319,"logger":"http","msg":"server is listening only on the HTTPS port but has no TLS connection policies; adding one to enable TLS","server_name":"srv0","https_port":443}
Dec 15 09:33:39 caddy[423143]: {"level":"info","ts":1671118419.3401732,"logger":"http","msg":"enabling automatic HTTP->HTTPS redirects","server_name":"srv0"}
Dec 15 09:33:39 caddy[423143]: {"level":"warn","ts":1671118419.3402288,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv1","http_port":80}
[... cut ...]
Dec 15 09:59:20 systemd[1]: caddy.service: Succeeded.
Dec 15 09:59:20 systemd[1]: Stopped Caddy.
Dec 15 09:59:20 systemd[1]: caddy.service: Consumed 54.490s CPU time.
-- Boot a263e35c830e4bbfbec9543ef559b940 --
Dec 15 10:00:26 systemd[1]: Starting Caddy...
Dec 15 10:02:12 systemd[1]: caddy.service: start operation timed out. Terminating.
Dec 15 10:02:12 systemd[1]: caddy.service: Failed with result 'timeout'.
Dec 15 10:02:12 systemd[1]: Failed to start Caddy.
Dec 15 10:27:44 systemd[1]: Starting Caddy...

mcguirepr89, 2022-12-15:
Hi, @dbrianhendrix -- please rerun the command with the amended extra ampersand:
```
sudo systemctl restart caddy && journalctl -n100 --no-pager --no-hostname -ucaddy
```

Please wrap the output in a code block by pasting the output after a line containing only \`\`\`

```
So that your log output is formatted for reading
                           like this is pre-formatted
```

dbrianhendrix, 2022-12-15:
Here's the output from your new command:

```
birdnetpi:~/BirdNET-Pi$ sudo systemctl restart caddy && journalctl -n100 --no-pager --no-hostname -ucaddy
-- Journal begins at Sun 2022-12-04 04:04:47 CST, ends at Thu 2022-12-15 13:56:55 CST. --
Dec 15 09:53:29 caddy[423143]: {"level":"info","ts":1671119609.9065757,"logger":"tls.cache.maintenance","msg":"stopped background certificate maintenance","cache":"0x40004ce8c0"}
Dec 15 09:53:29 caddy[423143]: {"level":"info","ts":1671119609.9071634,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
Dec 15 09:53:29 caddy[423143]: {"level":"info","ts":1671119609.9073758,"logger":"admin.api","msg":"load complete"}
Dec 15 09:53:29 systemd[1]: Reloaded Caddy.
Dec 15 09:53:30 caddy[423143]: {"level":"info","ts":1671119610.317003,"logger":"tls.issuance.acme.acme_client","msg":"trying to solve challenge","identifier":"hendrixhabitat.ddns.net","challenge_type":"tls-alpn-01","ca":"https://acme-v02.api.letsencrypt.org/directory"}
Dec 15 09:53:30 caddy[423143]: {"level":"info","ts":1671119610.4774451,"logger":"tls","msg":"served key authentication certificate","server_name":"hendrixhabitat.ddns.net","challenge":"tls-alpn-01","remote":"23.178.112.107:27318","distributed":false}
Dec 15 09:53:30 caddy[423143]: {"level":"info","ts":1671119610.4793615,"logger":"tls","msg":"served key authentication certificate","server_name":"hendrixhabitat.ddns.net","challenge":"tls-alpn-01","remote":"18.119.125.148:10838","distributed":false}
Dec 15 09:53:30 caddy[423143]: {"level":"info","ts":1671119610.52156,"logger":"tls","msg":"served key authentication certificate","server_name":"hendrixhabitat.ddns.net","challenge":"tls-alpn-01","remote":"35.92.9.112:23728","distributed":false}
Dec 15 09:53:30 caddy[423143]: {"level":"info","ts":1671119610.733575,"logger":"tls.issuance.acme.acme_client","msg":"validations succeeded; finalizing order","order":"https://acme-v02.api.letsencrypt.org/acme/order/556610676/152900199447"}
Dec 15 09:53:31 caddy[423143]: {"level":"info","ts":1671119611.59263,"logger":"tls.issuance.acme.acme_client","msg":"successfully downloaded available certificate chains","count":2,"first_url":"https://acme-v02.api.letsencrypt.org/acme/cert/034a03b4916cee8a25ae3b45fe81659cc3ee"}
Dec 15 09:53:31 caddy[423143]: {"level":"info","ts":1671119611.5935647,"logger":"tls.obtain","msg":"certificate obtained successfully","identifier":"hendrixhabitat.ddns.net"}
Dec 15 09:53:31 caddy[423143]: {"level":"info","ts":1671119611.5936074,"logger":"tls.obtain","msg":"releasing lock","identifier":"hendrixhabitat.ddns.net"}
Dec 15 09:59:20 caddy[423143]: {"level":"info","ts":1671119960.626706,"msg":"shutting down apps, then terminating","signal":"SIGTERM"}
Dec 15 09:59:20 caddy[423143]: {"level":"warn","ts":1671119960.6267936,"msg":"exiting; byeee!! 👋","signal":"SIGTERM"}
Dec 15 09:59:20 caddy[423143]: {"level":"info","ts":1671119960.6340935,"logger":"tls.cache.maintenance","msg":"stopped background certificate maintenance","cache":"0x40001209a0"}
Dec 15 09:59:20 caddy[423143]: {"level":"info","ts":1671119960.63572,"logger":"admin","msg":"stopped previous server","address":"tcp/localhost:2019"}
Dec 15 09:59:20 caddy[423143]: {"level":"info","ts":1671119960.6358004,"msg":"shutdown complete","signal":"SIGTERM","exit_code":0}
Dec 15 09:59:20 systemd[1]: Stopping Caddy...
Dec 15 09:59:20 systemd[1]: caddy.service: Succeeded.
Dec 15 09:59:20 systemd[1]: Stopped Caddy.
Dec 15 09:59:20 systemd[1]: caddy.service: Consumed 54.490s CPU time.
-- Boot a263e35c830e4bbfbec9543ef559b940 --
Dec 15 10:00:26 systemd[1]: Starting Caddy...
Dec 15 10:02:12 systemd[1]: caddy.service: start operation timed out. Terminating.
Dec 15 10:02:12 systemd[1]: caddy.service: Failed with result 'timeout'.
Dec 15 10:02:12 systemd[1]: Failed to start Caddy.
Dec 15 10:27:44 systemd[1]: Starting Caddy...
Dec 15 10:27:49 caddy[17720]: caddy.HomeDir=/var/lib/caddy
Dec 15 10:27:49 caddy[17720]: caddy.AppDataDir=/var/lib/caddy/.local/share/caddy
Dec 15 10:27:49 caddy[17720]: caddy.AppConfigDir=/var/lib/caddy/.config/caddy
Dec 15 10:27:49 caddy[17720]: caddy.ConfigAutosavePath=/var/lib/caddy/.config/caddy/autosave.json
Dec 15 10:27:49 caddy[17720]: caddy.Version=v2.5.0 h1:eRHzZ4l3X6Ag3kUt8nj5IxATprhqKq/wToP7OHlXWA0=
Dec 15 10:27:49 caddy[17720]: runtime.GOOS=linux
Dec 15 10:27:49 caddy[17720]: runtime.GOARCH=arm64
Dec 15 10:27:49 caddy[17720]: runtime.Compiler=gc
Dec 15 10:27:49 caddy[17720]: runtime.NumCPU=4
Dec 15 10:27:49 caddy[17720]: runtime.GOMAXPROCS=4
Dec 15 10:27:49 caddy[17720]: runtime.Version=go1.18.1
Dec 15 10:27:49 caddy[17720]: os.Getwd=/
Dec 15 10:27:49 caddy[17720]: LANG=en_GB.UTF-8
Dec 15 10:27:49 caddy[17720]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
Dec 15 10:27:49 caddy[17720]: NOTIFY_SOCKET=/run/systemd/notify
Dec 15 10:27:49 caddy[17720]: HOME=/var/lib/caddy
Dec 15 10:27:49 caddy[17720]: LOGNAME=caddy
Dec 15 10:27:49 caddy[17720]: USER=caddy
Dec 15 10:27:49 caddy[17720]: INVOCATION_ID=fd6d2f4c25d1445088098d6caaf7904d
Dec 15 10:27:49 caddy[17720]: JOURNAL_STREAM=8:102567
Dec 15 10:27:49 caddy[17720]: {"level":"info","ts":1671121669.6366777,"msg":"using provided configuration","config_file":"/etc/caddy/Caddyfile","config_adapter":""}
Dec 15 10:27:49 caddy[17720]: {"level":"info","ts":1671121669.6799848,"logger":"admin","msg":"admin endpoint started","address":"tcp/localhost:2019","enforce_origin":false,"origins":["//localhost:2019","//[::1 [... cut ...]
Dec 15 13:56:55 caddy[178085]: {"level":"info","ts":1671134215.5584583,"logger":"http","msg":"enabling automatic TLS certificate management","domains":["hendrixhabitat.ddns.net"]}
Dec 15 13:56:55 caddy[178085]: {"level":"info","ts":1671134215.5632617,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
Dec 15 13:56:55 caddy[178085]: {"level":"info","ts":1671134215.563503,"msg":"serving initial configuration"}
Dec 15 13:56:55 caddy[178085]: {"level":"info","ts":1671134215.5638664,"logger":"tls","msg":"finished cleaning storage units"}
Dec 15 13:56:55 systemd[1]: Started Caddy.

mcguirepr89, 2022-12-15:
Thanks for that log info. From what it seems from the logs and by visiting the site for which `caddy` most recently completed domain challenges (https://hendrixhabitat.ddns.net), there don't seem to be any issues with `caddy`.

My only thought on next steps to troubleshoot logging into using Basic Authentication (the Tools page) would be to double-check that you are entering the correct password. You can view what password is stored by running `grep 'CADDY_PWD' /etc/birdnet/birdnet.conf`. If the password you are entering matches what you find in `birdnet.conf`, then we can walk through resetting it again

dbrianhendrix, 2022-12-15:
The password being returned from the command ```grep 'CADDY_PWD' /etc/birdnet/birdnet.conf``` is being used with username ```birder``` and failing. The "Sign in" pop-up still states "Your connection to this site is not private" as HTTPS still does not work. Again, if I lose power or reboot, the RPI will boot up and not work HTTP or HTTPS.

mcguirepr89, 2022-12-15:
[image]
I'm afraid I will need more reliable information. That is proof that you are indeed being served a valid SSL certificate from Let's Encrypt that you were issued this morning. Something certainly is not right, but you can imagine I'm confused at this point.

dbrianhendrix, 2022-12-15:
I'm at a loss to explain. Just try to go to my site from your website that shows "- Tennessee, United States"
https://fairviewtennessee.birdnetpi.com/

It will not work with HTTPS.

dbrianhendrix, 2022-12-15:
I guess one thing at a time. What do you want me to do to reset password?

mcguirepr89, 2022-12-15:
That is not configured in you Caddyfile as is evident in the logs that you provided. I maintain that CNAME domain registration and can confirm with no doubt that it is still registered. You can verify that for yourself by looking it up with https://dnschecker.org/#CNAME/fairviewtennessee.birdnetpi.com that link.

Also, after reading the front page of the repo (I no longer have a BirdNET-Pi installation, so I'm helping by looking at the documentation), I see that the login username is not `birder` as you say you are using, which will fail every time, but is in fact [`birdnet`](https://github.com/mcguirepr89/BirdNET-Pi#access)

Please try the correct login and let me know if you have access to what is configured on your system. Remember that you cannot reach it at https://fairviewtennessee.birdnetpi.com because your web server (Caddyfile) is _not_ configured for that URL. To confirm this, once you _do_ sign into the Tools page, got to your "System Info" and scroll down to the "Caddyfile" section and you will see the address line does not contain the *.birdnetpi.com subdomain.

dbrianhendrix, 2022-12-15:
I have always used ```birder``` -- the user ```birdnet``` is not even in the /etc/passwd file.

mcguirepr89, 2022-12-15:
`/etc/passwd` is for *nix systems. Basic Authentication is an HTTP protocol. They are not related whatsoever and never have been.

dbrianhendrix, 2022-12-15:
Well, now I know! I'm now in with ```birdnet```.

mcguirepr89, 2022-12-15:
May I close this issue now?

dbrianhendrix, 2022-12-15:
Logging in was a minor issue. I still can't get HTTPS to work and if I reboot right now, it will NOT come up with a working HTTP or HTTPS page.

dbrianhendrix, 2022-12-15:
HTTPS was working on all these "sites" approximately 2 weeks ago.

mcguirepr89, 2022-12-15:
2 questions:

1. Did you perform the recommended step that I posted in the very first response to this thread? That would have increased your `caddy.service`'s `[Service] \TimeoutSec` setting, which is the problem.
2. Did you not see [this comment](https://github.com/mcguirepr89/BirdNET-Pi/issues/673#issuecomment-1353703549)? It's literally proof that HTTPS is working

dbrianhendrix, 2022-12-15:
Yes, I did those commands. They did not help.

HTTPS may be working for the address in your command but not the ones used by visitors to website.

https://fairviewtennessee.birdnetpi.com gives the following:

This site can’t provide a secure connection
fairviewtennessee.birdnetpi.com sent an invalid response.
ERR_SSL_PROTOCOL_ERROR

mcguirepr89, 2022-12-15:
I won't repeat this again: _that is not configured on your web server_.

You can _verify_ this by looking at the Caddyfile that is loaded. Does it say "fairviewtennessee" anywhere in that file?

Does `sudo grep -e '*fairview*' /etc/caddy/Caddyfile` provide _any_ result? I'm certain it doesn't.

dbrianhendrix, 2022-12-15:
I appreciate you "repeating" -- however, this WORKED before. Why it is NOT there anymore is a mystery to me. I do NOT understand how this all integrates together. If I did, I wouldn't be asking.

dbrianhendrix, 2022-12-15:
Do I need to add them back assuming they are now somehow missing?

dbrianhendrix, 2022-12-15:
I have never used https://hendrixhabitat.ddns.net to get to this site or shared that with others.

mcguirepr89, 2022-12-15:
I don't feel like recounting every step that has lead to this, but in essence:
1. You reached out to me for a subdomain that was not locatively semantic (a requirement)
2. so I offered to register a locatively semantic subdomain _and_ the one you wanted (hendrixhabitat.birdnetpi.com)
3. This required a manually configured Caddyfile
4. I made that Caddyfile for you
5. You have changed a setting somehwere and the custom Caddyfile was wiped
6. The Caddyfile that wiped it had your ddns.net subdomain configured
7. You forgot all of that
8. Then had trouble with the basic auth

Everything is working as it should right now, because you don't have your custom Caddyfile in place.

You can make this by adding the domains that you want to add _manually_ to your `/etc/caddy/Caddyfile` (`sudo nano /etc/caddy/Caddyfile`) and then reloading the `caddy` configuration: `sudo systemctl reload caddy`

It will take a minute to complete the domain challenges and then everything will be as you expect.

dbrianhendrix, 2022-12-15:
I did absolutely nothing to cause the Caddy file to be wiped. Your assumption I changed a setting is incorrect. Please just close this issue. I will reuse the Raspberry Pi 4 for something else and move on without the site.

Thank you for help.

closed 2022-12-15 by mcguirepr89