mcguirepr89/BirdNET-Pi #674: ERR_INSUFFICIENT_RESOURCES

jds3d, 2022-12-15:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
On the overview page, I'm getting repeated UI errors:

views.php?view=Overview:143          GET https://viennava.birdnetpi.com/overview.php?ajax_detections=true&previous_detection_identifier=/By_Date/2022-12-14/Eastern_Wood-Pewee/Eastern_Wood-Pewee-84-2022-12-14-birdnet-15:59:16.mp3 net::ERR_INSUFFICIENT_RESOURCES
loadDetectionIfNewExists @ views.php?view=Overview:143
(anonymous) @ views.php?view=Overview:170

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Overview Page.
2. It just keeps flashing/reloading
3. Hit F12 to bring up developer console.
4. See error ( possibly repeats forever, saw 40,000 of them before closing window)

**Expected behavior**
No error, loads normally.

**Screenshots**
[image]

**Your build**
[image]

mcguirepr89, 2022-12-15:
Hi, @jds3d -- thank you for filling out the issue template completely. I was very easily able to replicate the issue, though I hope that @ehpersonal38 will have the time and will to take a closer look with me. It seems to have to do with the way the `$iterations` value is calculated, though I'm not super sure where the problem lies.

Perhaps by placing these lines at the top of your `~/BirdNET-Pi/scripts/overview.php` file, we may have some more information to work with?
```
ini_set('display_startup_errors', 1);
error_reporting(E_ALL);
```

Pasting that on the line below the initial PHP opener `<?php` while also commenting out the `error_reporting(E_ERROR);` line with two forward slashes `//` (so that it is temporarily discounted while we troubleshoot in favor of the more verbose `error_reporting`) may be helpful. :crossed_fingers: 

The first ten lines of the new `overview.php` would look like the code block below:
```
<?php
ini_set('display_startup_errors', 1);
error_reporting(E_ALL);
//error_reporting(E_ERROR);
ini_set('display_errors',1);
ini_set('session.gc_maxlifetime', 7200);
session_set_cookie_params(7200);
session_start();
$myDate = date('Y-m-d');
$chart = "Combo-$myDate.png";
```

Solanicars, 2022-12-15:
As awesome as this project is!!!

Any idea how I can unsubscribe to every update????

Cheers

ehpersonal38, 2022-12-15:
This issue should be resolved as per a commit I just did.

mcguirepr89, 2022-12-15:
@Solanicars https://docs.github.com/en/account-and-profile/managing-subscriptions-and-notifications-on-github/managing-subscriptions-for-activity-on-github/managing-your-subscriptions

Solanicars, 2022-12-15:
Ah…

I could never figure out how to do it in the app …

TY!

Still can’t wait to try this device sometime!

closed 2022-12-23 by mcguirepr89

jds3d, 2023-01-03:
> This issue should be resolved as per a commit I just did.

Hi @ehpersonal38.  I am back from the holidays.  This fix you committed did not have any effect on my system.  Can we reopen this and continue looking?

ehpersonal38, 2023-01-03:
Hi, try updating again, it should be fixed

jds3d, 2023-01-03:
I just looked at your code.  It looks like it's my 3 second clip length that is the issue.  And yep, looks like that fix worked.  Thanks!  We can keep this closed now.

ehpersonal38, 2023-01-10:
Just want to make a note about the 3 second clip length, I changed my installation's recording length to that and it completely broke things. My disk space filled up because the analsys couldn't keep up with the number of recordings. 

I guess that seemingly doesn't affect your installation since you didn't mention any issues like that, but just a note for anyone else in the future