mcguirepr89/BirdNET-Pi #686: A multitude of issues/errors

trophyguy, 2022-12-26:
I've noticed my BirdNET page has taken quite a few seconds to display.
Sometimes maybe 10-15 before I see the recent detections.

[birdnet.cnyweather.com](url)

On December 25th, I noticed no detections in the middle of the afternoon.
So I looked at the log and it was blank.

Restarted birdnet within the webpage and the analyzing started again.
Log showed it was reviewing detections from December 22nd.
Kept an eye on it and it was processing everything.

December 26th checked again and there were no detections
so looking in the log there was an error code Error 28
[image]

Looking at Sys Info got this
[image]

I've had to restart it a few times after getting Error Code 28 and it is still analyzing everything from the previous day
at the time I'm writing this.

EDIT: And it has stopped again so i'll restart it once again.

mcguirepr89, 2022-12-26:
Hi, @trophyguy -- I'm sorry this has happened. The help I can offer is in helping you clear the disk space.
If possible, please connect to your installation's shell via SSH.

Once at a command line prompt, if you are using an RTSP stream as your source, run: 
```
rm -rfv ~/BirdSongs/StreamData/*
```

If you are _not_ using an RTSP stream as your audio source, continue to the steps below.

Paste the following at the command line to determine which directories ought to be removed:
```
cd ~/BirdSongs/December-2022 && du -cks | sort -rn | head -n
```

That command should show the directories under `~/BirdSongs` whose contents use the most disk space.

To remove the directories that have filled, please use the `rm -drv DIRECTORY` command, replacing `DIRECTORY` with the directory you need to remove. For example:
```
rm -drv ./22-Thursday ./23-Friday ./24-Saturday ./25-Sunday ./26-Monday
```

trophyguy, 2022-12-27:
I've had to continuously restart the services all afternoon. Still trying to analyze the last 12 hours of today.
I did see an error I believe in the log.

**Soundscape POST Response Status -  201**
and
**Detection POST Response Status -  201**

Not sure if thats something going on with my system.

closed 2022-12-28 by mcguirepr89