mcguirepr89/BirdNET-Pi #709: Can't write mp3 file to disk

Svardsten53, 2023-01-24:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
When testing the NEW model I have for the first time seen a mp3 file that can't be written to disk.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

**Additional context**
I suspect this is because the file name consists of two species names separated by a slash.

**Your build**
Raspberry Pi 4B

**Code or log snippets**

mcguirepr89, 2023-01-25:
Hi, Lars - your conclusion is correct. Please fix the species label and things will work again

Svardsten53, 2023-01-25:
The file name is created from the program's list of species and the user should not have to change it and probably shouldn't either, as we have seen that the species determinations in the new model do not work with any list.

The solution should be that when the file name is created, you replace the slash with some other character, e.g. an underscore.

mcguirepr89, 2023-01-25:
Hi, Lars -- I disagree. BirdNET-Pi's code should not be altered to accommodate a malformed species entry in the `labels.txt` file -- the entry should be fixed. Please make that correction and submit a pull request if you would like the repo to address this. The _only_ punctuation that is allowed in 'labels.txt` is the underscore character "\_".

Regarding "...as we have seen that the species determinations in the new model do not work with any list..." this is a result of a misunderstanding of how analysis works within `server.py`. The text file that gets loaded as the species labels is loaded as an indexes list -- the indices are used to label predictions, meaning each species must exist in its same index. You cannot just throw a `labels.txt` file in place with the species you want to detect and expect `server.py` to know what to do with the file. It expects a file that is exactly the size of the model, with the species in exactly the same order.

I hope that explanation is satisfactory, though if it is not, I hope you will be able to make the necessary changes so that you are satisfied.

Svardsten53, 2023-01-25:
Patrick, thanks so much for clarifying how this works! I thought this with multiple species names separated by a slash was a standard in BirdNET as it came with the installation for the new analysis model. It's easy for me to change in my list, but will more than one underscore per line work logically in the processing/analysis? Please, maybe one of you who can code can go through the species lists for all languages and fix this and also change to a capital letter at the beginning of the national species names so it will be the same for all countries? Otherwise, other users will encounter the same problem.

mcguirepr89, 2023-01-25:
Hi, Lars -- I see now what you mean -- the Swedish species labels have lots and lots of forward slashes. I don't know why that is the case for the non-English species list.

This `sed` command should do two things: 1) replaces all forward slashes with a "+", and 2) will capitalize the first letter of the translated species name:

```
sed -i 's/_[a-z]/_\U&/g;s/\/[a-z]/+\U&/g;s/\///g' PATH_TO_SWEDISH_LABELS.txt
```

Hopefully @ehpersonal38 will be able to put together some logic that will use the "+" character to handle parsing the label correctly.

Alternatively, you could run this `sed` command to only keep the first species entry before any forward slashes and then @ehpersonal38 will have not extra work:

```
sed -i 's/_[a-z]/_\U&/g;s/\/.*//' PATH_TO_SWEDISH_LABELS.txt
```

Svardsten53, 2023-01-26:
I have no problem fixing this myself, nor for almost everyone who runs the program in Sweden. I have recruited almost all of the users seen on BirdWeather and a few more and we keep in touch.

What I'm thinking of are the users in the 11 countries out of 25 that have forward slashes in their species lists. These are lists from the BirdNET team and we will not be able to change these I guess. If, for example, Anas crecca (Common Teal) flies over their location, the program will hang! To avoid these problems and the following issues reported here, this can be easily changed.

The easiest way is to change the forward slash to e.g. a space character when the file name is created, but keep the forward slash in other text on the pages, which looks best. Then the problem is permanently solved and we don't need to change and maintain the species lists from the Analyzer team.

Regarding the lack of a capital letter at the beginning of the species name, this only applies to Sweden, Norway and Finland. It's a cosmetic issue, but it's ugly. I have asked Stefan Kahl to change this in future lists.

**@ehpersonal38**
From what Patrick wrote above, you are the one who handles the coding now and I kindly ask you to fix this so that users in all countries will be happy. Thanks!

ehpersonal38, 2023-01-28:
Fixed!

> The easiest way is to change the forward slash to e.g. a space or underscore character when the file name is created, but keep the forward slash in other text on the pages, which looks best.

I tried implementing this, but I don't think it's possible due to how BirdNET-Pi works, but I could be wrong. As of now the code will just lob off all the characters after and including the /, so "Northern Cardinal/Red Bird/Big Red" will just become "Northern Cardinal"

Maybe if someone has more time they can code this to be better, but it fixes the fatal error which is the more important thing in my eyes

Svardsten53, 2023-01-28:
That solution works well. The file name is less important, the main thing is that the program continues with the analysis and does not stop writing the files. The analysis itself seems to continue. After I deleted the faulty files, the ones that came after immediately started writing.