mcguirepr89/BirdNET-Pi #712: Species stats and console hangs when accessed through web address and proxy

davidrutland, 2023-01-24:
I have two Raspberry Pis. The first (Pi1) hosts a bunch of other sites and service and runs Apache.

The second Pi (Pi2) is is a different part of the house, and only hosts BirdNetPi

I have bought another domain name and have an Apache conf on Pi 1, which redirects requests for my domain through to port 80 on the second Pi.

I'm able to access my BirdNet-Pi instance through my domain name, and I can log in, listen to the mic, recorded bird sounds etc. But the "species stats" page hangs after a "Please wait..." message.

The web console closes instantly with a "connection closed" message.

ehpersonal38, 2023-01-24:
The stats page is actually a separate service that runs on localhost:8501, meaning port 8501. So I think you'll need to also route port 8501 from Pi 1 to port 8501 on your Pi 2. 

Here are some other ports that need to be routed as well:

```
reverse_proxy /log* localhost:8080
reverse_proxy /stats* localhost:8501
reverse_proxy /terminal* localhost:8888
```

davidrutland, 2023-01-24:
All requests from Pi1 arrive at :80 on Pi2. And are picked up by Caddy. Caddy itself should proxy to :8501 because it matches with /stats*. Caddy doesn't accept external requests on those ports.

I'm not getting a 404, so the request from Pi1 is definitely getting through to Pi2 and to Caddy.

The bottom section of the page just gives up eventually

davidrutland, 2023-01-25:
This may be related. 

I have an indefinite "please wait" or hang  or connection closed when trying to access stats and logs through birdpi.local (direct to Pi2) when I have a VPN active. Every other part of the site is functional and active. It's just those three sections which are proxied by Caddy.

davidrutland, 2023-01-25:
I've turned on logging for Caddy (never used anything with Caddy before birdnet-pi). These are examples of relevant entries:

[birdpi_caddy_log_entries.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/10498337/birdpi_caddy_log_entries.txt)

mcguirepr89, 2023-01-25:
Please open this issue with your VPN service provider

closed 2023-01-25 by mcguirepr89