mcguirepr89/BirdNET-Pi #717: exception converting to int() in utils/notifications.py

hoover67, 2023-01-28:
Hi all, 

I just saw an exception in scripts/utils/notifications.py in line 47 when apprising whether to send a notification which apparently stopped all further processing (at least nothing was shown in the web interface nor in daemon.log).

Here's a quick workaround which hopefully might fix this issue: 

               try:
                    if int(timeim.time()) - species_last_notified[comName] < int(APPRISE_MINIMUM_SECONDS_BETWEEN_NOTIFICATIONS_PER_SPECIES):
                        return
                except:
                    return

All the best, Uwe

lloydbayley, 2023-01-31:
That's an interesting catchall and would seem to be ok but I'm not familiar-enough with the guts of it to pass comment. 
There could be a reason that it wouldn't work due to something further on in the code. I think @ehpersonal38 will be the one to check this. He is rather busy so give him a bit to check this out. :)

ehpersonal38, 2023-01-31:
I saw this and have no problem implementing it, but I did want to get to the root cause of this issue rather than just wrapping try/catch blocks around it and calling it a day (usually bad practice in the programming world, not that I ever follow best practice 😁).

I saw a thread a while ago where this exact error happened to a user who didn't have their Flickr API key set, but had Apprise notifications enabled with `$flickrimage` in the Notification Body. Can you confirm whether that is what's happening on your system or not?

lloydbayley, 2023-01-31:
Agreed. Prevention is better than the cure. I too, will be interested to see if it turns out to be something like that.

hoover67, 2023-01-31:
Hi,

the only apprise notifications I'm using are with telegram, so I guess it's
not related to this.

As for the quick fix I was just interested in not having to restart the
birdnet services anymore after this happened, often leading to a backlog of
processing wav files from over a day or so, and this try: / except: seems
to have worked a "first aid" to this problem.

In the meantime, I've set my flickr API key but only to have bird images on
the detections page, I am not using them in any notifications.

All the best, Uwe

ehpersonal38, 2023-01-31:
Interesting... well I guess I'll go ahead and push this change to `main` seeing as it won't cause any harm. I will have it print out the error in "View Logs" though instead of just returning. Thanks for the issue report!

closed 2023-01-31 by ehpersonal38