mcguirepr89/BirdNET-Pi #718: Some strange issue

Stepphenwolf2021, 2023-01-28:
I just make my birdnet-pi project up 3 days ago. Everything seems to be fine and I have my station registerd in BirdWeather as station no. 710. (https://app.birdweather.com/stations/710).

As I have to shutdown my raspberry pi 3B every night and start it every morning. I notice that sometimes the recording function didn't work after the raspberry pi start. (The play icon on the right top corner is dimmed and can not be clicked). However, after I ssh login into the raspberry pi,  everything back to normal. This happen 2-3 times when I have to restart the raspberry pi.

Is that normal ? or Do I miss something during installation process ?

Please advise. 

Thank you

blogdemoi, 2023-01-28:
Hello,
Out of topic... Is your Birdnet-Pi public? I would be happy to listen to Thai birds.
Thank you.

ehpersonal38, 2023-01-28:
Try running the following commands:
`sudo systemctl enable getty@tty1.service`
`bash ~/BirdNET-Pi/scripts/restart_services.sh` (this one will take a minute to run)

Stepphenwolf2021, 2023-01-29:
> 

Not yet, i am asking for fix public ip address from the service provider. Hope I can make it public within next week. However, you can listen Thai Bird at this site https://app.birdweather.com/stations/710

Cheers,

Stepphenwolf2021, 2023-01-29:
> `sudo systemctl enable getty@tty1.service`
> `bash ~/BirdNET-Pi/scripts/restart_services.sh`

Thank you ehpersonal38. Do I need to run this command every time I restart or just once. Sorry for asking, I am very new for raspberry pi.

ehpersonal38, 2023-01-29:
@Stepphenwolf2021 Just once 🙂

Stepphenwolf2021, 2023-01-30:
Thank you everyone.

Stepphenwolf2021, 2023-01-31:
@ehpersonal38  My birdnet-pi was not detect any bird for 2 days now. I did run your command many times, but it still does not work.

ehpersonal38, 2023-01-31:
I'll need to see your View Logs, or `/var/log/syslog`

DemBones78, 2023-01-31:
why do you have to shut it down every day ?  that's an odd problem to have.  if its a power issue they make a pi UPS 
https://www.amazon.com/gp/product/B08VS5HMLF/ref=as_li_tl?ie=UTF8&camp=1789&creative=9325&creativeASIN=B08VS5HMLF&linkCode=as2&tag=mklements-20&linkId=fbcfdd13281423cf5979cc2caf14312a&th=1

DemBones78, 2023-01-31:
[Stepphenwolf2021](https://github.com/Stepphenwolf2021)

on the index page under amazing what does it say ?  
my unit has been over run with birds and is a day behind, processing 24x7    maybe you are just behind because you are turning it off every night.      

look at the image i posted its from my setup.  mine shows no detection today because its still working on yesterday

[image]

Stepphenwolf2021, 2023-02-01:
> Here are my screen shot

[image]

> I'll need to see your logs with the following command: `tail -n 100 /var/log/syslog`

[image]

Stepphenwolf2021, 2023-02-01:
> on the index page under Analyzing what does it say?

Sorry, where is the index page ? I can't find it.

Stepphenwolf2021, 2023-02-02:
> A lot of people have problems with the Raspberry Pi 3B, so it's not really recommended anymore. See this thread for a few commands you can run to hopefully help it run better, but no guarantees: [#242 (comment)](https://github.com/mcguirepr89/BirdNET-Pi/discussions/242#discussioncomment-2694098)

Sorry for make thing confusing, this log is from another raspberry pi 4

ehpersonal38, 2023-02-13:
@Stepphenwolf2021 are you still facing this issue?

closed 2023-02-16 by ehpersonal38