mcguirepr89/BirdNET-Pi #729: Separate login for listening?

Svardsten53, 2023-02-07:
Would it be possible to create a separate login for "Live Audio" and "Spectrogram" for those who wish this? In Sweden, the program is used by at least two bird observatories and it would be good if the members could listen directly or see when the species are filled in the running spectrogram. The current login gives admin rights, which should not be spread. Maybe it can be solved as an option in the configuration.

ehpersonal38, 2023-02-08:
1. Run the following command, replacing `$PWD` with whatever password you want users to access live stream with: 
`caddy hash-password --plaintext $PWD`. Copy this output.
3. Open `/etc/caddy/Caddyfile`, find the line that says `basicauth /stream {`, and the line directly under it should say `birdnet <bunch of characters>`. Replace the bunch of characters with the hashword you got from step 1.
4. Execute: `sudo systemctl reload caddy`

Svardsten53, 2023-02-09:
Thanks! It works perfectly. I changed both user and password, didn't realize the solution was so simple.

ehpersonal38, 2023-02-09:
Great! Happy to hear that. 🙂  Closing this now as resolved

closed 2023-02-09 by ehpersonal38