mcguirepr89/BirdNET-Pi #734: Unicode Encode Error

lloydbayley, 2023-02-09:
Well, this is a bit of an odd one.
It has happened to me once before and I put it down to 'weird stuff happens' but it's happened again.
I noticed my recognitions had stopped and looked at the logs and saw the following:

[image]

Now the bit that didn't show in this one, was that in the last one (some months ago), mentioned Flickr so I'm wondering if there is some malformed text in Flickr descriptions...although, I'm not 100% sure.

I'm having a good think about how we could sanitise the text coming in to be send out via Apprise.

Anyone else seen this?

ehpersonal38, 2023-02-09:
Good spot! It's a little annoying that my try/except had an exception... >_>

I think you've described the problem pretty well, a weird character in the title of a Flickr image. I've just done a commit which may or may not address this issue. Time will tell I guess

Thanks again for the report!

closed 2023-02-09 by ehpersonal38

lloydbayley, 2023-02-10:
Well don't you get a gold star?! :) That was quick.
Will be interested to see your catchall. I shall keep an eye on this to see how it goes!
Thanks @ehpersonal38 !